esp8266/Arduino #1685: SPIFFS Creating Duplicate Files

craigfoo, 2016-02-26:
#### Hardware
Hardware:			Adafruit ESP8266 Feather Huzzah
Core Version:      	        Whatever core is current with PlatformIO

### Description

SPIFFS is creating duplicate files.

### Settings in IDE

Module:  			Adafruit ESP8266 Feather Huzzah
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		SERIAL
Reset Method:		On-board button

### Sketch/Description

 I'm using SPIFFS to create 3 files for reading and writing data. 

/data/config.json
/logs/errors.json
/logs/actions.json

`
#include <ArduinoJson.h>
#include "FS.h"

#define CONFIG_FILE       "/data/config.json"
#define ERROR_LOG_FILE       "/logs/errors.json"
#define ACTION_LOG_FILE      "/logs/actions.json"

bool loadConfig(const char* filename) {
     File file = SPIFFS.open(filename, "r");
     if (!file) {
       Serial.println("Failed to open config file");
       return false;
     }

     size_t size = file.size();
     if (size > 1024) {
       Serial.println("Config file size is too large");
       return false;
     }
     else if (size == 0)   {
      Serial.println("File is empty");
      return false;
     }
     std::unique_ptr<char[]> buf(new char[size]);
     file.readBytes(buf.get(), size);

     StaticJsonBuffer<200> jsonBuffer;
     JsonObject& json = jsonBuffer.parseObject(buf.get());

     if (!json.success()) {
       Serial.println("Failed to parse config file");
       return false;
     }

     const char* url = json["url"];
     const char* Token = json["Token"];

     Serial.print("Loaded url: ");
     Serial.println(url);
     Serial.print("Loaded Token: ");
     Serial.println(Token);

     file.close();
}

void setup() {
  Serial.begin(115200);
  Serial.println("");
  delay(1000);
  Serial.println("Mounting FS...");

  if (!SPIFFS.begin()) {
    Serial.println("Failed to mount file system");
    return;
  }

   StaticJsonBuffer<200> jsonBuffer;
   JsonObject& json = jsonBuffer.createObject();

   Dir dir = SPIFFS.openDir("/");
   while (dir.next()) {
      Serial.print(dir.fileName());
      File f = dir.openFile("r");
      Serial.println(f.size());
   }

   if (SPIFFS.exists(CONFIG_FILE)) {

      Serial.print(CONFIG_FILE); Serial.println(" exists!");
      if (!loadConfig(CONFIG_FILE))   {
         File configFile = SPIFFS.open(CONFIG_FILE, "w");
         json["url"] = "api.something.awesome";
         json["Token"] = "bd1ba53b5915krpnd14fd992b1d7349f49d8e5";
         json.printTo(configFile);

         configFile.close();
      }
   }
   else   {
      Serial.print("Creating "); Serial.println(CONFIG_FILE);
      File configFile = SPIFFS.open(CONFIG_FILE, "w");
      json["url"] = "api.something.awesome";
      json["Token"] = "bd1ba53b5915krpnd14fd992b1d7349f49d8e5";
      json.printTo(configFile);

      configFile.close();
   }

   if (SPIFFS.exists(ACTION_LOG_FILE)) {

      Serial.print(ACTION_LOG_FILE); Serial.println(" exists!");
      if (!loadConfig(ACTION_LOG_FILE))   {
         File actionFile = SPIFFS.open(ACTION_LOG_FILE, "w");
         json["url"] = "api.something.awesome";
         json["Token"] = "action";
         json.printTo(actionFile);

         actionFile.close();
      }
   }
   else   {
      Serial.print("Creating "); Serial.println(ACTION_LOG_FILE);
      File actionFile = SPIFFS.open(ACTION_LOG_FILE, "w");
      actionFile.close();
   }

   if (SPIFFS.exists(ERROR_LOG_FILE)) {

      Serial.print(ERROR_LOG_FILE); Serial.println(" exists!");
      if (!loadConfig(ERROR_LOG_FILE))   {
         File errorFile = SPIFFS.open(ERROR_LOG_FILE, "w");
         json["url"] = "api.something.awesome";
         json["Token"] = "error";
         json.printTo(errorFile);

         errorFile.close();
      }
   }
   else   {
      Serial.print("Creating "); Serial.println(ERROR_LOG_FILE);
      File errorFile = SPIFFS.open(ERROR_LOG_FILE, "w");
      errorFile.close();
   }
}

void loop() {

   Dir dir = SPIFFS.openDir("/");
   while (dir.next()) {
       Serial.print(dir.fileName());
       File f = dir.openFile("r");
       Serial.println(f.size());
   }
   delay(1500);
   Serial.println();

} 
`
On the first run, everything seems to be peachy:

### Console Output 1
`
Mounting FS...
Creating /data/config.json
Creating /logs/actions.json
Creating /logs/errors.json
/data/config.json80
/logs/actions.json0
/logs/errors.json0

/data/config.json80
/logs/actions.json0
/logs/errors.json0

/data/config.json80
/logs/actions.json0
/logs/errors.json0
`
On the second run (after all the files are created), it finds all the files, finds that the action and error files are empty attempts to write to them but it's just renaming them to config.json instead. Now I have 3 config.json files.

### Console Output 2

`Mounting FS...
/data/config.json80
/logs/actions.json0
/logs/errors.json0
/data/config.json exists!
Loaded url: api.something.awesome
Loaded Token: bd1ba53b5915krpnd14fd992b1d7349f49d8e5
/logs/actions.json exists!
File is empty
/logs/errors.json exists!
File is empty
/data/config.json80
/data/config.json48
/data/config.json47

/data/config.json80
/data/config.json48
/data/config.json47

/data/config.json80
/data/config.json48
/data/config.json47 
`

What am I missing? Why is it allowing me to have 3 identical file names?

igrr, 2016-03-10:
@craigfoo  Thanks for the report, we have updated SPIFFS library, the is fixed now in git.

sticilface, 2016-03-10:
Good, work.  I've been having this problem for a while, but could not really identify the cause, as it was random for me.  
I'll let you know if it solved too.  
As usual @igrr thanks for the hard work.

craigfoo, 2016-03-10:
Excellent @igrr! I will give it try soon and report back!

craigfoo, 2016-03-10:
@igrr, does the new code live here, https://github.com/esp8266/Arduino/tree/master/cores/esp8266/spiffs?

craigfoo, 2016-03-11:
I ran the same code I used to find the bug and it now works swimmingly! @igrr, good work sir. Thanks!

closed 2016-04-18 by igrr