esp8266/Arduino #1687: Esptool 0.4.7 broken?

WereCatf, 2016-02-27:
Using Nodemcu - devkit under Linux Mint 17.2 when trying to upload a sketch the upload-LED flashes briefly and then nothing happens, no error messages or anything, but also no sketch is uploaded. Reverting to the commit right before esptool was updated to 0.4.7 works as one would expect.

reaper7, 2016-02-27:
today updated esp8266 arduino version from github and windows version of esptool 0.4.7 from
https://github.com/igrr/esptool-ck/releases/download/0.4.7/esptool-0.4.7-win32.zip
 the same symptoms :(
sketch uploaded but not working and no error message
return to 0.4.6 and everything is ok

Rotzbua, 2016-02-27:
I confirm @WereCatf for board d1 mini and windows 8.1.
Replacing esptool.exe by old 0.4.6 worked. New 2.1.0 works.

jeffryr, 2016-02-27:
Confirmed on Linux Mint 17.3 64-bit that esptool-0.4.7 generates a segmentation fault on upload.  After building the latest version from https://github.com/igrr/esptool-ck.git with -g and running the non-stripped executable from the commandline, here is the backtrace:

````
m4a88tdm(jra-3555): gdb --args /home/jra/ESP-8266/esptool-ck/esptool -v -cd ck -cb 460800 -cp /dev/ttyUSB0 -ca 0x00000 -cf /home/jra/Arduino-1.6.5/Nokia-5110-HDC1000-ESP12e/Nokia-5110-HDC1000-ESP12e/Nokia-5110-HDC1000-ESP12e.bin
GNU gdb (Ubuntu 7.7.1-0ubuntu5~14.04.2) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/jra/ESP-8266/esptool-ck/esptool...done.
(gdb) run
Starting program: /home/jra/ESP-8266/esptool-ck/esptool -v -cd ck -cb 460800 -cp /dev/ttyUSB0 -ca 0x00000 -cf /home/jra/Arduino-1.6.5/Nokia-5110-HDC1000-ESP12e/Nokia-5110-HDC1000-ESP12e/Nokia-5110-HDC1000-ESP12e.bin
warning: the debug information found in "/lib64/ld-2.19.so" does not match "/lib64/ld-linux-x86-64.so.2" (CRC mismatch).

esptool v0.4.7 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
opening port /dev/ttyUSB0 at 460800
opening bootloader
resetting board
trying to connect
trying to connect
Uploading 337632 bytes from to flash at 0x00000000

Program received signal SIGSEGV, Segmentation fault.
__memcpy_sse2 () at ../sysdeps/x86_64/multiarch/../memcpy.S:196
196	../sysdeps/x86_64/multiarch/../memcpy.S: No such file or directory.
(gdb) bt
#0  __memcpy_sse2 () at ../sysdeps/x86_64/multiarch/../memcpy.S:196
#1  0x0000000000402b3c in memcpy (__len=1024, __src=<optimized out>, __dest=0x60e350 <flash_packet+16>) at /usr/include/x86_64-linux-gnu/bits/string3.h:51
#2  espcomm_upload_mem (src=<optimized out>, src@entry=0x7ffff7f75010 <error: Cannot access memory at address 0x7ffff7f75010>, size=<optimized out>, size@entry=337632) at espcomm/espcomm.c:411
#3  0x0000000000402ede in espcomm_upload_file (name=<optimized out>) at espcomm/espcomm.c:540
#4  0x000000000040134d in argparse_commcmd (num_args=1, arg_ptr=0x7fffffffe370) at argparse/argparse_commcmd.c:83
#5  0x00000000004010e5 in parse_arg (num_args=num_args@entry=2, arg_ptr=arg_ptr@entry=0x7fffffffe368) at argparse/argparse.c:143
#6  0x0000000000400f31 in main (argc=<optimized out>, argv=<optimized out>) at main.c:59
(gdb) 
````

WereCatf, 2016-02-27:
I would recommend @igrr go back to 0.4.6 until esptool is fixed.

igrr, 2016-02-27:
Sure, I'll fix everything in a few hours. Sorry for the trouble.

WereCatf, 2016-02-27:
@igrr Don't apologize, I'm pretty certain everyone here is just eternally grateful for all the hard work you and Links2004 and others are doing! :)

igrr, 2016-02-27:
Could you please try 0.4.8?
https://github.com/igrr/esptool-ck/releases/tag/0.4.8

reaper7, 2016-02-27:
v0.4.8 + Windows10 + Arduino esp8266 latest github
works OK on nodemcu0.9 and wemos d1 mini.

so, seems You fix it :)
TNX Igrr!

closed 2016-02-27 by igrr