esp8266/Arduino #1689: Can not download esp8266 board manager url

tortechnocom, 2016-02-27:
http://arduino.esp8266.com/stable/package_esp8266com_index.json

unic8s, 2016-02-27:
Same for me. Arduino IDE didn't handle the 404 correctly and just removed the whole board package.

TobiasBodewig, 2016-02-27:
I have the same problem:
Error downloading http://arduino.esp8266.com/stable/package_esp8266com_index.json

GizmoRA2, 2016-02-27:
Yup seems to only have just happened, as others have been working with it a couple of hours ago, hope it comes back online soon :) this is a very valuable tool

intelliarm, 2016-02-27:
same issue

tomdpi, 2016-02-27:
Same on stable & staging URLs:
Error downloading http://arduino.esp8266.com/stable/package_esp8266com_index.json
Error downloading http://arduino.esp8266.com/staging/package_esp8266com_index.json

tommck, 2016-02-27:
Me as well.  Tried it directly in the browser and inside Arduino IDE.  Guess I can try to get it to work with the source?

Omemanti, 2016-02-27:
aih aih,, same here

Rotzbua, 2016-02-27:
Server arduino.esp8266.com is down. No ping.

Temporary workaround:

1. add link to latest json from release actually: https://github.com/esp8266/Arduino/releases/download/2.1.0/package_esp8266com_index.json
1. go to boardmanager and update

note:
tool `xtensa` is missing
maybe you have do delete old 2.0.0 folder manually

TobiasBodewig, 2016-02-27:
I tried to install the package. But it doesn't work.

intelliarm, 2016-02-27:
@Rotzbua the json contains http://arduino.esp8266.com/win32-xtensa-lx106-elf-gb404fb9-2.tar.gz
It seems arduino.esp8266.com is down.

Zadinator, 2016-02-27:
Got the same problem.

Rotzbua, 2016-02-27:
@intelliarm that was what I wanted to express ;)
@TobiasBodewig  esptool 0.4.7 didnt worked for me (win7, wemos d1), I replaced it by old 0.4.6 and it works

JonathanFStewart, 2016-02-27:
Urgh, just as I was getting my project working, Oh well hope they get this fixed soon.

Zadinator, 2016-02-27:
@JonathanFStewart haha same. What you working on?

JonathanFStewart, 2016-02-27:
Im working on controlling some phillips hue lights. It took me quite a long time to figure out how to get the esp8266 to make an http put request, but I finally got it. How bout you?

tommck, 2016-02-27:
@JonathanFStewart what was difficult about doing a PUT?  I've never worked with it before, but was planning on doing a PUT or at least a POST with it

JonathanFStewart, 2016-02-27:
@tommck It was mostly in experience. There doesn't seem to be any libraries out there for sending put requests with the esp8266. Iv always used postman, or other libraries to generate http requests up to this point. Looking back on it it's actually pretty simple. I just didn't have any experience manually creating http requests before this.

igrr, 2016-02-27:
@JonathanFStewart ESP8266httpClient library is bundled with the esp8266 core — you can use it to send GET or POST or PUT.

On the main topic — the server is obviously down, I will check what's up when I get home in a few hours.

JonathanFStewart, 2016-02-27:
@igrr *dope I didn't see that before. Guess that's what I get for not reading the documentation, Off to change my code. Thanks!

Zadinator, 2016-02-27:
@JonathanFStewart That sounds cool man. I havent really tried HTTP requests, but I am making an auto-assembling wifi mesh of ESPs. I was just about to test the UDP transfer of messages but couldnt because the server played up :P

krzychb, 2016-02-27:
@igrr thank you for bringing hope on the main topic :smile: :+1:

Zadinator, 2016-02-27:
@igrr  Yeah, thanks a lot man!

JonathanFStewart, 2016-02-27:
@Zadinator  Wow that's really cool! any plans for the mesh network, or just doing it for fun?

AlexFreshner, 2016-02-27:
Phew, glad it's not me, I was tugging my remaining hair out (arduino noob here) and can now do something else in the meantime :dancer: :) 
Would be great if the error message in the Arduino IDE console was more descriptive (server down, for example). 
Thanks @igrr for looking into this :+1:

bmitov, 2016-02-27:
@igrr Thank you! BTW: You are doing a fantastic job with the ESP! Thank you for all of your work!

MoeAfilal, 2016-02-27:
@AlexFreshner  Same, was going crazy thinking I was doing something wrong

krzychb, 2016-02-27:
@igrr thank you for bringing http://arduino.esp8266.com back to life.
I have just installed fresh and stable 2.1.0, compiled and uploaded my sketch and everything works correctly :smile: :+1:

closed 2016-02-27 by igrr

BG4RFF, 2016-03-27:
some one help me
I I have just installed fresh and stable 2.1.0, arduino 1.6.8 form arduino.cc,but something is wrong.
the items is
Arduino:1.6.8 (Windows XP), 开发板："Generic ESP8266 Module, Serial, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS), ck, Disabled, None"

In file included from c:\documents and settings\localservice\local settings\application data\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\memory:62:0,

                 from C:\Documents and Settings\LocalService\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266\Esp.cpp:24:

c:\documents and settings\localservice\local settings\application data\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algobase.h:59:28: fatal error: bits/c++config.h: No such file or directory

 #include <bits/c++config.h>

                            ^

compilation terminated.

exit status 1
Error compiling for board Generic ESP8266 Module.

在文件 -> 首选项开启
“编译过程中显示详细输出”选项
这份报告会包含更多信息。
and another is

Arduino:1.6.8 (Windows XP), 开发板："Generic ESP8266 Module, Serial, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS), ck, Disabled, None"

D:\Arduino1.6.8\arduino-builder -dump-prefs -logger=machine -hardware "D:\Arduino1.6.8\hardware" -hardware "C:\Documents and Settings\LocalService\Local Settings\Application Data\Arduino15\packages" -tools "D:\Arduino1.6.8\tools-builder" -tools "D:\Arduino1.6.8\hardware\tools\avr" -tools "C:\Documents and Settings\LocalService\Local Settings\Application Data\Arduino15\packages" -built-in-libraries "D:\Arduino1.6.8\libraries" -libraries "D:\My Documents\Arduino\libraries" -fqbn=esp8266:esp8266:generic:UploadTool=esptool,CpuFrequency=80,FlashFreq=40,FlashMode=dio,UploadSpeed=115200,FlashSize=512K64,ResetMethod=ck,Debug=Disabled,DebugLevel=None____ -ide-version=10608 -build-path "C:\DOCUME~1\ADMINI~1\LOCALS~1\Temp\build61879de404a7e74c4508ada90b951e7e.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "D:\Arduino1.6.8\libraries\WiFi\examples\ConnectNoEncryption\ConnectNoEncryption.ino"
D:\Arduino1.6.8\arduino-builder -compile -logger=machine -hardware "D:\Arduino1.6.8\hardware" -hardware "C:\Documents and Settings\LocalService\Local Settings\Application Data\Arduino15\packages" -tools "D:\Arduino1.6.8\tools-builder" -tools "D:\Arduino1.6.8\hardware\tools\avr" -tools "C:\Documents and Settings\LocalService\Local Settings\Application Data\Arduino15\packages" -built-in-libraries "D:\Arduino1.6.8\libraries" -libraries "D:\My Documents\Arduino\libraries" -fqbn=esp8266:esp8266:generic:UploadTool=esptool,CpuFrequency=80,FlashFreq=40,FlashMode=dio,UploadSpeed=115200,FlashSize=512K64,ResetMethod=ck,Debug=Disabled,DebugLevel=None____ -ide-version=10608 -build-path "C:\DOCUME~1\ADMINI~1\LOCALS~1\Temp\build61879de404a7e74c4508ada90b951e7e.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "D:\Arduino1.6.8\libraries\WiFi\examples\ConnectNoEncryption\ConnectNoEncryption.ino"
"C:\Documents and Settings\LocalService\Local Settings\Application Data\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Documents and Settings\LocalService\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.1.0/tools/sdk/include" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DARDUINO=10608 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266  -DESP8266       "-IC:\Documents and Settings\LocalService\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266" "-IC:\Documents and Settings\LocalService\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\variants\generic" "C:\DOCUME~1\ADMINI~1\LOCALS~1\Temp\build61879de404a7e74c4508ada90b951e7e.tmp\sketch\ConnectNoEncryption.ino.cpp" -o "nul"
"C:\Documents and Settings\LocalService\Local Settings\Application Data\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Documents and Settings\LocalService\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.1.0/tools/sdk/include" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DARDUINO=10608 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266  -DESP8266       "-IC:\Documents and Settings\LocalService\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266" "-IC:\Documents and Settings\LocalService\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\variants\generic" "-IC:\Documents and Settings\LocalService\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\libraries\SPI" "C:\DOCUME~1\ADMINI~1\LOCALS~1\Temp\build61879de404a7e74c4508ada90b951e7e.tmp\sketch\ConnectNoEncryption.ino.cpp" -o "nul"
"C:\Documents and Settings\LocalService\Local Settings\Application Data\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Documents and Sett [... cut ...]
c:\documents and settings\localservice\local settings\application data\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algobase.h:59:28: fatal error: bits/c++config.h: No such file or directory

 #include <bits/c++config.h>

                            ^

compilation terminated.

使用 1.0  版本的库 SPI 在文件夹： C:\Documents and Settings\LocalService\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\libraries\SPI 
使用 1.2.7  版本的库 WiFi 在文件夹： D:\Arduino1.6.8\libraries\WiFi 
exit status 1
Error compiling for board Generic ESP8266 Module.

mcauser, 2016-04-20:
@igrr looks like arduino.esp8266.com is down again :(

mrpeu, 2016-04-20:
It worked again.

davidbliss, 2016-08-11:
If it helps someone else with this problem, I was unable to install using the Boards Manager. After trying many other things I tried a different WiFi network and it went through.

samiultahir, 2017-03-02:
I have the same problem here

carlosmatuku, 2017-04-07:
This one worked for me:
http://arduino.esp8266.com/staging/package_esp8266com_index.json

samiultahir, 2017-04-07:
I got it!
I had windows 7 installed then once i upgraded to windows 10 it worked for me!

virusvomit, 2017-04-08:
I had some problems with installing that ESP package. Now it works :)
First i had problem with Arduino Boards Manager. When i add url  to preferences, he still did not work. When i tried find ESP, i saw only classic arduino librabry (Arduino AVR Boards). I fixed it, when i download a JSON file and manually copied that file to \AppData\Local\Arduino15 folder. Then i could find ESP8266 package in boards manager. But, when i wanted install it, arduino download that package and write message "Error downloading http://arduino.esp8266.com/win32-xtensa-lx106-elf-gb404fb9-2.tar.gz". So i had to download that file and manually copy to folder \AppData\Local\Arduino15\staging\packages. Then i could finally install that ESP package

samiultahir, 2017-04-08:
you can use a program called nodemcu flasher and it will flash a preferred .bin to your device

virusvomit, 2017-04-08:
Yep, i tried that, but not working for me. I tried flash firmware (from https://nodemcu-build.com/) but not work. Was set correctly port, baudrate but did not work. When i click on flash, that program do nothing happened. On ESP flashing blue LED, but progres bar was still blank. Therefore i want to try flash throught Arduino IDE.

samiultahir, 2017-04-08:
Oh it did work for me though

iskakfatoni, 2017-05-05:
The site is down again...

tnapolet, 2017-05-05:
The ide console error I receive is "Error downloading http://arduino.esp8266.com/stable/package_esp8266com_index.json"
I have not been successful for yesterday or today. 
I am using a raspberry pi for my platform.
All has been well for the last few weeks of use.
The board manager will not display the 8266.

marcellocolombo, 2017-05-05:
I confirm the link doesn't work:
http://arduino.esp8266.com/stable/package_esp8266com_index.json"
Do you know if i can work with arduino and mac os with other link?

ColeJ2001, 2017-05-05:
Yeah same for me it's down again... 
This link lets me see the board package in boards manager at least: https://github.com/esp8266/Arduino/releases/download/2.1.0/package_esp8266com_index.json
But I'm still getting an error when I try to install it.

timothycoutlakis, 2017-05-05:
Down for me too.  What a crap place to host.

timothycoutlakis, 2017-05-05:
This is even a bigger issue than it seems - If you had previously downloaded the stuff, and open board manager, then the board library disappears and you can no longer compile!

michaelnewton, 2017-05-06:
In case it helps anyone, based on @virusvomit 's workaround for windows, on OSX i was able to use:
https://github.com/esp8266/Arduino/releases/download/2.1.0/package_esp8266com_index.json

Then manually download:
http://arduino.esp8266.com/osx-xtensa-lx106-elf-gb404fb9-2.tar.gz

Place the tar.gz file in /Users/[username]/Library/Arduino15/staging/packages/

Then install worked from board manager and was able to successfully upload a sketch.

Modelmat, 2017-05-06:
Working again.

senjacob, 2017-05-06:
I am having the same issue. So I used a workaround to install latest esp8266 board. 
Here are the steps I did.

1. Install [Fiddler](http://www.telerik.com/fiddler).
2. Created a folder as C:\staging_packages\
2. Downloaded the [package_esp8266com_index.json](http://arduino.esp8266.com/versions/2.3.0/package_esp8266com_index.json) to C:\staging_packages\
3. Added Auto responder in fiddler to take the downloaded file instead of fetching from github.
4. Open board manager. Now it should list esp8266 board.
5. Click to install the board. You might see downloading esp8266-2.3.0.zip has failed.
6. Find the url of esp8266-2.3.0.zip in package_esp8266com_index.json (This can vary depending up on the OS environment) and download the zip file to C:\staging_packages\
7. Add another autoresponder in fiddler to use the downloaded file. Now, you may need to change the URL protocol from https to http both in package_esp8266com_index.json and in fiddler autoresponder.
8. Repeat steps 5-7 for other files like xtensa-lx106-elf-gb404fb9-2, esptool-0.4.9, mkspiffs-0.1.2 to use the downloaded files.
9. Now install the esp8266 board :)

I have attached the edited package_esp8266com_index.json and the autoresponder file to import to fiddlers in ESP8266WorkaroundHelpers.zip. Note that the attached files are for windows environment. You may need to edit those for other environments.

[ESP8266WorkaroundHelpers.zip](https://github.com/esp8266/Arduino/files/980790/ESP8266WorkaroundHelpers.zip)

cooltheo17, 2017-05-06:
Still not working for me ;(

jvsalo, 2017-05-10:
Hi all,

I suddenly got this today. I inspected the traffic with Wireshark, and problem is that arduino.esp8266.com replies as follows:

HTTP/1.1 403 Forbidden: The owner of this website (arduino.esp8266.com) has banned your access based on your browser's signature (35c8bdffe6f35b1b-ua21).

Can someone here reach the esp8266.com guys to check this out?

jvsalo, 2017-05-10:
As a workaround, adding "-Dhttp.agent=mickey_mouse" to JAVA_OPTIONS in the arduino wrapper script works just fine.

kakaducsy, 2017-05-11:
https://ndsybkiot.wordpress.com/2017/05/05/how-to-fix-error-downloading-httparduino/

riteshp887, 2017-06-25:
github link provided here( https://github.com/esp8266/Arduino/releases/download/2.1.0/package_esp8266com_index.json )works.

nickjan, 2017-07-26:
Same here getting error http://arduino.esp8266.com/stable/package_esp8266com_index.json again and again wht
 to do??

marcellocolombo, 2017-07-26:
Hi,
i just tried to call the link and it's work!!!

Attached  you find the json ;)

Bye 

Marcello Colombo
PROJECT DEVELOPMENT & MANAGEMENT
 
 
Avvertenze ai sensi del DLGS 196/2003
 
Le informazioni trasmesse attraverso la presente e-mail ed i suoi allegati sono dirette esclusivamente al destinatario e devono ritenersi riservate con divieto di diffusione salva espressa autorizzazione. La diffusione e la comunicazione da parte di soggetto diverso dal destinatario è vietata dall'art. 616 e ss. c.p. e dal d.l.vo n.196/03. Se la presente e-mail ed i suoi allegati fossero ricevuti per errore da persona diversa dal destinatario siete pregati di distruggere tutto quanto ricevuto e di informare il mittente con lo stesso mezzo.

Data: Wed, 26 Jul 2017 01:53:03 +0000 (UTC)

riteshp887, 2017-07-26:
Put this link https://github.com/esp8266/Arduino/releases/download/2.1.0/package_esp8266com_index.json in your preferences column in your arduino ide instead of http://arduino.esp8266.com/stable/package_esp8266com_index.json and carry on the same process. You will be able to download. Best of luck

RAKTPS, 2017-11-18:
This one worked for me,I am on win 7
http://arduino.esp8266.com/staging/package_esp8266com_index.json

Vedjai, 2018-09-30:
i had the same problem but i got it solved
this is what i did
1) i downloaded the arduino ide from microsoft store instead of the zip file from arduino.cc
2) I opened the arduino ide (which was downloaded from microsoft store)
3) i opened file>preferences and in additional board manager urls i removed everything that was there and clicked ok
4) then i went to tools>boards manager and in the last was the esp8266.
5) i clicked on install and it got installed

if you get any error in between click on install again.

Hope it works for you all

Vedjai, 2018-10-01:
hi people
you will get many errors while downloading it but don't get sad
keep clicking on install again
the download starts from where it stopped
it too get errors 
i got around 25 errors but clicked on install and it got downloaded

cyres, 2018-10-23:
> This workaround worked for me. Though the URL was returning expected JSON it was returning error "Error downloading http://arduino.esp8266.com/stable/package_esp8266com_index.json" in IDE. I tried following and now I'm able to resolve this issue.
> 
> 1. Open preferences.txt from C:\Users\AppData\Local\Arduino15 ( this path may vary depending on your
>    OS) and remove value for "boardsmanager.additional.urls"
> 2. delete all .tmp files from this folder
> 3. copy & paste URL in File->preferences
> 4. Goto Board-> Manager board... now you dont see any error and ESP8266 should be there in the list
>    [image]

Its Working For ME! Tnx Bro

GMFalka, 2018-12-24:
> This workaround worked for me. Though the URL was returning expected JSON it was returning error "Error downloading http://arduino.esp8266.com/stable/package_esp8266com_index.json" in IDE. I tried [...]

It works! Thanks!!!

[5 more replies]