esp8266/Arduino #1690: ERROR DOWNLOADING http://arduino.esp8266.com/stable/package_esp8266com_index.json

sagitaurusram, 2016-02-27:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

spi-ke, 2016-02-27:
https://github.com/esp8266/Arduino/issues/1689

closed 2016-02-27 by igrr

sagitaurusram, 2016-02-28:
The ide console displayed " error downloading the esp board manager" , hence I had raised an issue, Thank you for the prompt response. It works now.