esp8266/Arduino #1696: errors from netif->output will bubble up to udp_send, but we aren't forwarding them to WiFiUdp class

everslick, 2016-02-28:
I see high packet loss when sending rapid UDP packets without yield() or delay(). A yield() after each packet does only help slightly, but delay(1) after each Udp.write() works around that bug.

Problem description:

The MAC layer seems to drop packets if the send queue is full, but this condition is not reported back to the application (endPacket() should return 0 in this case).

igrr, 2016-03-02:
I have addressed this in commit aa67d1c, however it appears that udp_sendto doesn't actually return errors when underlying packet buffer is full.

everslick, 2016-03-03:
so i guess it's up to espressif to fix this then. :-/ no idea where to
report this though.

closed 2016-04-18 by igrr

barsznica, 2016-09-29:
May be related to [my problem](Similar to my problem. [Issue #2340](https://github.com/esp8266/Arduino/issues/2340)).