esp8266/Arduino #1697: OneWire not working with v2.0 or v2.1

scottfuhrman, 2016-02-28:
It seems that OneWire 2.3.1 and 2.3.2 is not working with the latest stable build 2.1.0. Tested by using the Dallas sample sketch to read a DS18B20 sensor. Sensor always reports a value of -127 C.

Going back to 1.6.5 where OneWire is included, same sketch works fine.

scottfuhrman, 2016-02-29:
Also confirmed the older Onewire.h that used to be included at 1.6.5 still works with stable build 2.1.0 so that is the workaround for now.  Not sure if this is a bug with the esp8266 arduino project or the onewire project?

igrr, 2016-02-29:
Please verify that you have an external pull-up resistor on the data line. The official OneWire library configures the data pin as `INPUT`, unlike our fork, which used to configure it as `INPUT_PULLUP`.

scottfuhrman, 2016-02-29:
Thank you for the reply. You are correct, this change in behavior threw me off.  I configured the pin I am using as INPUT_PULLUP and the official library worked just fine.

closed 2016-02-29 by scottfuhrman