esp8266/Arduino #1698: compile error begin example - esp8266 - Blink

BorgMcz, 2016-02-28:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:		NodeMCU 0.9
Core Version:      	2.1.0

### Description

not compile 

### Settings in IDE

Module:  			NodeMCU 0.9 ESP8266 12
Flash Size: 		4MB
CPU Frequency:      80Mhz
### Sketch

/*
 ESP8266 Blink by Simon Peter
 Blink the blue LED on the ESP-01 module
 This example code is in the public domain
 
 The blue LED on the ESP-01 module is connected to GPIO1 
 (which is also the TXD pin; so we cannot use Serial.print() at the same time)
 
 Note that this sketch uses LED_BUILTIN to find the pin with the internal LED
*/

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);     // Initialize the LED_BUILTIN pin as an output
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, LOW);   // Turn the LED on (Note that LOW is the voltage level
                                    // but actually the LED is on; this is because 
                                    // it is acive low on the ESP-01)
  delay(1000);                      // Wait for a second
  digitalWrite(LED_BUILTIN, HIGH);  // Turn the LED off by making the voltage HIGH
  delay(2000);                      // Wait for two seconds (to demonstrate the active low LED)
}

### Debug Messages

C:\Documents and Settings\uzivatel\Local Settings\Data aplikacĂ­\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\variants\nodemcu" "C:\Documents and Settings\uzivatel\Local Settings\Data aplikacĂ­\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266\Esp.cpp" -o "C:\DOCUME~1\uzivatel\LOCALS~1\Temp\build7c955e20053892e726efe23569c11769.tmp\core\Esp.cpp.o"
In file included from c:\documents and settings\uzivatel\local settings\data aplikací\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\memory:63:0,

                 from C:\Documents and Settings\uzivatel\Local Settings\Data aplikací\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266\Esp.cpp:24:

c:\documents and settings\uzivatel\local settings\data aplikací\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\allocator.h:46:75: fatal error: bits/c++allocator.h: No such file or directory

 #include <bits/c++allocator.h> // Define the base class to std::allocator.

                                                                           ^

compilation terminated.

exit status 1
Chyba kompilace.

igrr, 2016-02-29:
This file should normally be present in the following directory:
`c:\documents and settings\uzivatel\local settings\data aplikací\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\xtensa-lx106-elf\bits/c++allocator.h`

Please check if the file is there. If not, please try deleting these two directories: `c:\documents and settings\uzivatel\local settings\data aplikací\arduino15\packages\esp8266` and `c:\documents and settings\uzivatel\local settings\data aplikací\arduino15\staging\packages`, close Arduino, and then open it again and install esp8266 boards manager package again.

If the file is there, then the issue may be due to the way gcc handles international characters in path names on windows. I noted that in one line the path component is `Data aplikacĂ­` and in another line it's `data aplikací`. I'm not sure these are considered the same on Windows.

igrr, 2016-02-29:
Looks like this is the same as #1631, closing this one.

closed 2016-02-29 by igrr