esp8266/Arduino #1699: ESP12-E : Panic from umm_malloc ( Inconsistent between similar ESPs )

uncletammy, 2016-02-29:
### Basic Infos

I have two ESP12E modules.  One came mounted to a NodeMCU breakout board.
https://drive.google.com/file/d/0BwFbwCQ9LBckUTdSR2RmdEVtMkE/view?usp=sharing

The other is a generic board that I have wired myself (see pics for wiring)
https://drive.google.com/file/d/0BwFbwCQ9LBckMTk4MWVVSUtGYTA/view?usp=sharing
https://drive.google.com/file/d/0BwFbwCQ9LBckdWhMUVRGNl9UMUk/view?usp=sharing

I am able to upload sketeches to both of them using the Arduino IDE (version 1.6.8 with version 2.1.0 of Arduino core for ESP8266).

Both boards produce identical output from the CheckFlashConfig sketch.  They behave the same until I try to use a sketch that connects to Wifi.  Then, the NodeMCU board works but the generic board panics.  Sketches that don't connect to Wifi work fine on both. My operating system is Debian (Jessie).

This is being continued from https://github.com/esp8266/Arduino/issues/1675 at the request of @Links2004 

#### Hardware
Hardware: ESP12E
Core Version:  2.1.0 ( I don't know which revision)

### Settings in IDE

Module: Generic ESP8266 Module
Flash Size: 4MB w/ 1MB SPIFFS (tried all 4MB settings) 
CPU Frequency: 80Mhz (tried 40Mhz also)
Flash Mode: qio
Flash Frequency: 80Mhz  (tried 40Mhz also)
Upload Using: SERIAL
Reset Method: ck

### Sketch

This is the original CheckFlashConfig script modified to connect to my local wireless network.

```cpp

/* 
 ESP8266 CheckFlashConfig by Markus Sattler
 
 This sketch tests if the EEPROM settings of the IDE match to the Hardware
 
 */

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <Hash.h>

ESP8266WiFiMulti WiFiMulti;

void setup(void) {
    Serial.begin(115200);
    Serial.setDebugOutput(true);

    uint32_t realSize = ESP.getFlashChipRealSize();
    uint32_t ideSize = ESP.getFlashChipSize();
    FlashMode_t ideMode = ESP.getFlashChipMode();

    Serial.printf("Flash real id:   %08X\n", ESP.getFlashChipId());
    Serial.printf("Flash real size: %u\n\n", realSize);

    Serial.printf("Flash ide  size: %u\n", ideSize);
    Serial.printf("Flash ide speed: %u\n", ESP.getFlashChipSpeed());
    Serial.printf("Flash ide mode:  %s\n", (ideMode == FM_QIO ? "QIO" : ideMode == FM_QOUT ? "QOUT" : ideMode == FM_DIO ? "DIO" : ideMode == FM_DOUT ? "DOUT" : "UNKNOWN"));

    if(ideSize != realSize) {
        Serial.println("Flash Chip configuration wrong!\n");
    } else {
        Serial.println("Flash Chip configuration ok.\n");
    }

    for(uint8_t t = 4; t > 0; t--) {
        delay(1000);
    }

    Serial.println("Joining network");

    WiFiMulti.addAP("myNetworkName", "hackon!!");

    while(WiFiMulti.run() != WL_CONNECTED) {
        delay(100);
    }

    Serial.println("Connected using IP:");
    Serial.println(WiFi.localIP());

    Serial.println("And MAC address:");
    Serial.println(WiFi.macAddress());

}

void loop() {

    Serial.println("...still connected!\n");
    delay(5000);

}
```

### Debug Messages

##### From NodeMCU board
```cpp
	tail 0
	chksum 0x42
	csum 0x42
	Flash real id:   001640E0
	Flash real size: 4194304

	Flash ide  size: 4194304
	Flash ide speed: 40000000
	Flash ide mode:  DIO
	Flash Chip configuration ok.

	Joining network
	Connected using IP:
	10.100.20.136
	And MAC address:
	5C:CF:7F:0B:70:E0
	...still connected!

	...still connected!

	...still connected!
```
##### From generic board
```cpp

	Flash real id:   001640E0
	Flash real size: 4194304

	Flash ide  size: 4194304
	Flash ide speed: 80000000
	Flash ide mode:  QIO
	Flash Chip configuration ok.

	Joining network
	[WIFI][APlistAdd] add SSID: ATXHackerspace
	[WIFI] delete old wifi config...
	there is no poison after the block. Expected poison address: 0x3fff0ff8, actual data: 0x0 0xef 0xef 0xfe

	Panic /home/tink/arduino-nightly/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c:851 check_poison_block

	ctx: cont 
	sp: 3fff0090 end: 3fff04d0 offset: 01b0

	>>>stack>>>

```

### Stack

See image for decoded stack

[image]

kaeferfreund, 2016-02-29:
"Flash ide speed: 80000000" never worked properly for me. Did you try 40 Mhz?

uncletammy, 2016-02-29:
@kaeferfreund, I have tried 40Mhz but I got the same results :(

tablatronix, 2016-02-29:
try esptool erase_flash

uncletammy, 2016-03-05:
@tablatronix, were you able to reproduce this?  I don't quite understand your comment.

tablatronix, 2016-03-05:
Yup wdt resets, but I dont have exceptions handy, ill try to enable debugging and get any stacks if they help. Ill install gdb if i have time also. And i was getting a similar issue with the serial terminal example so ill probably be debugging stuff anyway.

tablatronix, 2016-03-05:
I am using the `wifimutli` example.
latest commit, arduino ide  hourly from 2/9

I get a wdt reset on `[WIFI] delete old wifi config...` 

```
[WIFI][APlistAdd] add SSID: ssid_from_AP_1
[WIFI][APlistAdd] add SSID: ssid_from_AP_2
[WIFI][APlistAdd] add SSID: ssid_from_AP_3
Connecting Wifi...
[WIFI] delete old wifi config...
[WIFI] start scan
WiFi not connected!
WiFi not connected!
WiFi not connected!
scandone
[WIFI] scan done
[WIFI] 7 networks found
       0: [1][5E:CF:7F:06:FA:0E] ESPap (-52)  
       1: [6][5C:96:9D:64:37:2B] harlet (-44) *
       2: [6][44:E1:37:24:5C:20] ATT128 (-82) *
       3: [11][E2:88:5D:4D:70:51] xfinitywifi (-80)  
       4: [11][58:23:8C:68:57:72] HOME-5772 (-52) *
       5: [11][5A:23:8C:68:57:74] xfinitywifi (-51)  
       6: [11][E0:88:5D:4D:70:5F] HOME-705F (-81) *

[WIFI] no matching wifi found!
WiFi not connected!
[WIFI] delete old wifi config...

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
```

and this also

```
[WIFI][APlistAdd] add SSID: ssid_from_AP_1
[WIFI][APlistAdd] add SSID: ssid_from_AP_2
[WIFI][APlistAdd] add SSID: ssid_from_AP_3
Connecting Wifi...
[WIFI] delete old wifi config...

Exception (2):
epc1=0x3ffe89b4 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3ffe89b4 depc=0x00000000

ctx: cont 
sp: 3fff00e0 end: 3fff05c0 offset: 01a0

>>>stack>>>
3fff0280:  feefeffe feefeffe feefeffe feefeffe  
3fff0290:  feefeffe feefeffe feefeffe feefeffe  
3fff02a0:  feefeffe feefeffe feefeffe feefeffe  
3fff02b0:  feefeffe feefeffe feefeffe feefeffe  
3fff02c0:  feefeffe feefeffe feefeffe feefeffe  
3fff02d0:  feefeffe 00000004 3fff0330 3fff0341  
3fff02e0:  40106b00 00000000 3fff1848 00000484  
3fff02f0:  3fff0470 feefeffe 000000a5 feefeffe  
3fff0300:  feefeffe 00000000 00000000 3fff0290  
3fff0310:  feefeffe feefeffe feefeffe feefeffe  
3fff0320:  3fff0340 3fff0330 00000004 feefeffe  
3fff0330:  feefeffe 3ffe8aae 3fff1848 00000000  
3fff0340:  000000a5 3fff13c4 feefeffe feefeffe  
3fff0350:  feefeffe feefeffe feefeffe 4010031d  
3fff0360:  feefeffe 00000da8 3ffef4bc 40100426  
3fff0370:  feefeffe feefeffe feefeffe 40100537  
3fff0380:  feefeffe 00001000 00000484 401006f5  
3fff0390:  00000484 00001000 0000007d 40106d2c  
3fff03a0:  40004b31 3fff03d0 0000001c 40222f7d  
3fff03b0:  401059ba 40223065 3fff13c4 0000007f  
3fff03c0:  0000007d 3fff0470 3fff13c4 0000007d  
3fff03d0:  ffffff00 55aa55aa 00000134 0000001c  
3fff03e0:  0000001c 00000032 00000064 0000007f  
3fff03f0:  40223454 3fff13c4 3fff13c4 000000ff  
3fff0400:  00000001 3fff0490 402235af 00000008  
3fff0410:  3fff13c4 000000ff 3fff0470 00000000  
3fff0420:  3fff1484 3fff04d1 00000001 4022363c  
3fff0430:  3fff0470 3fff13c4 3ffef4dc fffffffe  
3fff0440:  3fff0490 3fff76fc 3fff13c4 3ffef4dc  
3fff0450:  40223678 00000000 00000000 6c6f2065  
3fff0460:  402024ec 63206966 69666e6f 2e2e2e67  
3fff0470:  3fff0000 40106b80 3ffef4dc 00000008  
3fff0480:  3ffef4dc 0000002a 3fff04da 40203640  
3fff0490:  3fff0400 3fff04a0 00000008 0000002b  
3fff04a0:  3ffef4dc 00000008 60000000 0000007e  
3fff04b0:  6000001c ff000000 3ffef32c 3fffdad0  
3fff04c0:  3ffe86d1 53532064 3fff04a0 3fff04d0  
3fff04d0:  3fff074c 40202554 000a335f fffffffe  
3fff04e0:  3ffef4dc 3ffef32c 00000000 4020225f  
3fff04f0:  3fff0530 3fff0500 00000008 401008c8  
3fff0500:  00000018 feefeffe 3fff134c 000000a2  
3fff0510:  3fff12c0 00000000 00000018 40202398  
3fff0520:  3ffe87bb 00000001 3ffe895d 40203640  
3fff0530:  00000000 00000011 3ffe843b 3ffef598  
3fff0540:  3fffdad0 00000012 3ffef4dc 40202f05  
3fff0550:  3ffe895c 3ffef318 3ffef318 40202f05  
3fff0560:  3ffe842a 3ffef598 3ffef4dc 40202ff0  
3fff0570:  3fffdad0 3ffef318 3ffef4dc 3ffef598  
3fff0580:  3fffdad0 3ffef318 3ffef4dc 40201cc8  
3fff0590:  feefeffe feefeffe feefeffe feefeffe  
3fff05a0:  feefeffe 00000000 3ffef590 402034c3  
3fff05b0:  feefeffe feefeffe 3ffef5a0 40100958  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset
```

If i erase_flash it works.... until I reboot then it goes into loop again.

uncletammy, 2016-03-06:
@tablatronix, I get similar behavior and It's always right around "delete old wifi config..." .  It also goes away (for a while) upon erasing the flash.

tablatronix, 2016-03-06:
Looks like it immediatly calls wifi.disconnect after, which might be causing some bad flash writes. This might be another bug already known. Hmm

tablatronix, 2016-03-06:
Ok i reverted to 2.1 worked fine.
Wiped clean and installed latest dev, also working. Fine.

I did notice that my flash check is correct now., it was not matching before, but. Ost stuff was working fine and my settigs were correct. Ill keep and eye on it and see if it corrupts again or whatever.

wkraus, 2016-04-03:
@uncletammy Hi, I am getting exception(2) as well whenever I use  WiFi.begin(...) . Have You solved Your problem yet?
I use a WeMos D1 - old version, running well if not connecting to WiFi. even WiFi scan works.
Arduino 1.6.8 - core 2.1.0

uncletammy, 2016-04-05:
@wkraus, I'm not certain but I think I had two separate problems.  One was definitely solved [here](https://github.com/Links2004/arduinoWebSockets/issues/58)

The other one which was the crash immediately upon calling `WiFi.begin` seems to resolve upon using the suggestion by @tablatronix. Simply use esptool's erase_flash command  `esptool -p <port> erase_flash`

Over the past month I have flashed 100 generic ESP modules.  I cannot find a pattern in the frequency of occurrence.  It seems random.  I do know that once I addressed the issue that I've linked above AND turned off serial logging completely, the problem almost never occurred.  When it did though, the erase_flash command did the trick.

wkraus, 2016-04-06:
@uncletammy good advice. I will try erase_flash but don't know how to do it. Can I put it into an arduino sketch?

uncletammy, 2016-04-12:
@wkraus, no.  You have to use a command line tool called esptool.py.  See https://github.com/themadinventor/esptool for installation instructions and usage.

tablatronix, 2016-04-12:
I don't know your platform but for example windows.

`arduino\hardware\esp8266\tools\esptool\esptool.exe`

From command line run esptool
`esptool -cp <port name> -cd ck -ce`
eg. `esptool -cp COM7-cd ck -ce`

if you have esptool.py and python
`esptool.py -p <port name> erase_flash`

wkraus, 2016-04-16:
Thanks for Your help, I was out for a week and will be out again next week but will try it as soon as possible and post a comment.

igrr, 2016-05-10:
Tracking the same issue in #1997, closing this one.

closed 2016-05-10 by igrr