esp8266/Arduino #1700: Thanks for a great Arduino overlay for the ESP8266. I have used it successfully for custom ESP8266 boards I have been designing. Here is an example:

kriswiner, 2016-02-29:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

Duality4Y, 2016-02-29:
correct me if i am wrong, but the spi bus for the flash chip is weird to some hardware in the chip that makes it appear memory mapped. there for if the chip has some different communictation protocol it's not supported and no software can do anything about it.

kriswiner, 2016-02-29:
There is no reason for the memory map or commo protocol to be different, but indeed if they are different in some way and this is not supported by the Arduino IDE, then yes, this could be the reason for the flash not allowing the ESP program to run.

The ESP SDK doesn't seem to have any trouble with it though.

igrr, 2016-02-29:
Could you please do the following:
- capture and post here serial output from bootloader at 74880 baud
- use esptool.py to read back the contents of flash and then use diff or some other tool to verify that the binary is uploaded correctly

kriswiner, 2016-02-29:
Here is the program I compiled and flashed:
`#include "Wire.h"   
//#include <i2c_t3.h>
#include <SPI.h>
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
#include <ArduinoOTA.h>

void setup() {
      pinMode(15, OUTPUT);
      Serial.begin(57600);
      
    }
     
    void loop() {
      digitalWrite(15, HIGH);
      Serial.println("test");
      delay(500);
      digitalWrite(15, LOW);
      delay(500);
    }`

Here's what I see upon flashing. I don't have a choice of 74880 baud so I picked 57600:

`Sketch uses 235,194 bytes (22%) of program storage space. Maximum is 1,044,464 bytes.
Global variables use 34,820 bytes (42%) of dynamic memory, leaving 47,100 bytes for local variables. Maximum is 81,920 bytes.
C:\Documents and Settings\Kris\Local Settings\Application Data\Arduino15\packages\esp8266\tools\esptool\0.4.6/esptool.exe -vv -cd ck -cb 57600 -cp COM4 -ca 0x00000 -cf C:\DOCUME~1\Kris\LOCALS~1\Temp\buildbb6dc20b0e4f3063ce19eebcbd902ae8.tmp/esp8266blinky.ino.bin 
esptool v0.4.6 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to ck
	setting baudrate from 115200 to 57600
	setting port from COM1 to COM4
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	stat C:\DOCUME~1\Kris\LOCALS~1\Temp\buildbb6dc20b0e4f3063ce19eebcbd902ae8.tmp/esp8266blinky.ino.bin success
	setting serial port timeouts to 1000 ms
opening bootloader
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_open
Uploading 239344 bytes from C:\DOCUME~1\Kris\LOCALS~1\Temp\buildbb6dc20b0e4f3063ce19eebcbd902ae8.tmp/esp8266blinky.ino.bin to flash at 0x00000000
	erasing flash
	size: 03a6f0 address: 000000
	first_sector_index: 0
	total_sector_count: 59
	head_sector_count: 16
	adjusted_sector_count: 43
	adjusted_size: 02b000
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	setting serial port timeouts to 10000 ms
	setting serial port timeouts to 1000 ms
	espcomm_send_command: receiving 2 bytes of data
	writing flash
..........................................................................................................................................................................................................................................
starting app without reboot
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
closing bootloader
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
`

It looks just like it should and on my other ESP modules with the 8 Mbit Winbond flash I would expect the TX led to be blinking once a second; not with the 32 Mbit flash though.

I am not sure I know how to use esptool.py. Where can I get this and is there a description of how to use it?

igrr, 2016-03-01:
Sure, you can get esptool.py here: `https://github.com/themadinventor/esptool`. This page also has instructions on how to use it. You need `read_flash` command.

Also please do post the output from serial port captured at 74880 baud. Open Arduino serial monitor, set baud rate to 74880, and then reset the ESP. Copy&paste what you see.

igrr, 2016-03-01:
[image]

Attaching an illustration of what I mean when I say "Serial monitor" and "74880 baud".

kriswiner, 2016-03-01:
[image]

igrr, 2016-03-01:
Okay, this shows that the binary was not written correctly to the chip (see `csum err` message).
Now you need to use esptool.py to read back contents of flash memory and compare them to the original binary. See if there some random errors, or the binary wasn't written at all...

kriswiner, 2016-03-01:
Yes, but which contents. it seems I need to specify a memory address, memory size and a filename. What are these then?

igrr, 2016-03-01:
It's there in the log you have posted above:
`Uploading 239344 bytes from C:\DOCUME~1\Kris\LOCALS~1\Temp\buildbb6dc20b0e4f3063ce19eebcbd902ae8.tmp/esp8266blinky.ino.bin to flash at 0x00000000`

So you need to read 239344 bytes starting at address 0. Pick any file name you like. esptool.py will read flash contents into the file you have specified. Then compare the contents of the file with the contents of `C:\DOCUME~1\Kris\LOCALS~1\Temp\buildbb6dc20b0e4f3063ce19eebcbd902ae8.tmp/esp8266blinky.ino.bin`.

kriswiner, 2016-03-01:
Not sure what I am looking for, but the first page seems to be identical:

[image]

kriswiner, 2016-03-01:
So as I reported, the board seems to get flashed correctly (modulo the checksum error). It really has to be something in the Arduino IDE since when I load the SDK firmware binaries I can get the board to respond with AT commands. In other words, there is no hardware design error or component malfunction but something about the Arduino IDE memory allocation which messes up the ESP function? I did not specifically add a SPIFFS and my only choices seem to be either 3M + 1 M SPIFFS or 1M + 3 M SPIFFS. Does this matter? I tried flashing at 9600 baud (painfully slow) but the behavior didn't change. Can't be power since using the same setup the AT commands work. How can I tell if the bootloader is missing? Running out of options...

igrr, 2016-03-01:
Serial output you have posted above indicates that the first stage bootloader (inside the ESP8266 mask ROM) was not able to load second stage bootloader from flash successfully, signalling a checksum error.

Did i get that correctly that you were able to run the same binary produced by Arduino IDE on other board, which used a different flash chip? If that is the case, then we can rule out the possibility that Arduino IDE generated a binary with incorrect checksum.

Since you have verified that the flash chip contents match the original binary *(have you verified just the first & last pages or the full contents?)*, only one option is left. Apparently ROM bootloader is not able to read the contents of flash correctly, thereby reporting a checksum error. ROM bootloader reads flash configuration for the header of binary image. Initially it talks to the flash chip over SPI at low speed, reads a few bytes at the beginning, obtains flash configuration parameters stored in these bytes, and proceeds to set up flash interface according to these parameters. Could you please do a hex dump of the first 16 bytes of a binary (produced by the SDK) which works for you and responds to the AT commands?
Then we can compare flash configuration from that binary with flash configuration from the binary generated by Arduino.

kriswiner, 2016-03-02:
Yes, I was able to run this program successfully using the exact same board design but with the W2580 1 Mbit flash.

I didn't check that every entry of the .bin files was the same; I don't seem to have a diff function in the CMD window.

This what I get from the board with the 32 Mbit N25 flash after loading with the ESP SDK and using AT+RST:

`AT version:0.51.0.0(Nov 27 2015 13:37:21)
SDK version:1.5.0
compile time:Nov 27 2015 13:58:02
OK
AT+RST

OK

 ets Jan  8 2013,rst cause:1, boot mode:(3,0)

load 0x40100000, len 1396, room 16 
tail 4
chksum 0x89
load 0x3ffe8000, len 776, room 4 
tail 4
chksum 0xe8
load 0x3ffe8308, len 540, room 4 
tail 8
chksum 0xc0
csum 0xc0

2nd boot version : 1.4(b1)
  SPI Speed      : 40MHz
  SPI Mode       : DOUT
  SPI Flash Size & Map: 32Mbit(1024KB+1024KB)
jump to run user1 @ 1000

r
ready
`

kriswiner, 2016-03-02:
This is what the first 16 bytes look like starting at address 0x000000:
[image]

kriswiner, 2016-03-02:
[image]
[image]

igrr, 2016-03-02:
Okay, so this binary has "DOUT" flash mode (third byte 0 = QIO, 1 = QOUT, 2 = DIO, 3 = DOUT).
Your binary, OTOH, has "DIO" flash mode. We don't have DOUT mode configurable via Arduino IDE menu, so you will have to add it manually. 
Close the IDE and find `boards.txt` file in the ESP8266 package. If you have installed it via Boards Manager, it'll be somewhere under AppData/Arduino15/packages/esp8266.
Find the following two lines:
```
generic.menu.FlashMode.dio=DIO
generic.menu.FlashMode.dio.build.flash_mode=dio
```

and add another two next to them:
```
generic.menu.FlashMode.dout=DOUT
generic.menu.FlashMode.dout.build.flash_mode=dout
```
Save the file, start the IDE, select new mode from the list and build the binary again.

kriswiner, 2016-03-02:
Outstanding!

That worked. Thanks you very much for your expert an patient help!

Can this option be added to a future Arduino IDE release or must anyone who uses this 32 Mbit flash also have to add these lines manually?

In any case it works for me now, so thank you again!

closed 2016-03-02 by kriswiner