esp8266/Arduino #1702: Remove OTA option from the menu since it is now misleading

igrr, 2016-02-29:


krzychb, 2016-03-08:
As there are no comments I will proceed with updating [Arduino IDE](https://github.com/esp8266/Arduino/blob/master/doc/ota_updates/ota_updates.md#arduino-ide) section of  [OTA updates](https://github.com/esp8266/Arduino/blob/master/doc/ota_updates/ota_updates.md) documentation and reference platform package release 2.2.0 where OTA option option will be removed from IDE menu.
Krzysztof

krzychb, 2016-03-09:
As far as I have checked, besides OTA option, we should remove from boardx.txt the whole menu entries like below. Otherwise we will have hanging and unused ```Upload Using: Serial``` menu entry:

```
generic.menu.UploadTool.esptool=Serial
generic.menu.UploadTool.esptool.upload.tool=esptool
generic.menu.UploadTool.esptool.upload.verbose=-vv
generic.menu.UploadTool.espota=OTA
generic.menu.UploadTool.espota.upload.tool=espota
```

I did not find it used anywhere in platform.txt. If I am missing something please let me know. I am planning to update doc/opta_updates and boardx.txt within this week.

Krzysztof

scropion86, 2016-03-30:
is this a confirmation that BasicOTA.ino not working
i tried it on latest release 2.1 and Arduino 1.6.8 and still can't see the OTA port

using same sketch on platformio and after the normal upload using USB to serial
i can upload the sketch after getting the IP and add it as upload port in platformio.ini

grzegorz189, 2016-03-30:
https://github.com/esp8266/Arduino/blob/master/doc/installing.md
according to the instructions on my 1.6.8. works

closed 2016-04-18 by igrr