esp8266/Arduino #1703: Node MCU serial port does not appear in the IDE

SmartSouth, 2016-02-29:
### Basic Infos

#### Hardware
Hardware:			ESP-12E 
Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			Node MCU
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		SERIAL
Reset Method:		nodemcu

I have installed the driver in Mac OS X 'El Capitan' and it worked wonderfully for a while. The board's USB driver seems to work since it appears in the terminal ('system_profiler SPUSBDataType'):

            CP2102 USB to UART Bridge Controller:

              Product ID: 0xea60
              Vendor ID: 0x10c4  (Silicon Laboratories, Inc.)
              Version: 1.00
              Serial Number: 0001
              Manufacturer: Silicon Labs
              Location ID: 0xfa140000

Yet it does not appear in the 'Port' section of the menu in the Arduino IDE. Any ideas what am I doing wrong?

igrr, 2016-03-01:
Correct me if i'm wrong, but i don't think this issue has anything to do with ESP8266 Arduino core itself, because it doesn't come into play until you actually upload anything to the ESP. It's likely an issue with the Arduino IDE, serial port drivers, or serial port adapter.

Anyway, could you please 
- run the following two commands in the Terminal? Make sure you have your adapter plugged in.

    `ls -l /dev/tty.*`

    `ls -l /dev/cu.*`

- specify the version of Arduino IDE which you use

SmartSouth, 2016-03-01:
It is totally possible! Can you close the issue?

Greg Lorincz

closed 2016-03-04 by Links2004

natsu90, 2017-03-11:
@SmartSouth did u get this working? i have the same problem.