esp8266/Arduino #1704: Board manager download error

Adrian8x, 2016-02-29:
### Description

Can you help me please to install the esp8266 board?
I get this error: http://postimg.org/image/7vfobkeiv/  with http://arduino.esp8266.com/stable/package_esp8266com_index.json on Aditional board manager URL's and I don't know what to do to make it work.
I tried do install it using git version but i don't understand this part: Download binary tools (you need Python 2.7)
cd esp8266/tools
python get.py
Where I should write this command? Maybe it's a stupid question for you or others but I'm not familiar with command line especially in python.

I have installed Python and I wrote cd esp8266/tools but I get this error:
http://postimg.org/image/qor1n9lev/

Thank you!

igrr, 2016-03-01:
Please check if you can open the URL given in error message (http://downloads.arduino.cc/packages/package_index.json) with your browser. If it works, please check Arduino IDE preferences and make sure your proxy settings in the IDE match your system proxy settings.

Adrian8x, 2016-03-01:
Thanks a lot!
It's working with git version but it's more simple with a single click with mouse on get.py file instead of commands. I don't know why the programmers all the time use commands instead of some clicks in total commander. It's a little bit annoying for people who doesn't know python or MS DOS commands.

Anyway thank you again!

closed 2016-03-01 by Adrian8x

jeanfredson, 2017-04-24:
Just Disable FIREWALL to solve this too !

moondemon68, 2017-10-13:
@jeanfredson Thanks bro, love you mate! :)

GoCloudHub, 2018-09-29:
having a similar issue.. Windows FW turned off.. unable to download https://raw.githubusercontent.com/VSChina/azureiotdevkit_tools/master/package_azureboard_index.json
for the az3166 board.. can connect in browser though.. any ideas?

bhawthorne, 2018-10-10:
Same issue on the package_azureboard_index.json. Not sure what the right fix is, though as a work-around, I was able to save that file locally and reference it (file://c:\path....)