esp8266/Arduino #1706: ESP-01 on 2.1.0 compiles but no WiFi

kfeger, 2016-02-29:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			?ESP-01?
Core Version:      	?2.1.0?

### Description

Software written under 2.0.0 compiles but WiFi doesn't seem to run
The update from 2.0.0 to 2.1.0 was done via Arduino's board Manager without any Problem.
Runing on a Win10-Mashine

### Settings in IDE

(Unchanged from Default)
Module:  			?Generic ESP8266 Module?
Flash Size: 		?512kB/4Mb?
CPU Frequency:		?80Mhz?
Flash Mode:			?dio?
Flash Frequency:	?40Mhz?
Upload Using:		?SERIAL?
Reset Method:		?ck/Manual switch?

### Sketch

Applies to supplied examples like wifiscan.ino (with my credentials, of Course)  as well as to my own sketches

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

non, just "no Network found"

igrr, 2016-03-01:
Could you please attempt erasing flash (using `esptool -cp <port_name> -cd ck -ce` command), and then flashing the sketch again?

furyfire, 2016-03-01:
Same issue using Platformio and WiFiScan.ino sketch 
Fresh install of platformio packages, debugging enabled after some duration a fatal exception occurs 
See attached log file
Hardware: NodeMCU 0.9
Sketch: libraries/ESP8266WiFi/examples/WiFiScan/WiFiScan.ino
Enviroment: Platformio 2.8.4
esp8266/Arduino: 2.1.0
Windows XP

I have cleared the flash using https://github.com/themadinventor/esptool
Since https://github.com/igrr/esptool-ck dies on using the -ce flag 
Log file with -DDEBUG_ESP_CORE -DDEBUG_ESP_WIFI
[issue_1706.txt](https://github.com/esp8266/Arduino/files/152518/issue_1706.txt)

kfeger, 2016-03-01:
I will try that. Could you please Forward a schematic for the best possible reset/setup-circuit. The board I use right now does that all Manual (two Switches on reset and GPIO0), which suffices for Arduino IDE-Programming. I didn't manage to run ESPTool with that.

os40la, 2016-03-03:
I am having the same problem. I had v2.0.0 on arduino 1.6.7 and all was working fine. installed v2.1.0 and no longer can connect to WIFI AP. all sketches compile fine and upload.

For example the while loop in HelloServer.ino never connects, just keeps looping forever
  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

2. I then uninstalled 2.1.0 and installed 2.0.0 and all works fine.
3. Then I uninstalled 2.0.0 again and installed 2.1.0 and cannot connect to AP anymore. It seems that something in 2.1.0 is no longer working in my environment as well.

    Hardware: NodeMCU 1.0 ESP-12E
    Sketch: libraries/ESP8266WiFi/examples (all) no longer connect to WIFI
        scan networks sketch works - it lists AP's
    OS: Windows 7 (fresh install in VM with all windows updates) Can't get any cleaner than this.
    Arduino 1.6.7
    esp8266/Arduino: 2.1.0

Bladder777, 2016-03-05:
i have a problem that might relate; the sketch seems to uplaod fine, then the blue led (TX) is constant. as if nothing was done. I cant roll back the esp installation except to 2.0 and some other, then it tells me platforms and baords are missing. hwo can we solve this issue? 
I went out and bought a new esp01 to check if its the board, but found the issue with a new installation of OS and new board...
Please assist

devyte, 2017-10-18:
A full erase of the flash with the esptool should clear this type of problem, something that should be done anyways on a new ESP and also when flashing a different type of sketch (i.e.: not an update to a current one).
I'm not aware of WiFi issues on the ESP01 with latest git. Closing.

closed 2017-10-18 by devyte