esp8266/Arduino #1707: Optional domain name verification

schokocappucino, 2016-03-01:
Since fcf9c0d I can't connect to a webserver in my local network anymore. Is it possible to make domain name verification optional (and only verify the fingerprint)? I connect to the webserver using its internal ip adress, the certificate doesn't contain a domain name.

igrr, 2016-03-01:
We can split verification into two functions (`verifyDomainName` and `verifySHA1Fingerprint`) so that you can choose which part to verify.

As a temporary workaround, i think you can generate a (self-signed) certificate which will contain an IP address. You can follow [this procedure](http://stackoverflow.com/a/21494483/5659071), except that you need to specify `IP.1 = xxx.xxx.xxx.xxx` instead of `DNS.1 = ` in `alternate_names` config section.

devyte, 2018-07-12:
This should already be possible with fingerprint or allowSelfCert, especially with bearssl. Closing.

closed 2018-07-12 by devyte