esp8266/Arduino #1708: Exception (29) HTTPS-Request

MarcusF2015, 2016-03-01:
### Basic Infos

#### Hardware
Hardware:			ESP-12
Core Version:      	2.1.0

### Description

Problem description

The sketch does every second a HTTPS-Request. Randomly an exception (29) occures. After this the sketch stops.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB
CPU Frequency:	80Mhz?
Flash Mode:		DIO
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

### Sketch

```
#include <ESP8266WiFi.h>
#include <ESP8266WiFiAP.h>
#include <ESP8266WiFiGeneric.h>
#include <ESP8266WiFiMulti.h>
#include <ESP8266WiFiScan.h>
#include <ESP8266WiFiSTA.h>
#include <ESP8266WiFiType.h>
#include <WiFiClient.h>
#include <WiFiClientSecure.h>
#include <WiFiServer.h>
#include <WiFiUdp.h>

IPAddress ip(myIPAddress xx.xx.xx.xx);   

//WiFiClientSecure client;

const char* ssid = "MyWiFi";
const char* password = "MyPassword";

unsigned long ulMessTime;

IPAddress   host(149,154,167,198);
const int httpsPort = 443;

int restart = 0;

void setup() {
  ulMessTime=millis();
  restart = 0;
  Serial.begin(115200);
  Serial.println();
  Serial.print("connecting to ");
  Serial.println(ssid);
  WiFi.config(ip, dns, gateway, subnet);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  configTime(3 * 3600, 0, "pool.ntp.org", "time.nist.gov");
}

void loop() {
  
  if ((millis()- ulMessTime) >= 1000)
  {
    ulMessTime=millis();

    // Use WiFiClientSecure class to create TLS connection

    Serial.println(String(millis()-ulMessTime));
    
    WiFiClientSecure client;
    client.setTimeout(100);
    Serial.print("connecting to ");
    Serial.println(host);
    int connres=client.connect(host, httpsPort);
    Serial.println(connres);
    
    if (!connres) {
      Serial.println("connection failed");
      return;
    }

    String url;
    if (restart!=0)
    {
      url = "/botMyTelegramBot/getUpdates?offset=0&limit=1";
    }
    else
    {
      url = "/botMyTelegramBot/sendMessage?chat_id=35549815&text=Restart&reply_markup=";
      restart=1;
    }
    
    Serial.print("requesting URL: ");
    Serial.println(url);

    client.print(String("GET ") + url + " HTTP/1.1\r\n" +
                 "Host: " + host + "\r\n" +
                 "User-Agent: BuildFailureDetectorESP8266\r\n" +
                 "Connection: close\r\n\r\n");

    Serial.println("request sent");
    while (client.connected()) {
      String line = client.readStringUntil('\n');
      Serial.println(line);
      if (line == "\r") {
        Serial.println("headers received");
        break;
      }
    }
//    String line = client.readStringUntil('\n');
    String line = client.readString();
    Serial.println("reply was:");
    Serial.println("==========");
    Serial.println(line);
    Serial.println("==========");
    Serial.println("closing connection");

    Serial.println(String(millis()-ulMessTime));

  }

}

```
### Debug Messages

```
Exception (29):
epc1=0x4000e01a epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff14b0 end: 3fff17f0 offset: 01a0

>>>stack>>>
3fff1650:  3ffe9010 00000035 00000005 00000001  
3fff1660:  4020375a 3fff15a1 3fff062c 00000030  
3fff1670:  00000000 00000005 00000000 000a0017  
3fff1680:  ffffffff 3fffc6fc 00000001 3fff062c  
3fff1690:  00000000 3fff1720 00000005 40203c06  
3fff16a0:  3ffe8db0 3fff1720 3fff247c 40229fe3  
3fff16b0:  3fff16d0 00000000 00000000 3fff1640  
3fff16c0:  00004400 0000002a 00000a69 4022619d  
3fff16d0:  000a0017 3fff16f0 3fff07d0 3fff21dc  
3fff16e0:  0000000a 3fff062c 40203fa4 3fff062c  
3fff16f0:  0000000a 3fff21dc 3fff247c 4022a3c9  
3fff1700:  00000005 00000001 3fff178c 40204060  
3fff1710:  3ffe9d98 3fff21dc 3fff21dc 402038d7  
3fff1720:  0000000a 3fff062c 40203fa4 3fff07d0  
3fff1730:  0000000a 3fff1770 3fff21dc 402039a5  
3fff1740:  3ffe9d98 3ffe8db0 3fff1770 40205903  
3fff1750:  3ffe8d88 3fff0628 3fff178c 40205d02  
3fff1760:  3fff0634 3ffe8db0 3fff1834 40202846  
3fff1770:  3ffe91d0 00000000 00000064 0001b553  
3fff1780:  00000000 3fff238c 3fff21dc 3fff21fc  
3fff1790:  0000000f 00000000 00000000 00000000  
3fff17a0:  00000000 3fff7bfc 0000004f 0000004d  
3fff17b0:  00000001 5b7aa8c0 feefeffe feefeffe  
3fff17c0:  00000000 00000000 00000001 3fff07c4  
3fff17d0:  3fffdad0 00000000 3fff07bc 40203ff0  
3fff17e0:  feefeffe feefeffe 3fff07d0 40100958  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

```

[URL=http://www.directupload.net/file/d/4280/9clee6ce_gif.htm][IMG]http://fs5.directupload.net/images/160301/temp/9clee6ce.gif[/IMG][/URL]

MarcusF2015, 2016-03-02:
Is a work around available?

igrr, 2016-03-02:
*Ouch!* https://github.com/igrr/axtls-8266/blob/master/ssl/tls1.c#L1425 :anguished: 
I'll fix that and another related issue soon.

MarcusF2015, 2016-03-02:
Ok, I'll wait patiently.

igrr, 2016-03-02:
@MarcusF2015 could you please try with the latest git version?
Thanks.

mkeyno, 2016-03-02:
hi  @MarcusF2015  which library you use , last stable or staging ?

igrr, 2016-03-02:
As mentioned at the top of the OP, `Core Version: 2.1.0` — that's the latest stable version.

mkeyno, 2016-03-02:
dear @igrr  I've just update my library to this new library , is there any new library soon?

MarcusF2015, 2016-03-02:
@igrr Thank you very much for your high speed software maintenance.
Now I have got a stupid problem: How to switch to the latest git version with Arduino IDE on a Windows PC? readme.md desribes this only for Linux.

igrr, 2016-03-03:
On windows you can open tools directory and double-click `get.py` script, if you have python installed.

MarcusF2015, 2016-03-03:
Now it runs 9 hours without an exception!

MarcusF2015, 2016-03-10:
The issue is back. Sometimes there is a restart with exception 28:
```
Exception (28): epc1=0x4022924a epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000008 depc=0x00000000

ctx: cont
sp: 3fff1570 end: 3fff19c0 offset: 01a0

>>>stack>>>

3fff1710:  00000001 00000014 6937a58a 40229204
3fff1720:  0663df8a 0c2d6983 01c350b1 00000004
3fff1730:  00000002 0000000f 3fff1770 3fff971b
3fff1740:  3fff9719 00000000 3fff2354 4022b1b0
3fff1750:  4021b42f 3fff2224 00001104 00000000
3fff1760:  3fff2354 00080000 00000003 4022a179
3fff1770:  00020315 3fff9702 3fffb20c 00001240
3fff1780:  00000002 00000015 00000019 00000004
3fff1790:  3fff2354 3fff9719 00000004 00000000
3fff17a0:  00000000 00080000 3fff2354 4022a782
3fff17b0:  00000001 3fff9719 3fff2354 4022af40
3fff17c0:  3ffe9530 3fff225c 3fff2354 402298f4
3fff17d0:  3fff1810 0000000e 00000010 00001387
3fff17e0:  0026c685 00000000 3fff2354 4022a7de
3fff17f0:  0026c685 3fff2354 3fff1b4c 40203a1a
3fff1800:  3fff0898 3fff07e8 3fff1b4c 40203de6
3fff1810:  00000000 3fff07e8 3fff07e8 402032c6
3fff1820:  c5a79a95 3fff05e6 40204838 3fff05e6
3fff1830:  3fff058c 000001bb 3fff07e8 40203faa
3fff1840:  3ffe93c0 c5a79a95 3fff063e 3fff05e6
3fff1850:  3fff058c 000001bb 3fff07e8 40203245
3fff1860:  3ffe93c0 c5a79a95 3ffe93c0 c5a79a95
3fff1870:  3fff18a0 3fff05e6 40204838 3fff09a0
3fff1880:  3fff03c4 3fff07e8 3fff1a00 40203fc0
3fff1890:  40201124 3fff05e6 3fff1a00 40205a7a
3fff18a0:  2e697061 656c6574 6d617267 67726f2e
3fff18b0:  40201100 00000004 3fff09a0 3fff05e6
3fff18c0:  3ffe8450 3ffe8450 40204838 3fff09a0
3fff18d0:  000001a2 3ffe8d84 3fff058c 3ffe8450
3fff18e0:  3ffe8450 3fff05e6 3fff058c 40205bd8
3fff18f0:  3ffe8dec 3ffe8dec 3ffe8dcc 3ffe8dcc
3fff1900:  3ffe8dec 3ffe8dcc 3ffe8dec 00000000
3fff1910:  00000001 00000001 00000001 00000000
3fff1920:  00000001 00000001 3ffe8d90 3ffe8da0
3fff1930:  3ffe8db0 3ffe8dc0 3ffe8dd0 3ffe8de0
3fff1940:  3ffe8d90 0000225c 00580058 00580058
3fff1950:  00000000 0000002c 40225fdc 3fff03b4
3fff1960:  40201184 01000802 3fff03c4 3ffe8450
3fff1970:  00000000 3fff058c 3fff0588 40202870
3fff1980:  00000000 00ffffff 3ffe93c0 017aa8c0
3fff1990:  00000000 00000000 00000001 3fff0998
3fff19a0:  3fffdad0 00000000 3fff0990 40204884
3fff19b0:  feefeffe feefeffe 3fff09a0 40100958
<<<stack<<<

ets Jan  8 2013,rst cause:2, boot mode:(3,2)

load 0x4010f000, len 1264, room 16

tail 0

chksum 0x42

csum 0x42

~ld
```

[image]

MarcusF2015, 2016-03-10:
[image]

igrr, 2016-03-11:
Is this the same sketch? If not, could you please upload the sketch which may be used to reproduce this?

MarcusF2015, 2016-03-11:
Sorry! I have confused the sketches. I will test the sketch above later and report.

MarcusF2015, 2016-03-11:
Now the data from the sketch above:
```

Exception (28): epc1=0x4022856a epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000008 depc=0x00000000

ctx: cont
sp: 3fff0fe0 end: 3fff1350 offset: 01a0

>>>stack>>>
3fff1180:  00000001 00000014 4eb62639 40228524
3fff1190:  d6ffbf1f 9bebc3b7 c7ded7f1 00000004
3fff11a0:  00000002 0000000f 3fff11e0 3fff1fc3
3fff11b0:  3fff1fc1 00000000 3fff1ef4 4022a4d0
3fff11c0:  40219edb 81c3d2d7 b1352ea9 00000000
3fff11d0:  3fff1ef4 00080000 00000003 40229499
3fff11e0:  00020315 3fff1f02 3fff3ab4 00001240
3fff11f0:  00000002 00000015 00000019 00000004
3fff1200:  3fff1ef4 3fff1fc1 00000004 00000000
3fff1210:  00000000 00080000 3fff1ef4 40229aa2
3fff1220:  00000001 3fff1fc1 3fff1ef4 4022a260
3fff1230:  3ffe933c 3fff1bd4 3fff1ef4 40228c14
3fff1240:  3fff1280 0000000e 00000010 00001387
3fff1250:  0034809b 00000000 3fff1ef4 40229afe
3fff1260:  0034809b 3fff1ef4 3fff1cd4 402035da
3fff1270:  3fff0218 3fff12d0 3fff1cd4 402039a6
3fff1280:  00000000 3fff12d0 3fff12d0 40202e86
3fff1290:  c6a79a95 3fff1390 3fff1390 3fff018c
3fff12a0:  3fff0194 3fff1390 3fff12d0 40203b6a
3fff12b0:  3ffe91b0 c6a79a95 3fff0194 402051bd
3fff12c0:  3fff0194 3fff1390 3fff0194 4020277a
3fff12d0:  3ffe8e80 00000000 00000064 00347f97
3fff12e0:  00000000 3fff1b9c 3fff1cd4 3ffe91b0
3fff12f0:  c6a79a95 00000000 00000000 00000000
3fff1300:  00000000 00000000 00000000 00000000
3fff1310:  00000001 5d7aa8c0 feefeffe feefeffe
3fff1320:  00000000 00000000 00000001 3fff0324
3fff1330:  3fffdad0 00000000 3fff031c 40203d10
3fff1340:  feefeffe feefeffe 3fff0330 40100958
<<<stack<<<

ets Jan  8 2013,rst cause:2, boot mode:(3,3)

load 0x4010f000, len 1264, room 16
tail 0
chksum 0x42
csum 0x42
~ld
```
[image]

closed 2016-04-18 by igrr

supersjimmie, 2016-06-03:
I'm not sure if I should open a new issue, or continue on this closed issue.
It looks like Stable 2.2.0 still has this https problem.
When my code repeats an https call, after a number of calls I receive:
```
Exception (29):
epc1=0x4000e1b2 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000004 depc=0x00000000

ctx: cont 
sp: 3fff0c40 end: 3fff1220 offset: 01a0
```
The stack point to a number of https stuff:
```
0x40222497: more_comps at /Users/igrokhotkov/e/axtls/e1/crypto/bigint.c line 1072
0x40222630: alloc at /Users/igrokhotkov/e/axtls/e1/crypto/bigint.c line 1106
0x402228e2: regular_multiply at /Users/igrokhotkov/e/axtls/e1/crypto/bigint.c line 822
0x4022343a: bi_barrett at /Users/igrokhotkov/e/axtls/e1/crypto/bigint.c line 1287
0x40223695: bi_mod_power at /Users/igrokhotkov/e/axtls/e1/crypto/bigint.c line 1412
0x402224fc: trim at /Users/igrokhotkov/e/axtls/e1/crypto/bigint.c line 1197
0x40224840: RSA_public at /Users/igrokhotkov/e/axtls/e1/crypto/rsa.c line 227
:  (inlined by) RSA_encrypt at /Users/igrokhotkov/e/axtls/e1/crypto/rsa.c line 260
0x4022091e: send_client_key_xchg at /Users/igrokhotkov/e/axtls/e1/ssl/tls1_clnt.c line 342
0x40220d1d: do_clnt_handshake at /Users/igrokhotkov/e/axtls/e1/ssl/tls1_clnt.c line 105
0x4022bd13: pbuf_free at ?? line ?
0x4022bd21: pbuf_free at ?? line ?
0x40202edf: ClientContext::_consume(unsigned int) at C:\...\esp8266\2.2.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 497
:  (inlined by) ClientContext::read(char*, unsigned int) at C:\...\esp8266\2.2.0\libraries\ESP8266WiFi\src\include/ClientContext.h line 169
0x40220754: do_handshake at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 1470
:  (inlined by) basic_read at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 1348
0x40220ab8: do_client_connect at /Users/igrokhotkov/e/axtls/e1/ssl/tls1_clnt.c line 148
0x4021f46c: ssl_new at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 576
0x4022088a: ssl_read at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 262
0x40203364: SSLContext::connect(ClientContext*, unsigned int) at C:\...\esp8266\2.2.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 497
:  (inlined by) WiFiClientSecure::_connectSSL() at C:\...\esp8266\2.2.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 262
0x40202752: WiFiClient::connect(IPAddress, unsigned short) at C:\...\esp8266\2.2.0\libraries\ESP8266WiFi\src/WiFiClient.cpp line 149
0x40204bec: esp_yield at C:\...\esp8266\2.2.0\cores\esp8266/core_esp8266_main.cpp line 43
0x4020363a: WiFiClientSecure::connect(IPAddress, unsigned short) at C:\...\esp8266\2.2.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 246
0x4010020c: _umm_free at C:\...\esp8266\2.2.0\cores\esp8266\umm_malloc/umm_malloc.c line 1285
0x402026d1: WiFiClient::connect(char const*, unsigned short) at C:\...\esp8266\2.2.0\libraries\ESP8266WiFi\src/WiFiClient.cpp line 149
0x40202fcc: WiFiClientSecure::~WiFiClientSecure() at C:\...\esp8266\2.2.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 497
0x40203650: WiFiClientSecure::connect(char const*, unsigned short) at C:\...\esp8266\2.2.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 252
0x40203924: HTTPClient::connect() at C:\...\esp8266\2.2.0\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.cpp line 308
0x40204454: String::~String() at C:\...\esp8266\2.2.0\cores\esp8266/WString.cpp line 720
0x40203da0: HTTPClient::sendRequest(char const*, unsigned char*, unsigned int) at C:\...\esp8266\2.2.0\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.cpp line 308
0x40204f48: Print::write(unsigned char const*, unsigned int) at C:\...\esp8266\2.2.0\cores\esp8266/Print.cpp line 64
0x40204239: Print::write(char const*) at C:\...\esp8266\2.2.0\cores\esp8266/Print.cpp line 64
0x40203e46: HTTPClient::GET() at C:\...\esp8266\2.2.0\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.cpp line 308
0x4020211c: loop at C:\Users\username\Documents\Arduino\octoblu-stream/octoblu-stream.ino line 70
0x40202268: ESP8266WiFiAPClass::softAPdisconnect(bool) at C:\...\esp8266\2.2.0\libraries\ESP8266WiFi\src/ESP8266WiFiAP.cpp line 276
0x40204356: Print::print(unsigned long, int) at C:\...\esp8266\2.2.0\cores\esp8266/Print.cpp line 64
0x40204f48: Print::write(unsigned char const*, unsigned int) at C:\...\esp8266\2.2.0\cores\esp8266/Print.cpp line 64
0x40238a28: sntp_getservername at ?? line ?
0x40204239: Print::write(char const*) at C:\...\esp8266\2.2.0\cores\esp8266/Print.cpp line 64
0x402042b0: Print::println(Printable const&) at C:\...\esp8266\2.2.0\cores\esp8266/Print.cpp line 64
0x40238a28: sntp_getservername at ?? line ?
0x40201fb4: setup at C:\Users\username\Documents\Arduino\octoblu-stream/octoblu-stream.ino line 45
0x40204c38: loop_wrapper at C:\...\esp8266\2.2.0\cores\esp8266/core_esp8266_main.cpp line 43
0x40100718: cont_norm at C:\...\esp8266\2.2.0\cores\esp8266/cont.S line 109

```