esp8266/Arduino #1714: ESP-01 crash (binary dump?) with ESP8266HTTPClient

synfinatic, 2016-03-03:
### Basic Infos

#### Hardware
Hardware:			ESP-01
Core Version:      	2.1.0

### Description

Running this code:
https://github.com/synfinatic/GoProControl/blob/master/src/goprocontrol/goprocontrol.ino

as of 0f63aeb38f1965553d22a28c7241b8ae7284ff88 and I seem to be able to connect to the AP on the GoPro and successfully make the initial HTTP request, but when I try a second HTTP request the ESP-01 seems to crash- not with a coredump, but with a binary data stream?  I've tried enabling debugging but the only indication of the error is this:
```
[HTTP-Client][begin] host: 10.5.5.9 port:80 url: /bacpac/SH? r l��r� �c�n����<CUT_BINARY_OUTPUT>
```
Everything after the ? in the URL is incorrect.  The actual line is _MUCH_ longer- it actually seems to go on forever if I let it.  The binary output is generated with or without debug enabled.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		512k
CPU Frequency:		80Mhz
Flash Mode:			DIO
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		CK

### Sketch

See above

### Debug Messages

```
[SETUP] WAIT 4...
scandone
state: 0 -> 2 (b0)
state: 2 -> 2 (a0)
[SETUP] WAIT 3...
state: 2 -> 0 (2)
reconnect
f 0, wifi evt: 1
STA disconnect: 2
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 

connected with ADTGoProBlack, channel 11
dhcp client start...
wifi evt: 0
ip:10.5.5.109,mask:255.255.255.0,gw:10.5.5.9
wifi evt: 3
[SETUP] WAIT 2...
[SETUP] WAIT 1...
Connecting to ADTGoProBlack
[WIFI][APlistAdd] add SSID: ADTGoProBlack
Connected!
Sending: /bacpac/PW?t=falcon123&p=%01
[HTTP-Client][begin] host: 10.5.5.9 port:80 url: /bacpac/PW?t=falcon123&p=%01 https: 0 httpsFingerprint: 
[HTTP-Client] connect http...
[hostByName] Host: 10.5.5.9 is a IP!
:ref 1
[HTTP-Client] connected to 10.5.5.9:80
:wr
:sent 162
:ww
:rn 219
[HTTP-Client][handleHeaderResponse] RX: 'HTTP/1.1 200 OK'
[HTTP-Client][handleHeaderResponse] RX: 'Allow: GET'
[HTTP-Client][handleHeaderResponse] RX: 'Accept-Ranges: bytes'
[HTTP-Client][handleHeaderResponse] RX: 'Cache-Control: no-cache'
[HTTP-Client][handleHeaderResponse] RX: 'Cache-Control: no-store'
[HTTP-Client][handleHeaderResponse] RX: 'Connection: Keep-Alive'
[HTTP-Client][handleHeaderResponse] RX: 'Server: GoPro Web Server v1.0'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Type: application/octet-stream'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Length: 1'
[HTTP-Client][handleHeaderResponse] RX: ''
[HTTP-Client][handleHeaderResponse] code: 200
[HTTP-Client][handleHeaderResponse] size: 1
[HTTP-Client][end] still data in buffer (1), clean up.
:c0 1, 219
[HTTP-Client][end] tcp stop
:ur 1
:close
:del
OK
Changing modes 3 -> 2
Changing modes 2 -> 0
Starting recording...
Sending: /bacpac/SH?t=falcon123&p=%01
[HTTP-Client][begin] host: 10.5.5.9 port:80 url: /bacpac/SH r l<REMOVE_BINARY>
```

Links2004, 2016-03-03:
looking strange at the first look,
some thinks to test:
- check the power supply
- change baud rate to 78800 and check the data again.
- can you change the line here:
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.h#L30

to:
```cpp
#define DEBUG_HTTPCLIENT(...) DEBUG_ESP_PORT.printf( __VA_ARGS__ ); DEBUG_ESP_PORT.flush()
```
hop we see then a bit more then since the binary stuff starts in the mit of the printf, but this comes form the buffering.

synfinatic, 2016-03-04:
Power supply should be good (off a large 12V battery and +1A rated switching regulator)

Changed to 78800 baud and got this:
```
!�������P��nan�+��-��5� e�j�����=%yn�����nE�� b�j�+n)v�k�I���¹h(~�X��|�Orr��CrɫI��6�t�X��t�dov׬z�-5%�&Hz�n�|�) 
 ets Jan  8 2013,rst cause:1, boot mode:(2,6)

Fatal exception (0): 
epc1=0x40100000, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x40100000, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x40100000, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x40100000, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x40100000, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x40100000, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x40100000, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
```
And the lines repeat forever.

synfinatic, 2016-03-04:
I did some more printf's and flush() in my code and have a better idea where it's crashing:

```
Changing modes 3 -> 2
Changing modes 2 -> 0
Starting recording...
Sending: /bacpac/SH?t=falcon123&p=%01
Trying run()
Trying begin()
[HTTP-Client][begin] host: 10.5.5.9 port:80 url: /bacpac/SH?t=falcon123&p=%01 https: 0 httpsFingerprint: 
Trying GET()
[HTTP-Client] connect http...
[hostByName] Host: 10.5.5.9 is a IP!
 r l��r� �c�n�����p�|��� �x��ǒ�
```

Links2004, 2016-03-04:
the:
```cpp
Fatal exception (0): 
epc1=0x40100000, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
```
looks more like a flash / ram access problem to me.
have you tried different modules?
can you try adding a 47uF cap to VCC?

synfinatic, 2016-03-05:
I switched modules and viola... problem went away!

Thanks for the help!

closed 2016-03-05 by synfinatic