esp8266/Arduino #1718: Webserver returns empty response

krebbi, 2016-03-03:
### Basic Infos

#### Hardware
Hardware: ESP-12E
Core Version:  2.1.0

### Description

this returns an empty response since 2.1.0:

  ESP8266WebServer webServer(80);
  webServer.sendHeader("Cache-Control", "no-cache, no-store, must-revalidate");
  webServer.sendHeader("Pragma", "no-cache");
  webServer.sendHeader("Expires", "-1");
  webServer.send(200, "text/html", "");
  webServer.sendContent(
    "<html><head></head>"
    "<body>"
    "<div><p>My HTML</p></div>"
    "</body></html>"
  );
webServer.client().stop();

but this works:

  ESP8266WebServer webServer(80);
  webServer.sendHeader("Cache-Control", "no-cache, no-store, must-revalidate");
  webServer.sendHeader("Pragma", "no-cache");
  webServer.sendHeader("Expires", "-1");
  webServer.send(200, "text/html", "<html><head></head><body><div><p>My HTML</p></div></body></html>");

### Settings in IDE

Module:  			NodeMCU 1.0
Flash Size: 		4MB/
CPU Frequency:	80Mhz
Flash Mode:		?
Flash Frequency:	?
Upload Using:		SERIAL
Reset Method:		?

### Sketch

see above

igrr, 2016-03-03:
If you intend to send more than once, you need to supply content length or let WebServer know that it is undefined. So before calling `send` the first time, call `server.setContentLength( length );` or `server.setContentSize(CONTENT_LENGTH_UNKNOWN)` if you don't know it upfront. I wouldn't recommend using CONTENT_LENGTH_UNKNOWN because in this case there will be good chance that server has to do close TCP connection actively, which is bad for heap usage. Better calculate the content length and pass it to the server, so that the client knows when the all data is received.

krebbi, 2016-03-03:
which version you are working with?
I get
'class ESP8266WebServer' has no member named 'setContentSize'
when trying your approach

igrr, 2016-03-03:
Sorry, that was a typo — the function is called `setContentLength`: https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WebServer/src/ESP8266WebServer.h#L114

I'm using the latest git version, but the function should be present both in latest stable (2.1.0) and staging (2.1.0-rc2).

krebbi, 2016-03-03:
that works, thx! 

maybe you should update the CaptivePortalAdvanced Example accordingly.

commit 2bc6d6b4 (2016-03-23): Merge pull request #1795 from fape/1718_CaptivePortalAdvanced_fix

CaptivePortalAdvanced fix #1718
commit ed13edf4 (2016-03-23): CaptivePortalAdvanced fix #1718
closed 2016-03-23 by igrr