esp8266/Arduino #1719: WiFi.Begin not broadcasting correct ssid

Mair, 2016-03-03:
#### Hardware
Hardware:		ESP-12
Core Version:      	2.1.0-rc2

### Description
When I use WiFi.begin() I can't find the SSID, however the SSID I do see is of the previous sketch where I used WiFi.softAP().
As an example, Lets say my previous sketch used the following line of code WiFi.softAP("mysoftacess","password"). then I write a new sketch that uses WiFi.begin("myhardap","password").

The ssid broadcasted is "mysoftacess",

moreover I find that my connection - even when using softAP - is really dodgy, only connecting on every 3rd or 4th attempt.
I've been at this for hours so any thoughts would be much appreciated. 

### Settings in IDE

Module:  			ESPino (ESP-12 module)
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			dio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

### Sketch

```cpp
#include <WiFiServer.h>
#include <ESP8266WiFi.h>

void setup() {
	Serial1.begin(115200);
	Serial1.print("Connecting...");
	WiFi.begin("myssid01","myssid01");
	WiFi.mode(WiFiMode::WIFI_AP);
}

void loop() {
	if(WiFi.status() != WL_CONNECTED) {
		Serial1.println("[loop] no wifi");
	}
	else {
		Serial1.println("[loop] WiFi connected");
		Serial1.print("[loop] IP address: ");
		Serial1.println(WiFi.localIP());
	}
	delay(500);
}
```

igrr, 2016-03-03:
If you want to start an AP, you need to use `WiFi.softAP(ssid, pass)`, not `WiFi.begin(ssid, pass)`.

If, on the other hand, you want the ESP to connect to another AP, you need to set `WiFi.mode(WIFI_STA)` and then call `WiFi.begin(ssid, pass)`. The example sketch you posted will never connect because you set the chip to WIFI_AP mode right after you asked it to act in WIFI_STA mode. 

Also please make sure that you SoftAP passwords are 8 characters or longer.

Mair, 2016-03-03:
Thanks for that.
 OK. So, that makes it slightly more clearer. Am I correct in saying that that 

1. we would always use `WiFi.softAP` to configure a new access point and always use WiFi.begin to logon to an existing wifi?
2. Does `WiFi.begin` implicitly sets the mode to `WIFI_STA`?
3. Does `WiFi.softAP` implicitly set the mode to `WIFI_AP`?
3. Can you use both `WiFi.begin` and `WiFi.softAP` at the same time by setting `WIFI_AP_STA`
Do you have any samples of using `WiFi.softAP` The attempt I have seems very unreliable and I would love to know if there is a better way.

Thanks again

gmag11, 2016-03-04:
I answer to each point: 

1. Correct. Remember that you may have both (Wi-Fi station and AP) at once, if you want it. 
2. I guess no. I would recommend to set working mode in your scketch. ESP8266 chip stores old Wi-Fi configuration on eeprom, so if you are reusing a board from other project you can get undesirable behavior. 
3. Same as 2
4. Yes but there is a problem with that. To connect to an existing Wi-Fi esp scans all channels until it finds the correct one. As radio module is shared between two modes it may happen that while esp is scanning channels you are unable to connect it as AP. 

Regards

Mair, 2016-03-05:
Fantastic. Thank you for the clarification

closed 2016-03-05 by Mair

Sovatna, 2018-08-02:
Dear Sir, 
Is there any way to solve the problem number 4? I am a student of agriculture and I want to control my farm using esp8266. The idea is that using esp(station) to send data from the inside field to the esp(access point) outside the field. Then the data will continuously send to website. I am doing fine of the connection between Station <--> Access Point && Access Point <--> Server separately. What I want is that Station <--> Access Point <--> Server

Best regards,

Sovatna, 2018-08-02:
Dear Sir, 

Is there anyway to solve problem No. 4?