esp8266/Arduino #1720: esp gets lost when using static IP configuration

vgam7, 2016-03-04:
#### Hardware
Hardware:			ESP-12E, ESP-01
Core Version:      	ALL, including last one

### Description

When configuring esp device with static IP using
WiFi.config(IPAddress(192,168,1,205), IPAddress(192,168,1,1), IPAddress(255,255,255,0));
after about 8-10 hours the device "gets lost in network", stopping to answering with it's IP address to newly connected devices which sends broadcast asking it's IP address.

### Sketch

#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <ArduinoJson.h>

#define ON 0
#define OFF 1

const char* ssid = "...";
const char* password = "...";

const int MAX_ON_TIME = 15*60*1000;

ESP8266WebServer _server(80);

int _onMillisec;
bool _state = false;
EspClass _esp;
int _initialFmem = _esp.getFreeHeap();
String _mac = WiFi.macAddress();
String _sdk = _esp.getSdkVersion();

const int _loadPin = 2;

void handleRoot() {
  _server.send(200, "text/plain", "Valid requests are: GET /power, PUT /power, GET /debug");
}

void handleNotFound(){
  _server.send(404);
}

void handleLoadStateChangeReq()
{
  String state = _server.arg("state");
  
  Serial.println("handleLoadStateChangeReq(" + state + ")");
  
  _state = (state == "on") ? ON : OFF;
  digitalWrite(_loadPin, _state);
  if (_state == ON)
    _onMillisec = millis();

  sendResponse(String("Load is ") + (_state == ON ? "ON" : "OFF"));
}

void handleDebugReq()
{
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  int fmem = _esp.getFreeHeap();

  StaticJsonBuffer<150> jsonBuffer;
  JsonObject& root = jsonBuffer.createObject();
  root["initFreeMem"] = _initialFmem;
  root["curFreeMem"] = fmem;
  root["uptime"] = millis() / 1000;
  root["mac"] = _mac;
  root["sdk_ver"] = _sdk;

  String js;
  root.printTo(js);

  sendResponse(js);
}

void sendResponse(String resp)
{
  _server.send(200, "text/plain", resp);
}

void connectToWifi()
{
  WiFi.begin(ssid, password);
  Serial.println("");
  WiFi.config(IPAddress(192,168,1,205), IPAddress(192,168,1,1), IPAddress(255,255,255,0));

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
}

void setup(void){
  pinMode(_loadPin, OUTPUT);
  _state = OFF;
  digitalWrite(_loadPin, _state);
  Serial.begin(115200);

  connectToWifi();

  _server.on("/", handleRoot);
  _server.on("/power", [](){ sendResponse(_state == ON ? "on" : "off"); });
  _server.on("/setpower", handleLoadStateChangeReq);
  _server.on("/debug", handleDebugReq);

  _server.onNotFound(handleNotFound);

  _server.begin();
  Serial.println("Server started");
}

void loop(void){
  _server.handleClient();
  if (_state == ON && millis() - _onMillisec >= MAX_ON_TIME)
  {
    _state = OFF;
    digitalWrite(_loadPin, _state);
  }
}

igrr, 2016-03-09:
I tried to reproduce this with the above sketch, changing only IP configuration and WiFi credentials. After 24 hours, device is still visible. To simulate a "new device connecting", I have cleared ARP cache on OS X with `sudo arp -a -d`. I will monitor the device for another day, but so far I'm not able to reproduce this.

vgam7, 2016-03-09:
Hi Ivan,
did you constantly ping the device or left it "inactive"? 

BTW, I found this issue was discussed on espressif forum almost 1 year ago and it still not resolved... 
[http://bbs.espressif.com/viewtopic.php?f=7&t=333&sid=af60f699403d2fba34d98a9b13c87199](url)

devyte, 2017-10-18:
I have 40+ 12F-based devices up and running for 24h at a time, and I don't see this issue. Closing.

closed 2017-10-18 by devyte

VoLinhTruc, 2018-08-13:
I also meet this issue about a year
I set static IP (in station mode) is 192.168.1.69
Upload sketch to board, us "Advance IP Scanner" to scan device which access to my home-Access Point
ESP's IP is correct (192.168.1.69)
But after a few minutes, it change to 192.168.1.2
I hope espressif's developer will fix it