esp8266/Arduino #1722: hanging ESP.restart()

Pfannex, 2016-03-05:
Hi,

what´s the difference between 

ESP.reset() and ESP.restart()?

Both work, but sometimes the ESP does not starts again.

working restart:
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
@cp:0 ld    <------ this means it work
```

hanging restart:
```
 ets Jan  8 2013,rst cause:2, boot mode:(1,7)
 ets Jan  8 2013,rst cause:2, boot mode:(1,7)

wdt reset    <----- this means it does not work
```

Links2004, 2016-03-05:
```ESP.reset()``` is a hard reset and can leave some of the registers in the old state which can lead to problems, its more or less like the reset button on the PC.

```ESP.restart()``` tells the SDK to reboot, so its a more clean reboot, use this one if possible.

the ```boot mode:(1,7)``` problem is known and only happens at the first restart after serial flashing.
if you do one manual reboot by power or RST pin all will work more info see: #1017

Pfannex, 2016-03-06:
OK, thanks for your explanation......

I was reading the ESP.restart() issues. 
But is there a final solution or is the, manually reset after serial flashing, the latest work around?

Links2004, 2016-03-06:
yes, the manually reset after serial flashing is the latest work around.
the problem is some here in the hard coded Silicium part of the ESP8266.
there may is a register / ram address which store the mode to boot in, but no one has finde it yes.
Its dam hard to search for.

Pfannex, 2016-03-06:
OK, thanks....

I think, I can live with it...... :-)

closed 2016-03-06 by Pfannex

PhilColbert, 2016-04-18:
Would it be possible for the ESP gurus to write a README guide to the ESP, just a list of known issues that a newbie could read, I just spent a few hours trying to work out why system restart seemingly randomly works...  and this is why, would be great to have a list of known strange issues someone just starting with the ESP could read and save a lot of headaches ! :)

Thanks

jkandasa, 2016-09-14:
I faced this issue. I added a `delay` before `ESP.restart()`. No more failure rate..
```
void reboot(){
  delay(5000); // wait few seconds for serial flashing... 
  ESP.restart();
}
```

aidanruff, 2017-03-24:
I've had the same problem, even with a pull up on GPOIO#0, it didn't work.

Then I reasoned that as I was using GPIO as an output, maybe the hardware wasn't reseting it to an input  on a soft reset, so the reset would depend upon the state of GPIO#0. 

So, I set GPIO#0 high before doing the restart and it works!

gochisar, 2017-08-11:
i have a solution;
`WiFi.forceSleepBegin();
  wdt_reset();
  ESP.restart();
  while(1)wdt_reset();`

close WiFi; forceSleepBegin();
and reset wdt contuniously.

karenYerevan, 2018-01-02:
hi there 
i have ESP8266-12, it is have other power
I flashing esp8266 from arduino uno 
Flashing programm (XTCOM_UTIL)
aruduino uno -- esp8266
RX -> RXD
TX -> TXD
GND -> GND,GPIO0

After flashing all AT commands worck good.
But when esp8266 is restart (AT+RST) , AT commands stopping works
and show this text 

> AT
> OK
> 
> AT+RST
> OK
> 
>  ets Jan  8 2013,rst cause:4, boot mode:(1,7)
> wdt reset

[image]

who can help me ???

devyte, 2018-01-02:
@karenYerevan this repo has nothing to do with the AT command set. I suggest asking at a community forum like esp8266.com.

jkandasa, 2018-01-02:
@karenYerevan AFAIK, after flash via serial, you need to do a hard/manual reboot at the very first time(power OFF and ON, ESP8266).

See in this [comment](https://github.com/esp8266/Arduino/issues/1722#issuecomment-192829825)

karenYerevan, 2018-01-02:
yes i done that many many time, but that dosnt worck

devyte, 2018-05-06:
GPIO pins will be re-inited before setup()

garyrylander, 2018-12-26:
You have to have a jumper on D0 and Rst. That worked for me

verygoodguy, 2023-08-01:
Does reset cut full power?