esp8266/Arduino #1723: SD card with SPIFFS

CosmicBoris, 2016-03-05:
Hardware:		ESP-07
Core Version:      	2.1.0-rc2

### Description
Can I use both SD card lib and SPIFFS?

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		1MB/512kb
CPU Frequency:      160Mhz
Flash Mode:	        DIO
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

#include <FS.h>
#include <SPI.h>
#include <SD.h>

File hFile;

In file included from ESP_FileServer.ino:7:0:
C:\Users\boris\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\libraries\SD\src/SD.h:26:7: error: redefinition of 'class fs::File'
 class File : public Stream {
       ^
In file included from ESP_FileServer.ino:5:0:
C:\Users\boris\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266/FS.h:48:7: error: previous definition of 'class fs::File'
 class File : public Stream
       ^
Multiple libraries were found for "SD.h"
 Used: C:\Users\boris\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\libraries\SD
 Not used: C:\Program Files (x86)\Arduino\libraries\SD

gmag11, 2016-03-06:
I'm not an expert in C++, but SPIFFS library (FS.h) defines a namespace called "fs". It should be possible to differenciate class definitions using it.

CosmicBoris, 2016-03-07:
I found this useful [https://github.com/esp8266/Arduino/issues/1524](url)
But now I have to specify what namespace am using:
```
void readFromSD(String path){
    using sd::File;
    File hFile = SD.open(path);
    ...
}
void readFromFS(String path){
    using fs::File;
    File hFile =  SPIFFS.open(path, "r");
    ...
}
```

igrr, 2016-03-09:
Currently this isn't possible, because both SD and SPIFFS provide a class called `File`.
There is, however, a suggestion to put SD library into its own namespace (#1524).

closed 2016-03-11 by CosmicBoris

WheresWally, 2017-07-03:
Any progress on being able to use SD and SPIFFS at the same time?