esp8266/Arduino #1727: Cannot compile example of http://github.com/boseji/BASE64 library

robertgregor, 2016-03-05:
Hello, I have something strange after upgrade to 2.1.0
In my project, I am using Base64 library.
Now with 2.1.0 I cannot compile my sketch:

SmokeSensor:813: error: expected primary-expression before '.' token

        client.println(base64.encode("aaaa"));

                             ^

exit status 1
expected primary-expression before '.' token
If I change the board to Arduino UNO, no problem to compile.

It can be easily reproduced by installing the http://github.com/boseji/BASE64 library and try to compile the example. With the esp8266 board I've got the error:

C:\Users\gregorro\Disk Google\RH\RemoteHomeArduinoSketches\WiFi_ESP\libraries\BASE64\examples\base64-demo\base64-demo.ino: In function 'void loop()':

base64-demo:26: error: expected primary-expression before '.' token

   Serial.println(base64.encode("Hello There, I am doing Good."));

                        ^

base64-demo:27: error: expected primary-expression before '.' token

   Serial.println(base64.decode("SGVsbG8gVGhlcmUsIEkgYW0gZG9pbmcgR29vZC4="));

                        ^

exit status 1
expected primary-expression before '.' token

With Arduino UNO, everything is fine...

Links2004, 2016-03-06:
the core has a base64 lib include, with the name base64 too this will make your problem.
the used lib is libb64.
https://github.com/esp8266/Arduino/tree/master/cores/esp8266/libb64

but the Arduino API is not fully implemented yet. (only encode is there)
https://github.com/esp8266/Arduino/blob/master/cores/esp8266/base64.h

robertgregor, 2016-03-06:
OK< I have used the core libb64/cdecode and it works fine...

closed 2016-03-06 by robertgregor

Humancell, 2017-03-04:
Does someone have a complete ESP example they can post?  I'm missing something, but can not get this to work at all ...

What do I include, if it is in the core?

vicnevicne, 2017-03-04:
Hi, humancell, 
Please refrain from cross-posting. You seem to have an issue in #1758 but asking the same question in #507 and "a complete ESP example" here will not motivate developers to help you.
This issue tracking system is meant for problems and bugs with the Arduino environment. If you really need help of a first ESP example, please head to the esp8266.com forums where many people will be able to help you.
KR,

Humancell, 2017-03-06:
Sorry ... I had found several people asking the same question as this post, and although this issue was closed there was no answer given for the original question ... instead the poster was pointed at a different library.

I'll refrain from cross posting in the future, but was not clear if the closed issues generate the same notifications.  In addition, I'll post an answer to the original posters question, as I realized my error was the same as his.

To use the Base64 library included, if you don't want to use libb64, you can do the following:

```
#include "base64.h"

// base64-encoding of Aladdin:OpenSesame, or QWxhZGRpbjpPcGVuU2VzYW1l
String basicAuth = "Aladdin:OpenSesame";
Serial.print("Base64: ");
Serial.println(base64::encode(basicAuth));
Serial.print("Match:  ");
Serial.println("QWxhZGRpbjpPcGVuU2VzYW1l");

```
The issue that both the original poster and myself did was to use the '.' instead of '::' when referencing the class method.  The error message was just throwing me off.  Hopefully this will assist someone in the future.

Sorry for the extra noise ...

vicnevicne, 2017-03-06:
Nice catch. Thanks for posting the answer.
KR,