esp8266/Arduino #1729: Sleep Mode Interrupted by Sensor

cadywompus, 2016-03-06:
Hi,

I've been trying to crack this design for while, but I haven't had any luck. What I'm trying to implement in my project is sleep mode that is interrupted when my sensor is sending data. From what I've gathered, sleep has to be initiated by setting an amount of time for the device to go into sleep mode and GPIO16 and RST have to be tied together.

I'm using a flow sensor that runs with a hall sensor and sends a pulsed signal to our ESP8266. Is it possible to take this and somehow trigger the RST to wake from sleep on a falling edge? Maybe putting a NMOS transistor with the gate connected to GPIO16 (since it's pulled HIGH in sleep), source to the RST and drain to the signal. I'd appreciate any suggestions I can get.

I am using the ESP8266 Huzzah from adafruit. Link below:
[ESP8266 Huzzah breakout](https://learn.adafruit.com/adafruit-huzzah-esp8266-breakout/overview)

The sensor: [flow sensor](https://www.adafruit.com/products/828)

**Settings in Arduino IDE**
Module:  			Adafruit HUZZAH ESP8266
Flash Size: 		4MB/ 3MSPIFFS
CPU Frequency:      80Mhz
Upload Speed:        115200
Upload Using:		Serial

Programmer: AVRISP mkII

chaeplin, 2016-03-06:
Use deepSleep(0) ---> sleep forever.
Connect Sensor output to RST. You don't need to connect GPIO 16 and RST.

cadywompus, 2016-03-07:
Awesome! I knew there was a way to do it, but I couldn't think of it. Thank you!

closed 2016-03-16 by igrr

aderbas, 2018-07-18:
But how it will wake up by the sensor if only the RTC is activated.

GuenbriMed, 2021-12-24:
The problem is that sensor output could be on high or low level.
Its a random status.
How can we manage this ?