esp8266/Arduino #1730: SoftwareSerial incorrect baud rate for TX

Elektrik1, 2016-03-07:
Hey,

I got an strange issue when trying to implement SoftwareSerial library. I noticed it using 4800 baud rate where it was reading incorrect chars (and reads more chars - 15, while 9 should be read). I tried to describe actual 4800 baud as 4700 - and it reads fine. So I decided to do some investigation:
1) I was using pins 5 and 4 - changed ones to 14 and 12 (from swsertest example) - doesn't matter, it is not PIN issue
2) Tried 115200 - it works fine.
3) 9600 baud value reads wrong value (basically all baud rates lower than 115200)
4) Attaching screenshots from FT232 connected Realterm + Saleae logic analyzer hardwired in between FT232<->ESP8266.
5) It also seems that pressing a restart several times (it prints out message on start) sometimes prints correct message (all ascii chars), so definitely timing issue of several ms, because sometimes it works.

Would be nice to have it fixed (I am planning to use baud rates as low as 240bps).

*Note SoftwareSerial library:
m_bitTime = ESP.getCpuFreqMHz()*1000000/speed; <- not sure if this is how we calculate baudrate

Test done using amica nodemcu devkit (ESP12-E) 80MHz cpu speed.

[sketch.txt](https://github.com/esp8266/Arduino/files/160638/sketch.txt)
[image]
[image]

igrr, 2016-03-09:
/cc @plerup

plerup, 2016-03-09:
Which version of SoftwareSerial are you using?

Elektrik1, 2016-03-09:
Using library SoftwareSerial at version 1.0 in folder: C:\Users\<user>\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\libraries\SoftwareSerial

ESP Board installed - 2.1.0 stable

Elektrik1, 2016-03-09:
I have found sources a bit updated on official repo (https://github.com/plerup/espsoftwareserial), tested new library - seems to be fine now, even at 240bps - data sent is correct, so perhaps may be checked as 'fixed' - latest library works fine, sorry for rushing anyone.

closed 2016-03-10 by Elektrik1