esp8266/Arduino #1736: How do BasicOTA for esp-12e

saytinh, 2016-03-08:
I tried many times the BasicOTA.ino for espj-12e with Arduino IDE 1.6.7 and 2.0.0 core version as official instructions (https://github.com/esp8266/Arduino/blob/master/doc/ota_updates/ota_updates.md) but not successful (it does not appears the esp8266-ota port in Arduino IDE)

Anyone already done with the esp-12e. Please advice me. Thank so much.

### Settings in IDE (I did tried almost  possible setting but still not appears)

Module:  			Generic ESP8266 Module
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

krzychb, 2016-03-08:
Hi saytinh,

Let's focus on troubleshoot this issue on Windows. 

1. What is displayed on serial monitor after upload using serial port is complete?
2. Is your PC on the same network as your ESP module?
3. Do you see your ESP advertised as described in [this post](http://www.esp8266.com/viewtopic.php?p=41869#p41869)?
4. If you do not see it then check your firewall settings. Maybe you did not allow access through firewall after doing Arduino IDE installation (see prompt below).

Krzysztof

[image]

saytinh, 2016-03-10:
Hi Krzysztof,

I follow your suggest on Windows but still not appears ota port on Arduino IDE. 

1. What is displayed on serial monitor after upload using serial port is complete? => Booting, Ready and 192.168.100.110 (I can ping to that IP)

2. Is your PC on the same network as your ESP module? => I'm sure all is the same network (my computer IP: 192.168.100.150)

3. Do you see your ESP advertised as described in this post? => The esp8266 is esp-12e. I checked by flash id showed that device 4016 (it means 4MB)
[image]

4. Firewall? => I already tried turn-off windows firewall. 

My computer is Windows 7 32-bit and using Arduino IDE 1.6.7 and 2.0.0 core version.

Do you have any more suggest? Thansk

Regards,

krzychb, 2016-03-10:
Hi saytinh,

This looks like a tougher cookie :smile: 
I just saw another [similar post](http://www.esp8266.com/viewtopic.php?p=42767#p42767) on ESP8266 forum and really curious what the root case it.

Regarding item 3 - please verify if you can see your ESP using a network sniffer like [WhoIsConnectedSniffer](http://www.nirsoft.net/utils/who_is_connected_sniffer.html) and what protocols are detected..

[image]

If you see similar picture (under Detected Protocols - wait couple of minutes to have complete update) and still do not see your ESP among network ports I propose that you reinstall Arduino IDE and try again.

I have been using OTA primary on 64bit version of Windows 7. I will check it on my other PC with 32bit Windows 7 later today afternoon and let you know.

Krzysztof

Testato, 2016-03-31:
same here
Win7-64
Core 2.1.0
yesterday i used a 4M esp12, at the beginning the Network port is not present, after many times closing IDE, changinbg serial port number, etc the Network port appear and the OTA worked.

Today the port do not appear.
Whit whoisconnected sniffer i have the mDNS

krzychb, 2016-03-31:
Hi @Testato
Whar is your Arduino IDE version?
Krzysztof

Testato, 2016-03-31:
1.6.8 in official Portable configuration

krzychb, 2016-03-31:
@Testato,

I have just checked BasicOTA.ino with:

- Core 2.1.0
- Win7-64
- Arduino IDE portable 1.6.8

I does work for me and the ota port shows up.

Have you tried to temporarily disable Windows Firewall?
I my configuration it is enabled all the time and I did not have to do anything with it except [Allow access](https://github.com/esp8266/Arduino/issues/1736#issuecomment-193950843) on first run of Arduino IDE.

Krzysztof

Testato, 2016-03-31:
windows firewall is disabled here

krzychb, 2016-03-31:
@Testato,

I assume that you have confirmed that your ESP on your Wi-Fi and PC is able to preform basic network functions like serving web pages using example sketch like HelloSever.ino.

If this is the case then I would bring out the bigger guns.

I would be looking for root cause in two areas - windows installation and network, step by step as follows:

1. Check if you are able to do [OTA using a Web Browser](https://github.com/esp8266/Arduino/blob/master/doc/ota_updates/readme.md#web-browser)
2. Try another PC - Widows or Linux (I am not proposing Mac because I have not tested it)
3. If you have another Wi-Fi router handy, set up a separate Wi-Fi network and hook up PC and ESP to this network.

Krzysztof

grzegorz189, 2016-03-31:
https://github.com/esp8266/Arduino/blob/master/doc/installing.md

saytinh, 2016-04-01:
I change to other wifi router and it's now successfully on both of Windows and Linux. You could need a few (one minutes) in order to appear wifi port. 

My problem is router. It solved so I close this issue. Thanks all of you for your suggests.

closed 2016-04-01 by saytinh

Testato, 2016-04-01:
i formatted my win7 and installed Kubuntu.
At the beginning i have same random rpoblem, after many test i found that "_" underscore character is not admitted in the hostname. CHanging _ whit - all module start to work.

I do no know if on Win7 the rpoblem si solved, i do not know if i want return to Windows :-)

krzychb, 2016-04-01:
@Testato,
Congratulations on determination (up to wiping out Windows) to get this issue diagnosed :+1:
I think that now Windows would work as well :smile:
Krzysztof

gmag11, 2016-04-03:
I had problems when first tried to implement or a on my code. I noticed that you must wait for WiFi connection before you do ArduinoOTA.begin(). Doing so, it takes just some seconds Arduino IDE to detect it.

saytinh, 2016-04-03:
Sorry I don't really understand what you mean. The waiting time I mean is after uploading BasicOTA.ino sketch and I have to waiting a moment (about 1 minute) for appearance wifi port.

gmag11, 2016-04-21:
If `ArduinoOTA.begin()` happens when WiFi has not yet connected, mDNS anouncement does not reach network. It is only my guess but, after adding 5 seconds wait time after `WiFi.begin()` OTA port appeared on Arduino.