esp8266/Arduino #1737: httpUpdate - failed to upload - Sketch not flashed

Pfannex, 2016-03-08:
### Basic Infos

#### Hardware
Hardware:			ESP-12e
Core Version:      	2.1.0-rc2

### Description

In the root of my WEB-Server I got a small *.bin (x.bin),
now I try to flash the ESP with this Sketch......
but all attempts failed.

Did anyone one this problem or where I´m wrong?

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB/1MB?
CPU Frequency:		80Mhz
Flash Mode:			DIO
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

### Sketch

```cpp
#include <ESP8266WiFi.h>
//#include <ESP8266WiFiMulti.h>

//#include <ESP8266HTTPClient.h>
#include <ESP8266httpUpdate.h>

void setup() {
  Serial.begin(115200);
  Serial.println("");
  Serial.println("");

  WiFi.begin("MySSID", "MYPass");

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("connected");

  Serial.println("");
  Serial.println("start flashing......");
  t_httpUpdate_return ret = ESPhttpUpdate.update("192.168.1.3", 80, "/x.bin");  

  switch(ret) {
    case HTTP_UPDATE_FAILED:
        Serial.println("[update] Update failed.");
        break;
    case HTTP_UPDATE_NO_UPDATES:
        Serial.println("[update] Update no Update.");
        break;
    case HTTP_UPDATE_OK:
        Serial.println("[update] Update ok."); // may not called we reboot the ESP
        break;
  }  
}

void loop() {
}
```

### Debug Messages

```
...........
connected
start flashing......
 ets Jan  8 2013,rst cause:4, boot mode:(1,6)
wdt reset
rl<\0>l??|<\0>?l?|<2><\f><\f><\f>?<\f>l?<\f>b|??<2>?<18>?r?b?<\f>b??nn?lnn???<\f>b<28>p??lrlrlp?n?<16><2><\f><\f>?<\f>l<\f>??<\f><\f><\f>b<\f>n?|<2>l????b??nn?<\0>l??l`<2>?
```
or

```
connected
start flashing......
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)
wdt reset
load 0x4010f000, len 1264, room 16
tail 0
chksum 0x42
csum 0x42
~Fatal exception (0): 
epc1=0x400ef278, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0):
```

Links2004, 2016-03-08:
please enable debug output in the IDE, and post the result.

Pfannex, 2016-03-08:
```
[httpUpdate] Header read fin.<\n>
[httpUpdate] Server header:<\n>
[httpUpdate]  - code: 200<\n>
[httpUpdate]  - len: 234224<\n>
[httpUpdate] ESP8266 info:<\n>
[httpUpdate]  - free Space: 2854912<\n>
[httpUpdate]  - current Sketch Size: 287460<\n>
[httpUpdate] runUpdate flash...<\n>

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)<\r>
wdt reset<\r><\n>
```

Pfannex, 2016-03-08:
```
............<\r><\n>
connected<\r><\n>
<\r><\n>
start flashing......<\r><\n>
[begin] roundedSize:       0x0003A000 (237568)<\n>
[begin] updateEndAddress:  0x00300000 (3145728)<\n>
[begin] currentSketchSize: 0x00046000 (286720)<\n>
[begin] _startAddress:     0x002C6000 (2908160)<\n>
[begin] _currentAddress:   0x002C6000 (2908160)<\n>
[begin] _size:             0x000392F0 (234224)<\n>
<\r><\n>
Soft WDT reset<\r><\n>
<\r><\n>
ctx: cont <\r><\n>
sp: 3fff14e0 end: 3fff18f0 offset: 01b0<\r><\n>
<\r><\n>
>>>stack>>><\r><\n>
3fff1690:  00000000 4000444e 00000000 3fff2630  <\r><\n>
3fff16a0:  00000000 4000420c 60000200 3fff27c0  <\r><\n>
3fff16b0:  00000100 40004ae8 00000100 002c6000  <\r><\n>
3fff16c0:  00000100 3fff26a0 3fffc718 002c6100  <\r><\n>
3fff16d0:  00001000 0000000f 40105b64 00001000  <\r><\n>
3fff16e0:  3fff26a0 3fff26a0 3fff0824 00000000  <\r><\n>
3fff16f0:  3fffc718 3fff26a0 40205a58 3fff08d0  <\r><\n>
3fff1700:  002c6000 4020485e 002c6000 00001000  <\r><\n>
3fff1710:  3fff07d4 00000001 3fff0824 4020563a  <\r><\n>
3fff1720:  00001000 00000000 3fff22a0 40205896  <\r><\n>
3fff1730:  00001000 000392f0 00300000 00046000  <\r><\n>
3fff1740:  3ffe9d78 00000000 3fff17b0 00000001  <\r><\n>
3fff1750:  3fff0824 000392f0 3fff06f4 4020376b  <\r><\n>
3fff1760:  3ffe9358 00000000 000003e8 40206f2c  <\r><\n>
3fff1770:  3fff2688 0000000f 00000000 4020402c  <\r><\n>
3fff1780:  3fff22a0 3fff0824 3fff17b0 3fff08d0  <\r><\n>
3fff1790:  40204694 001640e0 3fff22a0 3fff07d4  <\r><\n>
3fff17a0:  3fff22a0 3fff06f4 3fff1810 40203a83  <\r><\n>
3fff17b0:  3fff2670 0000000f 00000000 400201e9  <\r><\n>
3fff17c0:  00000000 00000000 3ffe8f94 40206e06  <\r><\n>
3fff17d0:  000392f0 00000000 00000000 00000001  <\r><\n>
3fff17e0:  00400000 454e656e 616c7754 43415f6e  <\r><\n>
3fff17f0:  53534543 efeffe00 3fff1810 00000000  <\r><\n>
3fff1800:  3fff06f4 3ffe9d78 00000001 40203b7d  <\r><\n>
3fff1810:  3fff22a0 00000000 3fff19d8 0000000f  <\r><\n>
3fff1820:  0000000b 50000050 6e011f40 3fff1f48  <\r><\n>
3fff1830:  0000000f 00000006 3ffe9d00 3fff21e8  <\r><\n>
3fff1840:  0000000f 00000000 3fff2480 000000ef  <\r><\n>
3fff1850:  000000e7 3fff2248 0000001f 00000013  <\r><\n>
3fff1860:  3fff2230 0000000f 00000000 3fff2204  <\r><\n>
3fff1870:  00000001 000000c8 000392f0 40205100  <\r><\n>
3fff1880:  00000000 3ffe9d78 3fff07fc 40205115  <\r><\n>
3fff1890:  3ffe8c40 00000050 3ffe8c4c 40206418  <\r><\n>
3fff18a0:  3fff1a18 4020257c 3fff07fc 3fff08bc  <\r><\n>
3fff18b0:  3fff0660 3ffe9d78 3fff07fc 40202228  <\r><\n>
3fff18c0:  3ffe9d78 00000001 feefeffe feefeffe  <\r><\n>
3fff18d0:  3fffdad0 00000000 3fff08b5 40205a9b  <\r><\n>
3fff18e0:  feefeffe feefeffe 3fff08d0 40100114  <\r><\n>
<<<stack<<<<\r><\n>
<\r><\n>
 ets Jan  8 2013,rst cause:2, boot mode:(1,7)<\r><\n>
<\r><\n>
<\r><\n>
 ets Jan  8 2013,rst cause:4, boot mode:(1,7)<\r><\n>
<\r><\n>
wdt reset<\r><\n>

```

Pfannex, 2016-03-08:
```
.......<\r><\n>
connected<\r><\n>
<\r><\n>
start flashing......<\r><\n>
[httpUpdate] Header read fin.<\n>
[httpUpdate] Server header:<\n>
[httpUpdate]  - code: 200<\n>
[httpUpdate]  - len: 234224<\n>
[httpUpdate] ESP8266 info:<\n>
[httpUpdate]  - free Space: 2854912<\n>
[httpUpdate]  - current Sketch Size: 288316<\n>
[httpUpdate] runUpdate flash...<\n>
[begin] roundedSize:       0x0003A000 (237568)<\n>
[begin] updateEndAddress:  0x00300000 (3145728)<\n>
[begin] currentSketchSize: 0x00047000 (290816)<\n>
[begin] _startAddress:     0x002C6000 (2908160)<\n>
[begin] _currentAddress:   0x002C6000 (2908160)<\n>
[begin] _size:             0x000392F0 (234224)<\n>
<\r><\n>
 ets Jan  8 2013,rst cause:4, boot mode:(1,7)<\r><\n>
<\r><\n>
´´´

Pfannex, 2016-03-09:
Did anyone use the http update on demand
or a idea why the update failed?

My http server shoud run normal
or is there someting special to know?

Should my code work?

Pfannex, 2016-03-09:
---

Pfannex, 2016-03-11:
@Links2004 
can you share a httpUpdate and advanced httpUpdate with php code?
Maybe with php update script.

Links2004, 2016-03-11:
the  ```ets Jan  8 2013,rst cause:4, boot mode:(1,7)``` looks like the update is done.
but you run in the known reboot problem see: #1722 or #1017 for some more details.

Pfannex, 2016-03-11:
hmm... but if flashing was successful, a manual reset should boot the new sketch.
But the old one is still flashed!

So it does not look like a reset problem.....

Links2004, 2016-03-13:
yes a reset (toggle reset pin) but no power off, since the commands for the bootloader (to copy the new over the old sketch) are stored in the rtc memory.

more how the update work internally:
https://github.com/esp8266/Arduino/blob/master/doc/ota_updates/readme.md#updater-class

devyte, 2017-10-18:
Closing as duplicate of physical reset issue.

closed 2017-10-18 by devyte