esp8266/Arduino #1739: Memory access violations when using Malloc after HTTPS handshake

madsci1016, 2016-03-09:
----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			ESP-12
Core Version:      	2.1.0

### Description

Using Malloc after HTTPS handshake yields either WDT resets or exception 28 errors.  I've baked the code down so you can test/verify with no additional hardware. sketch writes 11KB of 'A' to memory after handshake, and always crashes. 

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB
CPU Frequency:		160Mhz
Flash Mode:			dio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck 

### Sketch

#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>
//#include <SPI.h>

#define SDAp 0
#define SCLp 2
#define SENSORPWR 5
#define TI1WIRE 4
#define MISO 12
#define MOSI 13
#define SPICLK 14
#define SPICS 15
#define BATTV A0

const char* ssid     = "***";
const char* password = "***";

const char* host = "google.com";

//pointer to location to save audio samples  
byte *samples;
int reading;

#define SENSORS_ON digitalWrite(SENSORPWR, LOW)
#define SENSORS_OFF digitalWrite(SENSORPWR, HIGH)

//routine called at constant rate to sample and store audio
void sample(){
  digitalWrite(SPICS, LOW);
 // SPI.transfer(0b00011010); //send config with some pad bytes to get bit allignment right
  samples[reading++] = 'A';// SPI.transfer(0); //read high byte back
  digitalWrite(SPICS, HIGH);
  
  //see if we are done sampling
  if(reading>11025) timer1_disable();
  
}

void setup() {
  // put your setup code here, to run once:

    //Serial port for Debug
  Serial.begin(115200);

    //enable sensors
  pinMode(SENSORPWR,OUTPUT);
  SENSORS_ON;
  delay(200);

    //start WiFi nad connect
  WiFi.begin(ssid, password);

    //start SPI to external ADC
  //pinMode(SPICS, OUTPUT);
  //SPI.begin();

}

void loop() {
  // put your main code here, to run repeatedly:

    //WiFiClient client;
  WiFiClientSecure client;
  const int httpPort = 443;
  if (!client.connect(host, httpPort)) {
    Serial.println("Server connection failed, sleeping");
    ESP.deepSleep(285000000, WAKE_RF_DEFAULT);
  }
   
  Serial.print("Connected to ");
  Serial.println(host);
  
  //clear out certificate from memory
  client.read();

  Serial.println("Free Mem after read " + (String)ESP.getFreeHeap() + " free");
  
  //verify we have enough memory to make sound recording
  if(ESP.getFreeHeap() > 12000){
    Serial.println("Recording Audio");

    samples = (byte*)malloc(11025);

    if(samples == NULL) Serial.print("Malloc Failed");

    Serial.println("Free Mem wav malloc " + (String)ESP.getFreeHeap() + " free");
    //sample audio using interrupt
    
    //SPI.beginTransaction(SPISettings(2000000, MSBFIRST, SPI_MODE0));
    reading = 0;
    timer1_isr_init();
    timer1_attachInterrupt(sample);
    timer1_enable(TIM_DIV1, TIM_EDGE, TIM_LOOP);
    timer1_write(7256);    //80 ticks per uS

    while(reading < 11025){ Serial.print("."); delay(100); }

    Serial.println("Mem check 1 " + (String)ESP.getFreeHeap() + " free");

    //now we append wave file by sending binary
   // client.write((byte*)&recording, sizeof(recording));
    Serial.print("Header sent");
    //client.write((byte*)samples, 11025);
    Serial.print("data sent");
    //Serial.write((char*)&recording, sizeof(recording));
    //Serial.write((char*)samples, 11025);

  }

}
```

### Debug Messages

```
messages here
```

igrr, 2016-03-09:
I was able to reproduce this. 
However, i think you have an off-by-one error in your interrupt handler. When I replaced
```
if(reading>11025) timer1_disable();
```
with
```
if(reading>=11025) timer1_disable();
```
The code started working fine. Oh, and I have also added `free(samples);` line near the end of your loop.

madsci1016, 2016-03-09:
Dammit, I'm sorry @igrr I though I checked this well enough. Have I mentioned you are awesome?

Verified that mod works, however when I pass that 11KB buffer to transmit using client.write((byte*)samples, 11025), i get an exception. If I use client.write_P((char*)samples, 11025); it seems to work fine. 

The full sketch calls deepsleep at the end so I didn't bother freeing the memory, as loop should never return. But for best practice I'll call it after I'm done sending the wav file to the server.

igrr, 2016-03-11:
Yep, current client.write implementation can not handle buffers of arbitrary length. I'm not sure why it crashes, but we try to only pass it chunks of max 1460 bytes everywhere we use it.
There is a pull request about client.write refactoring, which will address this (and many other issues), but i haven't finished working on it yet.

closed 2016-03-11 by igrr