esp8266/Arduino #1740: TLS server

igrr, 2016-03-09:
Wrapping TLS server support into something like WiFiServerSecure should be pretty easy. 

- [ ] share axTLS context (`SSL_CTX`) between WiFiClientSecure and WiFiServerSecure
- [ ] do SSL initialization in WiFiServerSecure
- [ ] add some methods to load server certificate and private key from FS (similar to WiFiClientSecure)
- [ ] might need some tweaks to ESP8266WebServer to make it https-aware

bbasil2012, 2016-03-09:
do SSL initialization in WiFiServerSecure

AdamMiltonBarker, 2016-07-18:
Is this completed now? I am very happy if it is :D Any docs anywhere for this ?

igrr, 2016-07-18:
No, this is an open issue.

AdamMiltonBarker, 2016-07-18:
Ok thanks is there much progress with it or ETA ?

yhua537, 2016-09-06:
Any update on TSL server?

me-no-dev, 2016-09-06:
I have Tls server working with ESPAsyncTCP. Requires git version of this repo and a define in the lib to be changed. Running ssl server eats heap ;) monitor it or get exceptions if you run out.

yhua537, 2016-09-06:
Thanks, i was reading your ESPAsyncTCP, look like have to convert my standard web server to async version. Any chance of merging async server code back into standard esp8266/Arduino library? And,  (outside this discussion) using a similar style of functions server.arg instead of request -> arg fro an easier conversion?

me-no-dev, 2016-09-07:
Server.arg is not an option here, because in async you are handling multiple requests at the same time and each has its own args and more. I doubt changing server.arg to request->arg is that big of a deal ;) and also i never meant it to be a drop-in replacement

yhua537, 2016-09-07:
Thanks for explanation.

igrr, 2017-05-08:
https://github.com/esp8266/Arduino/pull/3001 contains an implementation of WiFiServerSecure. Haven't reviewed/tested it yet, if anyone is interested in this, please try the PR.

closed 2018-03-08 by igrr