esp8266/Arduino #1742: System crash on servo library after moving from stable 2.0.0 to 2.1.0

mvdbro, 2016-03-09:
### Basic Infos

#### Hardware
Hardware:		NodeMCU V1.0
Core Version:      	2.1.0

### Description

Spontaneous crash using servo library after moving from core 2.0.0. stable to 2.1.0 stable. No changes in the sketch.

### Settings in IDE

Module:  			NodeMCU 1.0
Flash Size: 		4MB
CPU Frequency:	80Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

### Sketch

```
ESP Easy R84, to large to copy here...
https://github.com/ESP8266nu/ESPEasy
```

### Decoded ESP Exception
```

0x4021d968: isTimerActive at C:\Tools\arduino-1.6.5P210\portable\packages\esp8266\hardware\esp8266\2.1.0\libraries\Servo\src\esp8266/Servo.cpp line 306
0x40107354: timer0_isr_handler at C:\Tools\arduino-1.6.5P210\portable\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266/core_esp8266_timer.c line 80
0x40107370: timer0_isr_handler at C:\Tools\arduino-1.6.5P210\portable\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266/core_esp8266_timer.c line 85
0x40229d0e: ieee80211_ht_updateparams at ?? line ?
0x40248771: set_dpd_bypass at ?? line ?
0x4022a749: ieee80211_parse_beacon at ?? line ?
0x40106415: ets_timer_disarm at ?? line ?
0x40106762: spi_flash_read at ?? line ?
0x401077e4: pvPortZalloc at C:\Tools\arduino-1.6.5P210\portable\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266/heap.c line 33
0x4022639d: pm_set_sleep_time at ?? line ?
0x40226832: pm_get_sleep_type at ?? line ?
0x402231cc: pp_noise_test at ?? line ?
0x402268df: pm_get_sleep_type at ?? line ?
0x402402dd: ets_timer_handler_isr at ?? line ?
0x40201a70: delay_end at C:\Tools\arduino-1.6.5P210\portable\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266/core_esp8266_wiring.c line 40
0x402402dd: ets_timer_handler_isr at ?? line ?
0x402402dd: ets_timer_handler_isr at ?? line ?
0x40240322: ets_timer_handler_isr at ?? line ?
0x4021ec8f: loop_task at C:\Tools\arduino-1.6.5P210\portable\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266/core_esp8266_main.cpp line 43

```

igrr, 2016-03-09:
@mvdbro Is it exception 0 or some other exception? (should be in the first line of crash dump)

@Makuna my guess since you have moved disabling of timer into the interrupt in 2425ee34,  `isTimerActive` and `finISR` should also be marked with `ICACHE_RAM_ATTR`.

mvdbro, 2016-03-09:
Yes it's an exception (0)

Danorexic, 2016-03-30:
Experienced the same exception and issues today. Took me a bit to track down it being tied to 2.2 and Servo. Downgrading to 2.0 and restarting the IDE fixed it.

Johboh, 2016-04-13:
I also experience this on 2.1. After downgrade to 2.0 it works fine.

0x40227fc1: ets_timer_handler_isr at ?? line ?
0x4010174b: ppProcessTxQ at ?? line ?
0x40228022: ets_timer_handler_isr at ?? line ?
0x40207b8b: loop_task at C:\Users\johboh\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266/core_esp8266_main.cpp line 4

igrr, 2016-04-13:
Please check staging 2.2.0-rc1.

closed 2016-04-18 by igrr

p4yl04d3r, 2016-06-02:
I to have been hit by this bug in 2.2.0 and 2.1.0.  But I am confused, 2.2.0 is the current stable yet 2.2.0-rc1 has a fix. I am accustom to "rc" being "release candidate". What is 2.2.0-rc1 a candidate to?  I would expect 2.2.0 but that is already released and marked stable.

Makuna, 2016-06-02:
@p4yl04d3r I see the changes in 2.2.0 locally, and this is from a Arduino IDE 1.6.8 Board Manager install and it is working correctly.  How are you installing the board support?

p4yl04d3r, 2016-06-02:
@Makuna I found the commit 8d43c40311708867626120b9d7a57d5cdd8c9aed and verified that I did have the fix from this but.
I am using IDE 1.6.9. I added the Boards URL in Preferences "http://arduino.esp8266.com/stable/package_esp8266com_index.json"
I can confirm I do have an issue with 2.2.0  where 2.0.0 is OK.
I will open a new issue.