esp8266/Arduino #1745: mktime return bad parameter

pipi61, 2016-03-09:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			?ESP-03?
Core Version:      	??

git version 2016 marc 01 Arduino167 esp8266-2.1.0  sdk1.5.1_16_01_08
### Description

Problem description

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?512k?
CPU Frequency:		?80Mhz?
Flash Mode:			?dio?
Flash Frequency:	?40Mhz?
Upload Using:		? SERIAL?
Reset Method:		?ck?

### Sketch

```cpp
#include <stdio.h>
#include <time.h>
void setup(void)
{
	int year;
  struct tm *idotmpstruct, timetmps32;
  time_t timetmp32;

  char buff[100];
  //  1332640800 - 2; //1332640800 2012/3/25  hetn:1   2:0:0
  
   Serial.begin(115200);
  timetmp32 = 0;
  idotmpstruct = &timetmps32;
//idotmpstruct=localtime(&timetmp32);
  for (year = 2012; year < 2015; year++)
	  {
		idotmpstruct->tm_year = year - 1900;
		idotmpstruct->tm_mon = 2;	/*//02->marcius */
		idotmpstruct->tm_mday = 25;	/*/lehetsges vasrnap :25..31 */
		idotmpstruct->tm_hour = 2;
		idotmpstruct->tm_min = 0;
		idotmpstruct->tm_sec = 0;
		idotmpstruct->tm_wday = 0;	/*/dummy */
		timetmp32 = mktime(idotmpstruct);	/*/elllit timestamp */
		//printf(" %lu ", timetmp32);
		Serial.print(timetmp32);
		idotmpstruct = localtime(&timetmp32);
		sprintf(buff, " %d/%d/%d  %d:%d:%d  hetn:%d \r\n",	/* Weekday (0--6; Sunday,vas = 0) */
				idotmpstruct->tm_year + 1900, idotmpstruct->tm_mon + 1, idotmpstruct->tm_mday, idotmpstruct->tm_hour, idotmpstruct->tm_min, idotmpstruct->tm_sec, idotmpstruct->tm_wday);	/*/,tb.millitm); */
		//printf("%s", buff);
		Serial.print(buff);
	  }
}
void loop ()
{
}

### Debug Messages

```
messages here
ESP 8266 mktime return bad parameter, display output:
-582025952 1951/7/23  14:7:28  hetn:1 
-550489952 1952/7/22  14:7:28  hetn:2 
-518953952 1953/7/22  14:7:28  hetn:3 

http://www.tutorialspoint.com/compile_c_online.php
if compile on linux C, the display output ok:
 1332640800 2012/3/25  2:0:0  hetn:0
 1364176800 2013/3/25  2:0:0  hetn:1
 1395712800 2014/3/25  2:0:0  hetn:2

Thank You
Istvan

igrr, 2016-07-04:
Fixed in git version after the switch to full newlib build.

bhcuong2008, 2016-10-15:
Hi igrr,

My program is running well on 2.3 stable. Please tell me how to patch mktime for right working.

Thank you.

pipi61, 2016-10-16:
Hi!
I not tested this new patch,
I search long time ago and use this: http://www.jbox.dk/sanos/source/lib/time.c.html

bhcuong2008, 2016-10-16:
Thank pipi61 :)

I will try it.

closed 2018-01-02 by igrr