esp8266/Arduino #1748: WeMos D1 R2

roboticboyer, 2016-03-10:
The WeMos Board D1 version R1 and R2 have a different pin mapping.
[Wemos_Pins_00.pdf](https://github.com/esp8266/Arduino/files/166799/Wemos_Pins_00.pdf)

The [pins_arduino.h ](https://github.com/esp8266/Arduino/blob/master/variants/d1/pins_arduino.h) is related to version R1.
The R2 version should have the pins_arduino.h of the Nodemcu V0.9 kit

Using a command like digitalWrite(D3, HIGH) it doesn't work, due to wrong GPIO mapping.

I suggest to add 2 boards D1 R1 and D1 R2 with the right pins_arduino.h

thank you
Ciao
Gio

roboticboyer, 2016-03-10:
I've made a further investigation.
The new core 2.1.0 correctly implement the [pins_arduino.h](https://github.com/esp8266/Arduino/blob/master/variants/d1_mini/pins_arduino.h) for D1 R2 (it's the same of D1 mini)

There are 2 boards:
- D1 mini & D1 R2
- D1 R1

igrr, 2016-03-11:
@roboticboyer Is the issue solved for you, or is something still wrong in 2.1.0?

roboticboyer, 2016-03-11:
Hi @igrr 
Now in 2.1.0 The Pin Mapping is OK for all WeMos boards (D1 R1, D1R2 and D1 mini).
You can close the issue.

Thank you for your replay.
Ciao
Gio

igrr, 2016-03-12:
- [ ] b

closed 2016-03-12 by igrr