esp8266/Arduino #1755: Android + softap

sxserjio, 2016-03-11:
I recently purchased the ESP module-12E and installed the Arduino IDE 1.6.6. I ran into a problem! The standard example WiFiAccessPoint does not work in conjunction softap is enabled + Android. Phone and tablet display the authorization error, while iOs devices work perfectly! The solution posted here [https://github.com/esp8266/Arduino/issues/515](https://github.com/esp8266/Arduino/issues/515) not helping me.

igrr, 2016-03-12:
Are you able to connect to an AP set up with AT firmware on the same module?

sxserjio, 2016-03-14:
No! I used the firmware AT21SDK95-2015-01-24 and AT22SDK100-2015-03-20-boot1.3b.bin. None of them, I could not connect to the esp8266 module. But I found the Samsung Galaxy S3. It connects to the access point you created as the firmware and the Arduino IDE without problems.

sxserjio, 2016-03-14:
Sketch TestEspApi when attempting to connect displays the message "EVENT_MAX".

igrr, 2016-03-14:
@sxserjio if your issue persists with AT firmware, then it is out of scope of this project. We can not go troubleshooting the bugs in the SDK. Your best bet is to report this issue to Espressif.

devyte, 2017-10-18:
I can connect with any device, including Android/iOS/laptop+Windows/laptop+Linux, to an ESP12's SoftAP.
Can't reproduce => closing.

closed 2017-10-18 by devyte