esp8266/Arduino #1757: WeMOS R1 D2  issuing "wdt reset" messages

dspdon, 2016-03-12:
### Basic Infos
Running WiFiScan.ino example with WeMOS R1 D2, started failing with "wdt reset" message

#### Hardware
Hardware:	WeMos R1 D2

### Description
Ran WiFiScan.ino example with WeMOS R1 D2, ran fine a few times.  I changed serial port rates, ran fine.  Changed back, code started failing with message:

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)
 wdt reset

Sometimes a stack dump would also display repetitively.

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch
/*
 *  This sketch demonstrates how to scan WiFi networks. 
 *  The API is almost the same as with the WiFi Shield library, 
 *  the most obvious difference being the different file you need to include:
 */
#include "ESP8266WiFi.h"

void setup() {
  Serial.begin(115200);

  // Set WiFi to station mode and disconnect from an AP if it was previously connected
  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  //ESP.wdtDisable();

  delay(100);
}

void loop() {
  Serial.print(F("Scanning ... "));

  int n = WiFi.scanNetworks();
  if (n == 0) {
    Serial.println(F("no networks"));
  } else {
    Serial.print(n);
    Serial.println(F(" networks"));
    for (int i = 0; i < n; ++i) {
      Serial.print(i + 1);
      Serial.print(": ");
      Serial.print(WiFi.SSID(i));
      Serial.print(", ch ");
      Serial.print(WiFi.channel(i));
      //WiFiPhyMode_t pm = WiFi.getPhyMode();
      Serial.print(", ");
      Serial.print(WiFi.RSSI(i));
      Serial.print(" dBm, ");
      Serial.println((WiFi.encryptionType(i) == ENC_TYPE_NONE) ? "" : "*");
      delay(10);
    }
  }
  Serial.println();

  delay(5000);
}

igrr, 2016-03-12:
Which version of the core are you using (2.1.0 or latest git?)

dspdon, 2016-03-14:
How do I determine which core I am using?

krzychb, 2016-03-15:
@dspdon,

1. Go to Tools > Board: (Board Name) > Boards Manager.
2. Scroll down to esp8266
3. You should see _version N.N.N INSTALLED_ (see below).
4. If version is not listed, you are likely using git version

Krzysztof

[image]

dspdon, 2016-03-16:
I have version 2.1.0:

[image: Inline image 1]

On Tue, Mar 15, 2016 at 1:04 AM, Krzysztof <notifications@github.com> wrote:

> @dspdon <https://github.com/dspdon>,
>
>    1. Go to Tools > Board: (Board Name) > Boards Manager.
>    2. Scroll down to esp8266
>    3. You should see *version N.N.N INSTALLED* (see below).
>    4. If version is not listed, you are likely using git version
>
> Krzysztof
>
> [image: core version]
> <https://cloud.githubusercontent.com/assets/12953787/13768100/577aeeaa-ea72-11e5-983b-bab3d03bfe01.png>
>
> —
> You are receiving this because you were mentioned. Reply to this email
> directly or view it on GitHub: <a href="
> https://github.com/esp8266/Arduino/issues/1757#issuecomment-196662818">
> https://github.com/esp8266/Arduino/issues/1757#issuecomment-196662818</a>
>

yomasa, 2016-04-06:
delay too long in loop maybe?

stylerucom, 2016-06-25:
@yomasa , I've got the same error, in my case because of loop - if I add delay(1) works fine.
What's the solution? I have loop of approximately 1500 repeats. What can you advise? Thnx!

stylerucom, 2016-06-25:
Removing writing to Serial is the solution in my case.

devyte, 2017-10-18:
Scanning is being done on every loop pass, including print out to Serial. The execution of the loop body should be spaced out by polling millis(), and actually do a scan every e.g.: 6s.
Also, the relevant underlying code has changed quite a lot since this issue was opened.
Closing.

closed 2017-10-18 by devyte