esp8266/Arduino #1758: Example for new libb64

koffienl, 2016-03-12:
I want to move my current project to the 2.1.0 core, where can I find an example for the usage of the new build-in libb64 ?

Links2004, 2016-03-13:
a Arduino like API is only for encode implemented right now:
https://github.com/esp8266/Arduino/blob/master/cores/esp8266/base64.h#L28-L33
the libb64 itself can handle encode and decode.

may the libb64 examples help you:
https://sourceforge.net/p/libb64/git/ci/master/tree/examples/
https://sourceforge.net/p/libb64/git/ci/master/tree/examples/c-example1.c

koffienl, 2016-03-13:
Thanks, will look into it :+1:

Humancell, 2017-03-04:
I'm not able to get this working ... what do I have to include to get access to these functions?

error: 'encode' was not declared in this scope

vicnevicne, 2017-03-04:
Hi, humancell, 
Please refrain from cross-posting. You seem to have an issue here but asking the same question in #507 and requesting "a complete ESP example" in #1727 will not motivate developers to help you.
This issue tracking system is meant for problems and bugs with the Arduino environment. If you really need help of a first ESP example, please head to the esp8266.com forums where many people will be able to help you.
KR,

devyte, 2017-10-18:
Closing per #3655 .

closed 2017-10-18 by devyte