esp8266/Arduino #1759: ADC_MODE doesn't work

mattwilliamson, 2016-03-12:
I'm trying to get use of `ESP.getVcc()`, so I added the following line, per documentation.

```
ADC_MODE(ADC_VCC);
```

But I get the following error messages during compilation:

```
Arduino: 1.6.8 (Mac OS X), Board: "NodeMCU 1.0 (ESP-12E Module), 160 MHz, Serial, 115200, 4M (3M SPIFFS)"

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware "/Applications/Arduino.app/Contents/Java/hardware" -hardware "/Users/matt/Library/Arduino15/packages" -hardware "/Users/matt/Documents/Arduino/hardware" -tools "/Applications/Arduino.app/Contents/Java/tools-builder" -tools "/Applications/Arduino.app/Contents/Java/hardware/tools/avr" -tools "/Users/matt/Library/Arduino15/packages" -built-in-libraries "/Applications/Arduino.app/Contents/Java/libraries" -libraries "/Users/matt/Documents/Arduino/libraries" -fqbn=esp8266:esp8266:nodemcuv2:CpuFrequency=160,UploadTool=esptool,UploadSpeed=115200,FlashSize=4M3M -ide-version=10608 -build-path "/var/folders/2l/mx56mt0d1yq98jkg_1p04gym0000gr/T/build225067a66c2411afaf5db2c19e6f1e44.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/Users/matt/Dropbox/Projects/Home/tanklevel/tanklevel_blynk/tanklevel_blynk.ino"
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware "/Applications/Arduino.app/Contents/Java/hardware" -hardware "/Users/matt/Library/Arduino15/packages" -hardware "/Users/matt/Documents/Arduino/hardware" -tools "/Applications/Arduino.app/Contents/Java/tools-builder" -tools "/Applications/Arduino.app/Contents/Java/hardware/tools/avr" -tools "/Users/matt/Library/Arduino15/packages" -built-in-libraries "/Applications/Arduino.app/Contents/Java/libraries" -libraries "/Users/matt/Documents/Arduino/libraries" -fqbn=esp8266:esp8266:nodemcuv2:CpuFrequency=160,UploadTool=esptool,UploadSpeed=115200,FlashSize=4M3M -ide-version=10608 -build-path "/var/folders/2l/mx56mt0d1yq98jkg_1p04gym0000gr/T/build225067a66c2411afaf5db2c19e6f1e44.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/Users/matt/Dropbox/Projects/Home/tanklevel/tanklevel_blynk/tanklevel_blynk.ino"
"/Users/matt/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/Users/matt/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/tools/sdk/include" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=160000000L -DARDUINO=10608 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266  -DESP8266       "-I/Users/matt/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266" "-I/Users/matt/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/variants/nodemcu" "/var/folders/2l/mx56mt0d1yq98jkg_1p04gym0000gr/T/build225067a66c2411afaf5db2c19e6f1e44.tmp/sketch/tanklevel_blynk.ino.cpp" -o "/dev/null"
"/Users/matt/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/Users/matt/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/tools/sdk/include" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=160000000L -DARDUINO=10608 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266  -DESP8266       "-I/Users/matt/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266" "-I/Users/matt/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/variants/nodemcu" "-I/Users/matt/Documents/Arduino/libraries/SimpleTimer" "/var/folders/2l/mx56mt0d1yq98jkg_1p04gym0000gr/T/build225067a66c2411afaf5db2c19e6f1e44.tmp/sketch/tanklevel_blynk.ino.cpp" -o "/dev/null"
"/Users/matt/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/Users/matt/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/tools/sdk/include" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=160000000L -DARDUINO=10608 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266  -DESP8266       "-I/Users/matt/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266" "-I/Users/matt/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/variants/nodemcu" "-I/Users/matt/Documents/Arduino/libraries/SimpleTimer" "-I/Users/matt/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/libraries/ESP8266WiFi/src" "/var/folders/2l/mx56mt0d1yq98jkg_1p04gym0000gr/T/build225067a66c2411afaf5db2c19e6f1e44.tmp/sketch/tanklevel_blynk.ino.cpp" -o "/dev/null"
"/Users/matt/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/Users/matt/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/tools/sdk/include" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=160000000L -DARDUINO=10608 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266  -DESP8266       "-I/Users/matt/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266" "-I/Users/matt/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/variants/nodemcu" "-I/Users/matt/Documents/Arduino/libraries/SimpleTimer" "-I/Users/matt/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/libraries/ESP8266WiFi/src" "-I/Users/matt/Documents/Arduino/libraries/Blynk" "/var/folders/2l/mx56mt0d1yq98jkg_1p04gym0000gr/T/build225067a66c2411afaf5db2c19e6f1e44.tmp/sketch/tanklevel_blynk.ino.cpp" -o "/dev/null"
"/Users/matt/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/Users/matt/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/tools/sdk/include" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=160000000L -DARDUINO=10608 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266  -DESP8266       "-I/Users/matt/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266" "-I/Users/matt/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/variants/nodemcu" "-I/Users/matt/Documents/Arduino/libraries/SimpleTimer" "-I/Users/matt/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/libraries/ESP8266WiFi/src" "-I/Users/matt/Documents/Arduino/libraries/Blynk" "/Users/matt/Documents/Arduino/libraries/SimpleTimer/SimpleTimer.cpp" -o "/dev/null"
"/Users/matt/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/Users/matt/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/tools/sdk/include" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=160000000L -DARDUINO=10608 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266  -DESP8266       "-I/Users/matt/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266" "-I/Users/matt/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/variants/nodemcu" "-I/Users/matt/Documents/Arduino/libraries/SimpleTimer" "-I/Users/matt/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/libraries/ESP8266WiFi/src" "-I/Users/matt/Documents/Arduino/libraries/Blynk" "/Users/matt/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/libraries/ESP8266WiFi/src/ESP8266WiFi.cpp" -o "/dev/null"
"/Users/matt/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/Users/matt/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/tools/sdk/include" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=160000000L -DARDUINO=10608 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266  -DESP8266       "-I/Users/matt/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266 [... cut ...]
Using library ESP8266WiFi at version 1.0 in folder: /Users/matt/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/libraries/ESP8266WiFi 
Using library Blynk at version 0.3.3 in folder: /Users/matt/Documents/Arduino/libraries/Blynk 
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).
```

### Basic Infos

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			NodeMCU v3
Flash Size: 		4MB
CPU Frequency:		160MHz
Flash Mode:			?qio?

mattwilliamson, 2016-03-12:
Arduino 1.6.8

igrr, 2016-03-12:
This line has to be outside of function scope, as mentioning in the docs. If it's not, please post the sketch.

mattwilliamson, 2016-03-12:
It's right at the top of the file:

```
ADC_MODE(ADC_VCC);

// http://aimatt.com

// Configure this stuff to your setup
char blynk_auth[] = ""; // Blink auth token - get from app http://blynk.cc
char wifi_ssid[] = "";                            // Wifi name
char wifi_pass[] = "";                       // Wifi password

uint echoPin = 5;                                       // D1 on NodeMCU - Echo Pin, 
uint trigPin = 4;                                       // D2 on NodeMCU - Trigger Pin

// --------------------------------

#define SAMPLES 5                 // How many samples to average per reading
#define US_TO_CM 58.2             // How many centimeters sound travels per microsecond in air

// Virtual GPIO from Blynk
#define VPIN_MIN_RANGE V10
#define VPIN_MAX_RANGE V11
#define VPIN_WARN_LEVEL V12
#define VPIN_AVG V5
#define VPIN_PERCENT V6
#define VPIN_BATTERY V7

// Defaults for UI sliders
uint maximumRange = 300;       // Maximum range needed - filters out and used to calculate %
uint minimumRange = 6 * 2.5;   // Minimum range needed - filters out and used to calculate %
double warnPercent = 25;       // What % to warn at when below

// Probably don't need to change anything down here
#include <SimpleTimer.h>
#define BLYNK_PRINT Serial
#include <ESP8266WiFi.h>
#include <BlynkSimpleEsp8266.h>

SimpleTimer timer;
double startTime = 0;
volatile double endTime = 0;
double avgDistance = 0;
double lastDistancePercent = 0;
double distance = 0;
double distancePercent = 0;
uint readings = 0;

// Trigger pin on SRF04 sonar, then wait for echo
void startMeasure() {
  BLYNK_LOG("startMeasure");

  digitalWrite(trigPin, LOW); 
  delayMicroseconds(2);
  
  digitalWrite(trigPin, HIGH);
  delayMicroseconds(10);

  distance = 0;
  endTime = 0;
  startTime = micros();
  
  digitalWrite(trigPin, LOW);
}

// Received echo on ECHO pin for sonar
void endMeasure() {
  endTime = micros();
}

// If we have a good value, send the reading
void sendMeasurement() {
  if (startTime > 0 && endTime > 0) {
    distance = (endTime - startTime) / US_TO_CM;

    Serial.println("Got measurement");
    Serial.println(endTime - startTime);

    Serial.println(distance);
    
    // A valid reading
    if (endTime > startTime && distance >= minimumRange && distance <= maximumRange) {
      readings++;
      avgDistance += distance;

      // Write these so that the app updates with the last settings after restart
      Blynk.virtualWrite(VPIN_MIN_RANGE, minimumRange);
      Blynk.virtualWrite(VPIN_MAX_RANGE, maximumRange);
      Blynk.virtualWrite(VPIN_WARN_LEVEL, warnPercent);
      Blynk.virtualWrite(VPIN_BATTERY, ESP.getVcc());
    }

    startTime = 0;
    endTime = 0;
  }

  // We have enough readings. Average them out and send them.
  if (readings >= SAMPLES) {
    Serial.println("Got enough samples");

    // Divide by the number of readings to get the average
    avgDistance /= readings;
    lastDistancePercent = distancePercent;
    Serial.println(avgDistance);
    distancePercent = (avgDistance - minimumRange) / maximumRange * 100;
    Blynk.virtualWrite(VPIN_AVG, avgDistance);
    Blynk.virtualWrite(VPIN_PERCENT, distancePercent);

    // Reset
    readings = 0;
    avgDistance = 0;

    // Check if the reading is below the warn level
    if (distancePercent < warnPercent && lastDistancePercent > warnPercent && lastDistancePercent > 0) {
        // Send a push notification that the tanks is low
        // Checck a few times to make sure it is not erratic readings
        BLYNK_LOG("Sending low level notification");
        Blynk.notify("Level below warning!");
    }
  }
}

void setup() {    
    Serial.begin(115200);

    BLYNK_LOG("booted");

    Blynk.begin(blynk_auth, wifi_ssid, wifi_pass);

    pinMode(trigPin, OUTPUT);
    pinMode(echoPin, INPUT);

    attachInterrupt(echoPin, endMeasure, FALLING);

    // Run measurement once per second
    timer.setInterval(1000L, startMeasure);

    BLYNK_LOG("done booting");
}

// When the app sets the minimum range via a slider, set the variable
BLYNK_WRITE(VPIN_MIN_RANGE)
{
  if(param.asDouble() > 0)  {
    BLYNK_LOG("Set minimumRange to %s", param.asStr());
    minimumRange = param.asDouble();
  }
}

// When the app sets the maximum range via a slider, set the variable
BLYNK_WRITE(VPIN_MAX_RANGE)
{
  if(param.asDouble() > 0)  {
    BLYNK_LOG("Set maxRange to %s", param.asStr());
    maximumRange = param.asDouble();
  }
}

// When the app sets the warning level (percent), set the variable
BLYNK_WRITE(VPIN_WARN_LEVEL)
{
  if(param.asDouble() > 0)  {
    BLYNK_LOG("Set warn to %s", param.asStr());
    warnPercent = param.asDouble();
  }
}

void loop() {
  Blynk.run();
  timer.run();
  sendMeasurement();
}

```

mattwilliamson, 2016-03-12:
For the original error I pasted, the line number is not 1 because I was moving it around to see if it would fix it.

igrr, 2016-03-12:
This seems to be caused by arduino-builder not generating correct forward declarations for functions marked with `extern C`.
I'll see if we can implement a workaround for this, since the just-released arduino 1.6.8 is affected by this issue.

mattwilliamson, 2016-03-12:
Thanks!

Matt Williamson
845-702-6652

mozgy, 2016-03-13:
At the moment plain
```
#include <Arduino.h>

ADC_MODE(ADC_VCC);

void setup() {}
void loop() {}
```
cannot get compiled with 1.6.8

mattwilliamson, 2016-03-17:
I've confirmed that downgrading to Arduino 1.6.6 bypasses the issue for now. Thanks for looking into this.

closed 2016-04-18 by igrr

tamptus, 2016-04-26:
I had the same Problem with : ADC_MODE(ADC_VCC); and Version 1.6.8
My workaround:
Add a new Tab with file adc.h
with one line: 
ADC_MODE(ADC_VCC)

add
 #include "adc.h"
 as first include in first tab.

pandacriss, 2017-02-04:
i have a related problem :  https://github.com/esp8266/Arduino/issues/2913
thanks tamptus, your trick work for me too.
> I had the same Problem with : ADC_MODE(ADC_VCC); and Version 1.6.8
My workaround:
Add a new Tab with file adc.h
with one line:
ADC_MODE(ADC_VCC)
add
#include "adc.h"
as first include in first tab.

twanek, 2017-02-20:
hello!

for me it compiles, but i constantly get value 2.75 when ESP.getVcc();
i'm using wemos d1 mini pro + arduino ide 181

any solution for this?

thanks!

mkh595, 2017-12-04:
remove 100k resistor attached to A0