esp8266/Arduino #1761: tone() causes an exception

urish, 2016-03-12:
----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			WeMos D1 Mini
Core Version:      	2.1.0

### Description

Calling `tone()` randomly causes a CPU exception a reset. With the attached sketch, the issue usually reproduces within 10 seconds.

### Settings in IDE

Module:  			WeMos D1 R2 & Mini
Flash Size: 		4MB
CPU Frequency:		80Mhz
Upload Using:		SERIAL

### Sketch

```cpp

void setup() {
  Serial.begin(9600);
  pinMode(D6, OUTPUT);
}

void loop() {
  Serial.println("Loop");
  tone(D6, 500, 200);
  delay(200);
  noTone(D6);
  delay(200);
}

```

### Stack Trace (Decoded)

```
Decoding 23 results
0x4020268c: t1IntHandler() at C:\Users\uri\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266/Tone.cpp line 103
0x4021980d: ieee80211_parse_beacon at ?? line ?
0x401016d1: ets_timer_disarm at ?? line ?
0x4010234e: spi_flash_read at ?? line ?
0x4010724c: pvPortZalloc at C:\Users\uri\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266/heap.c line 33
0x40211cc1: pm_set_sleep_time at ?? line ?
0x4021215a: pm_get_sleep_type at ?? line ?
0x402140dc: pp_noise_test at ?? line ?
0x40212207: pm_get_sleep_type at ?? line ?
0x4020512d: ets_timer_handler_isr at ?? line ?
0x4020512d: ets_timer_handler_isr at ?? line ?
0x40205172: ets_timer_handler_isr at ?? line ?
```

### Stack Trace (Original)

```
Exception (0):
epc1=0x4020268c epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffc50 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffdf0:  00000006 00000006 42d80000 42d80000  
3ffffe00:  00000005 00000000 3fffc230 4000050c  
3ffffe10:  40004376 00000030 00000016 ffffffff  
3ffffe20:  60000200 00000008 0001e100 80000000  
3ffffe30:  20000000 3fff0768 80000000 203fc280  
3ffffe40:  00000000 3fffc6fc fc70ffff 3fff076c  
3ffffe50:  00000074 003fc280 60000600 00000030  
3ffffe60:  00000000 3ffe9325 3ffe9320 3ffebb74  
3ffffe70:  4021980d 00000002 00000000 00000030  
3ffffe80:  ffffffd4 3ffebb50 ffffffd4 3ffebb60  
3ffffe90:  00000000 00000000 0000001f 401016d1  
3ffffea0:  4000050c 00000001 3fffc230 4000050c  
3ffffeb0:  40000f68 00000030 0000001c ffffffff  
3ffffec0:  40000f58 00000000 00000020 00000000  
3ffffed0:  003744f7 00000001 00003a98 00000001  
3ffffee0:  ffffffff 3fffc6fc fc70ffff 3fffdab0  
3ffffef0:  00000000 3fffdcb0 3ffe8870 00000030  
3fffff00:  00000000 400042db 3ffebb95 60000600  
3fffff10:  40004b31 3fff04ec 000002f4 003fc000  
3fffff20:  4010234e 3ffe8860 3ffe9000 4010724c  
3fffff30:  40211cc1 3ffe9000 3ffe8860 013c706d  
3fffff40:  3fff04ec 00001000 4021215a 00000008  
3fffff50:  402140dc 3ffe9154 40212207 3ffe90b4  
3fffff60:  3ffe8860 4020512d 3ffe8ccc 3ffe8860  
3fffff70:  4020512d 60000600 3ffe93b0 3ffe93b0  
3fffff80:  40205172 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffe8878 00000000 40000f65 3fffdab0  
3fffffa0:  40000f49 0001883a 3fffdab0 40000f49  
<<<stack<<<

```
```

igrr, 2016-03-12:
`t1IntHandler` needs to be marked with ICACHE_RAM_ATTR.

urish, 2016-03-12:
@igrr brilliant, seems to fix the issue. Do you want me to send a PR?

igrr, 2016-03-12:
Yes, if could add that to your pull request that would be cool.

urish, 2016-03-12:
sure thing

urish, 2016-03-12:
done :)

closed 2016-04-18 by igrr