esp8266/Arduino #1763: Compilation time with Arduino 1.6.7

vlast3k, 2016-03-12:
ever since i upgraded to Arduino 1.6.7, the compilation time of the sketches has increased quite a lot. I asked at the Arduino forum and i was told that this is a problem of the new linker. With ESP8266 sketches the difference is really huge - from 1 sec to recompile a modified sketch with 1.6.5, to more than 20 seconds in 1.6.7.
Do you experience the same or am i doing something wrong. I tried to do several clean installations, but nothing helped :(

krzychb, 2016-03-13:
Hi vlast3k,

Inspired by [this post](http://www.esp8266.com/viewtopic.php?p=42816#p42816) and your question please find below by results.

[image]

If I do not change my sketch and compilation options, then the times are about 2x shorter. I have not observed this process to take only 1s :smile:

On my PC I believe it depends on the number of object files (`*.c.o`  and `*.cpp.o`) created / reused by compiler / linker. For the sample sketch above there are about 50 object files created. By going to File > preferences and checking out “compilation” under “Show verbose output during:”, I can see several `Using previously compiled file:` messages depending on that I change in the project. I am planning to replace my HDD with a SDD to improve performance.

Regarding question asked on Arduino form I would also consider that a typical Arduino sketch has likely half that number of object files and the total size of files is likely 10x smaller.

Krzysztof

vlast3k, 2016-03-13:
> about 2x shorter. I have not observed this process to take only 1s [image:

on my laptop the recompilation time of this sketch is 6 sek with 1.6.5 and
12 sek with 1.6.8.. (first compilation is 30 sek, so perhaps the difference
is from the SSD)., which is bearable
Unfortunately the sketch that i use recompiles for 30 sek on 1.6.8, and on
1.6.5 i can no longer compile it for some reason )some strange compilation
issues)
but .. yeah.. apparently there are no magic spells to make it faster :)

vlast3k, 2016-03-13:
hmm i noticed also another thing that may be the root cause of my issues, since i had some problems with the automatic upload to the boards, I started exporting the binary and then uploading from the command line.. and in this case - in case there are unsaved files in the sketch - the whole sketch is copied to a new folder and compiled from scratch....
if i save before this then the compilation is for 20-30 seconds, which is still better than a minute :)

closed 2016-03-13 by vlast3k