esp8266/Arduino #1765: Donations

msurguy, 2016-03-13:
Hello! This repository is perhaps one of the most important IoT repos on the web currently and I'd like to support the contributors by means of small donation. 

Is there a way to Donate towards support of this project? If so, I think that should be also specified in the Readme documentation :)

Thanks for creating this and working on it so much!

shirish47, 2016-03-13:
Yes its mention on on first page ..paypal. I tried to donate, but its paypal I don't own a credit card, I live in india. I really wanted to thanks this community. Currently I am not using this core in any commercial purpose. But in case if we do I would like to mention it, all libraries, etc.

Humancell, 2016-03-14:
Hello,  I too want to donate ... and I did see the Donate button on the home page of the project.  This appears to send money to "support-esp8266@esp8266.com" ... is this really going to the team of contributors of the Arduino Core port?

igrr, 2016-03-14:
Guys,
thank you for your support. The donate button on front page sends money to help maintain the awesome esp8266.com community forum.

My suggestion for anyone who wants to contribute *to this project* by donating is as follows. 
- Pick any of the [open issues](https://github.com/esp8266/Arduino/issues) which you find meaningful and which you would like to see implemented/fixed. If it doesn't exist yet, create one. 
- Go to https://www.bountysource.com/ and post a bounty on that issue.

We really have more open stuff than we can handle, so attracting new contributors by posting bounties is a good way to spend money.

krzychb, 2016-03-14:
Simple and beautiful initiative :+1:

msurguy, 2016-03-25:
Thank you @igrr !

closed 2016-06-23 by igrr