esp8266/Arduino #1766: analog pin paramter for analogRead not zero based

jnsbyr, 2016-03-13:
### Basic Infos

#### Hardware
Hardware:		ESP8266 - all boards
Core Version:      	2.1.0

### Description

The analog pin parameter for analogRead() is currently not zero based but absolute (A0 = 17) in contrast to the standard Arduino library implementation (see https://www.arduino.cc/en/Reference/AnalogRead). This makes ports of existing projects (e.g. https://github.com/firmata/arduino) harder.

### Possible Solution

* change in all variants/.../pins_arduino.h: 
```C
...
static const uint8_t A0 = 0;
...
```

* change in core_esp8266_wiring_analog.c:
```C
...
extern int __analogRead(uint8_t pin) {
  if(pin == 0){
    return system_adc_read();
  }
  return digitalRead(pin) * 1023;
}
...
```

igrr, 2016-03-13:
It's not zero based for at least two variants of the AVR core:
https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/variants/standard/pins_arduino.h#L47-L54
https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/variants/leonardo/pins_arduino.h#L114-L125

jnsbyr, 2016-03-13:
Thx for considering the suggestion. Keeping A0 at 17 will not be a problem if analogRead works with pin=0.

jnsbyr, 2016-03-15:
An analysis has shown that the wifio board has some extras that need to be considered, especially A1 to A7 in addition to A0. Best I can think of would be to do it like we agreed for A0 and support access both absolute and zero based. There are no overlaps between absolute pin numbers and zero based pin numbers even with wifio so this should be possible. This issue blocks the integration of the ESP8266 into firmata/arduino (see https://github.com/firmata/arduino/pull/278#issuecomment-196550504).

igrr, 2016-03-16:
@jnsbyr this is now fixed in git version.

jnsbyr, 2016-03-16:
Thanks, will let you know how it works.

jnsbyr, 2016-03-18:
With the new version ```analogRead()``` now works for A0 as expected.

There are similar problems with the macros ```digitalPinToInterrupt(p)``` and ```digitalPinHasPWM(p)```. E.g. the second macro returns true for pins 1 to 17 with an Adafruit board. But it should be 0 to 16 or even better 0 to 5 and 12 to 16, because you should not do PWM on your flash IOs. The reason is that the current macro uses NOT_A_PIN but that resolves to -1 and that is true for 17 and it returns the pin number and that is false for 0.

My suggestions:

* minimum change
```C
#define digitalPinToInterrupt(p)  	(((p) < EXTERNAL_NUM_INTERRUPTS)?1:0)
#define digitalPinHasPWM(p)         (((p) < NUM_DIGITAL_PINS)?1:0)
```

* saveguard approach
```C
#define digitalPinToInterrupt(p)  	(((p) < EXTERNAL_NUM_INTERRUPTS && ((p) < 6 || (p) > 11))?1:0)
#define digitalPinHasPWM(p)         (((p) < NUM_DIGITAL_PINS && ((p) < 6 || (p) > 11))?1:0)
```

jnsbyr, 2016-04-01:
@igrr 
From my point of view this issue is resolved and can be closed.

igrr, 2016-04-01:
@jnsbyr Thanks! It will get closed (automatically) once we deploy the new release (2.2.0).

closed 2016-04-18 by igrr