esp8266/Arduino #1767: Exception(2) after connect to Accespoint

mreschka, 2016-03-13:
### Basic Infos

#### Hardware
Hardware:			ESP-01
Core Version:      	2.1.0 (via "stable" in Board-Manager; Arduino 1.6.8)

### Description

Exception(2) after connect to Accesspoint.

I am using a ESP-01 with Arduino 1.6.8. I started with the Serial Telnet Example and are getting the above Exception after I got a connection to my acces point.

I tested using two different ESP-Modules.

The epc1=0x3ffe86d0 is constant, so no sparadic problem.

I used two different Access Points an different WPA settings.

Power is from a 3.3V 800 mA Buck power Supply from 12V 2 Amp Supply, not extra capacitors.

Serial is a FTDI 3,3V Board.

I included the elf file and the sketch here:
[elf_and_bin.zip](https://github.com/esp8266/Arduino/files/171117/elf_and_bin.zip)
[sketch.zip](https://github.com/esp8266/Arduino/files/171118/sketch.zip)

I have kept the reset of the sketch-temp-folder, if you need more

The error is 100% reproducible with other sekteches in my envoronment. Strangely using the Wifimanager (another sektch) works - you can connect to the accespoint, access the captive gui and choose a wifi. As soon as you have done that the same error occurs until you reset the wfifimanger-data.

Any ideas?

Thanks in advance.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		512K (64K SPIFFS)
CPU Frequency:		80Mhz
Flash Mode:			DIO
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

### Sketch
`
#include <ESP8266WiFi.h>

//how many clients should be able to telnet to this ESP8266
#define MAX_SRV_CLIENTS 1
const char* ssid = "markusTest";
const char* password = "passwdTestasdf";

WiFiServer server(23);
WiFiClient serverClients[MAX_SRV_CLIENTS];

void setup() {
  Serial.begin(115200);
  WiFi.begin(ssid, password);
  Serial.print("\nConnecting to "); Serial.println(ssid);
  uint8_t i = 0;
  while (WiFi.status() != WL_CONNECTED && i++ < 20) delay(500);
  if(i == 21){
    Serial.print("Could not connect to"); Serial.println(ssid);
    while(1) delay(500);
  }
  //start UART and the server
  Serial.begin(115200);
  server.begin();
  server.setNoDelay(true);
  
  Serial.print("Ready! Use 'telnet ");
  Serial.print(WiFi.localIP());
  Serial.println(" 23' to connect");
}

void loop() {
  uint8_t i;
  //check if there are any new clients
  if (server.hasClient()){
    for(i = 0; i < MAX_SRV_CLIENTS; i++){
      //find free/disconnected spot
      if (!serverClients[i] || !serverClients[i].connected()){
        if(serverClients[i]) serverClients[i].stop();
        serverClients[i] = server.available();
        Serial.print("New client: "); Serial.print(i);
        continue;
      }
    }
    //no free/disconnected spot so reject
    WiFiClient serverClient = server.available();
    serverClient.stop();
  }
  //check clients for data
  for(i = 0; i < MAX_SRV_CLIENTS; i++){
    if (serverClients[i] && serverClients[i].connected()){
      if(serverClients[i].available()){
        //get data from the telnet client and push it to the UART
        while(serverClients[i].available()) Serial.write(serverClients[i].read());
      }
    }
  }
  //check UART for data
  if(Serial.available()){
    size_t len = Serial.available();
    uint8_t sbuf[len];
    Serial.readBytes(sbuf, len);
    //push UART data to all connected telnet clients
    for(i = 0; i < MAX_SRV_CLIENTS; i++){
      if (serverClients[i] && serverClients[i].connected()){
        serverClients[i].write(sbuf, len);
        delay(1);
      }
    }
  }
}
`

### Debug Messages

Connecting to markusTest
Ready! Use 'telnet 192.168.178.30 23' to connect

Exception (2):
epc1=0x3ffe86d0 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3ffe86d0 depc=0x00000000

ctx: sys 
sp: 3ffffc90 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffe30:  40211835 3ffffed0 3ffed830 3ffed258  
3ffffe40:  40211895 3ffffed0 4021177d 3ffffed0  
3ffffe50:  6b72616d 65547375 40007473 00000100  
3ffffe60:  7fffffff 3ffe9650 3ffe9650 00000001  
3ffffe70:  4020df59 00080000 3fffc200 00000022  
3ffffe80:  ffffffda 00000004 3ffffee0 3ffffef1  
3ffffe90:  40107020 00000000 3fff0f18 000adeab  
3ffffea0:  4010549c 00000001 000000a5 3fff0114  
3ffffeb0:  3fff00fc 00000000 00000000 3ffffe40  
3ffffec0:  402038c0 3ffeef3c 00000000 3fffd9d0  
3ffffed0:  3ffffef0 3ffffee0 00000004 00000045  
3ffffee0:  3ffeafb0 3ffe87dc 3fff0f18 000000e5  
3ffffef0:  000000a5 00003a98 3ffeafbc 00000000  
3fffff00:  00000000 00000000 3ffeafcb 4010031d  
3fffff10:  00000000 00000f28 3ffeeef0 40100426  
3fffff20:  40209402 00000000 0000003a 40100537  
3fffff30:  4020707c 3ffee354 3fff0064 401008c1  
3fffff40:  40211e8e 3ffec830 00000000 40107218  
3fffff50:  3ffeaf8c 000000aa 00000000 4021c969  
3fffff60:  00000000 40223ab1 3ffee608 3ffee630  
3fffff70:  018e2c54 60000600 3ffe8ec0 3ffe8ec0  
3fffff80:  40223af6 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffee658 3fffdad0 3ffeef3c 402038e3  
3fffffa0:  40000f49 40000f49 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

Stack Decoded:

Decoding 23 results
0x40211835: scan_parse_beacon at ?? line ?
0x40211895: scan_parse_beacon at ?? line ?
0x4021177d: scan_parse_beacon at ?? line ?
0x4020df59: ieee80211_parse_beacon at ?? line ?
0x40107020: printf at C:\Users\markus\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266/libc_replacements.c line 206
0x4010549c: chm_get_current_channel at ?? line ?
0x402038c0: loop_task at C:\Users\markus\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266/core_esp8266_main.cpp line 43
0x4010031d: check_poison at C:\Users\markus\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266\umm_malloc/umm_malloc.c line 817
0x40100426: check_poison_block at C:\Users\markus\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266\umm_malloc/umm_malloc.c line 851
0x40209402: esf_buf_recycle at ?? line ?
0x40100537: check_poison_all_blocks at C:\Users\markus\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266\umm_malloc/umm_malloc.c line 880
0x4020707c: ppRecycleRxPkt at ?? line ?
0x401008c1: free at C:\Users\markus\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266\umm_malloc/umm_malloc.c line 1717
0x40211e8e: sta_input at ?? line ?
0x40107218: vPortFree at C:\Users\markus\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266/heap.c line 18
0x4021c969: sys_check_timeouts at ?? line ?
0x40223ab1: ets_timer_handler_isr at ?? line ?
0x40223af6: ets_timer_handler_isr at ?? line ?
0x402038e3: loop_task at C:\Users\markus\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266/core_esp8266_main.cpp line 43

chrwei, 2016-03-14:
I'm also getting this with the included telnet serial example on a ESP-12 (adafruit huzzah if that matters) on board package version 2.1.0 but it works fine on 2.0.0.  many other examples work fine.

my serial output is:
`
ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

Exception (2):
epc1=0x3ffe849f epc2=0x00000000 epc3=0x00000000 excvaddr=0x3ffe849c depc=0x00000000

ctx: sys 
sp: 3ffffc90 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffe30:  40211601 3ffffed0 3ffed5c0 3ffecfac  
3ffffe40:  40211661 3ffffed0 40211549 3ffffed0  
3ffffe50:  736c6977 00006e6f 40103052 00000100  
3ffffe60:  7fffffff 3ffe93d4 3ffe93d4 00000001  
3ffffe70:  4020dd25 00080000 00000004 40107208  
3ffffe80:  00000000 00000004 3ffffee0 3ffffef1  
3ffffe90:  40107020 00000000 3fff0518 000611c1  
3ffffea0:  4010549c 00000030 000000a5 ffffffff  
3ffffeb0:  402036b3 00000000 00000000 3ffffe40  
3ffffec0:  feefeffe 00000000 3fffd9d0 3ffeeccc  
3ffffed0:  3ffffef0 3ffffee0 00000004 3ffeece0  
3ffffee0:  00000000 3ffe8595 3fff0518 000000e5  
3ffffef0:  000000a5 00003a98 3ffee3d0 00000030  
3fffff00:  00000000 00000000 00000000 4010031d  
3fffff10:  00000000 00000798 3ffeec80 40100426  
3fffff20:  402091ce 00000000 00000020 40100537  
3fffff30:  40206e48 3ffee0e4 3ffefdf4 401008c1  
3fffff40:  40211c5a 3ffec5c0 00000000 40107218  
3fffff50:  00000000 00000000 00000001 4021c735  
3fffff60:  00000000 4022387d 3ffee398 3ffee3c0  
3fffff70:  0013c9ad 60000600 3ffe8c28 4022387d  
3fffff80:  402238c2 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffee3e8 3fffdad0 3ffeeccc 402036ab  
3fffffa0:  40000f49 40000f49 3fffdab0 40000f49  
<<<stack<<<
`

(FWIW, the "code" markdown doesn't seem to be working here.)

fussel1976, 2016-03-16:
Hi... change your sketch setup: remove the line with
server.setNoDelay(true)

or change the parameter to false.... it worked for me.

for more information have  look at issue #1695 

greetz fussel

mreschka, 2016-03-20:
Hi fussel,
that makes it work :-) 

Thanks a lot!

closed 2016-03-20 by igrr