esp8266/Arduino #1769: serial transparent  bridge sample

mkeyno, 2016-03-13:
hi @igrr 
there is [couple of code](https://github.com/jeelabs/esp-link) written in lua which enable the serial transparent bridge between the serial pin connected to ESP and web interface which is inserted to module flash chip , I'm not interested to Lua and  I  suggested to @Links2004  to create sample for such wireless link which is would be so helpful for project like control 3d printer wirelessly, whereas already done by Bluetooth  
I know you may so busy for that but if you know guys or links lets to create such sample
best regards

luc-github, 2016-03-13:
What is the difference with : https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/WiFiTelnetToSerial/WiFiTelnetToSerial.ino

mkeyno, 2016-03-13:
hi @luc-github , but this code not secure and open to many clients, and as we discuss with @Links2004  the websocket protocol   is so reliable regards to that bare sample code , and I suggest  to create sample with  websocket library where web interface file store in flash
but Luc as I know you are also professional in 3d printer firmware, I've had plan to discuss with you about this wireless  link, whereas need small webpage to send the g code to board   and take back its respond to webpage , I did it with Bluetooth and Repetier server but I thinks ESP can have both in same time
I really interested to have your respond
BR

closed 2016-04-10 by mkeyno