esp8266/Arduino #1770: Error -8 during  tcp_output()

ba-us, 2016-03-14:
#### Hardware
Hardware:			[ESP-ADC from InCircuit](http://shop.in-circuit.de/product_info.php?cPath=21&products_id=67)

Core Version:      	Sorry, don't know how to find that out... downloaded from git last week

### Description

the webserver stops working properly after several request.
Transmitting data to the client is interrupted with error -8

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		1MB
CPU Frequency:		does not affect the error
Flash Mode:			does not affect the error
Flash Frequency:	does not affect the error
Upload Using:		SERIAL
Reset Method:		ck

Example that produces the error:

```cpp
#include <ESP8266WiFi.h>
#include <WiFiUdp.h>
#include <ESP8266WebServer.h>

// WiFi connection
char Receive_ssid[33] = "SSID";
char Receive_password[65] = "PASSWORD";
String sResponse;

int WiFiStatus = WL_IDLE_STATUS;

// storage for Measurements; keep some mem free; allocate remainder
#define KEEP_MEM_FREE 10240
unsigned long ulReqcount;       // how often has a valid page been requested
unsigned long ulReconncount;    // how often did we connect to WiFi

// Create an instance of the server on Port 80
ESP8266WebServer server(80);

// needed to avoid link error on ram check
extern "C"
{  
  #include "user_interface.h"
}

void logDetails();

void notFoundHandler()
{
  Serial1.println(F("Not Found Handler"));
  logDetails();
}

String methodToString(HTTPMethod method)
{
  switch(method)
  {
    case HTTP_GET:
      return F("GET");
    case HTTP_POST:
      return F("POST");
    case HTTP_PUT:
      return F("PUT");
    case HTTP_PATCH:
      return F("PATCH");
    case HTTP_DELETE:
      return F("DELETE");
  }
  return F("Unknown");
}

void sendTable()
{
  ulReqcount++;  
  sResponse  = F("<!DOCTYPE HTML>");
  sResponse += F("\r\n<html>\r\n<head>\r\n<meta http-equiv=\"content-type\" content=\"text/html; charset=utf-8\">");
  sResponse += F("\r\n\t<meta http-equiv=\"refresh\" content=\"2; url=/\">");
  sResponse += F("\r\n\t<style>");
  sResponse += F("\r\n\t\tbody {background-color: #dfdfdf;}");
  sResponse += F("\r\n\t\tfont {color: #000000;}");
  sResponse += F("\r\n\t\ttable {font-size: large; width:100%;}");
  sResponse += F("\r\n\t\ttable, th, td {border: 2px solid black; border-collapse: collapse;}");
  sResponse += F("\r\n\t\tth, td {padding: 5px;}");
  sResponse += F("\r\n\t\tth {text-align: left;}");
  sResponse += F("\r\n\t</style>\r\n<title>Test</title>\r\n</head>\r\n<body>");
  sResponse += F("\r\n<h1>Testdata</h1>");

  sResponse += "\r\n<table>";
  sResponse += "\r\n\t<tbody><tr><th>#</th><th>Time</th><th>Calibration used</th><th>Sesnor 1</th><th>Sensor2</th></tr>";
  sResponse += "\r\n\t<tr><td>1</td><td>04.02.2016 09:40:12</td><td>Standard</td><td>25,12</td><td>10,60</td></tr>";
  sResponse += "\r\n\t<tr><td>1</td><td>04.02.2016 09:40:14</td><td>Standard</td><td>25,14</td><td>10,67</td></tr>";
  sResponse += "\r\n\t<tr><td>1</td><td>04.02.2016 09:40:16</td><td>Standard</td><td>25,09</td><td>10,58</td></tr>";
  sResponse += "\r\n\t<tr><td>1</td><td>04.02.2016 09:40:18</td><td>Standard</td><td>25,11</td><td>10,62</td></tr>";
  sResponse += "\r\n\t<tr><td>1</td><td>04.02.2016 09:40:20</td><td>Standard</td><td>25,12</td><td>10,61</td></tr>";
  sResponse += "\r\n\t<tr><td>1</td><td>04.02.2016 09:40:22</td><td>Standard</td><td>25,15</td><td>10,70</td></tr>";
  sResponse += "\r\n\t<tr><td>1</td><td>04.02.2016 09:40:24</td><td>Standard</td><td>25,10</td><td>10,63</td></tr>";
  sResponse += "\r\n\t<tr><td>1</td><td>04.02.2016 09:40:26</td><td>Standard</td><td>25,13</td><td>10,67</td></tr>";
  sResponse += "\r\n</tbody></table>";

    
  sResponse += F("\r\n<div style=\"font-size:x-small\">");
  sResponse += F("\r\n\t<BR>Aufrufz&auml;hler = ");
  sResponse += ulReqcount;
  sResponse += F("\r\n\t<BR>Verbindungsz&auml;hler = ");
  sResponse += ulReconncount;
  sResponse += F("\r\n\t<BR>Freies RAM = ");
  sResponse += (uint32_t)system_get_free_heap_size();
  
  Serial1.println(F("send table...")); 
  logDetails();
  server.send(200, "text/html", sResponse);
}

void logDetails()
{
  Serial1.println("URL is: ") + server.uri();
  Serial1.println("HTTP Method on request was: " + methodToString(server.method()));
  
  // Print how many properties we received and then print their names and values.
  
  Serial1.println("Number of query properties: "+ String(server.args()));
  int i;
  for (i=0; i<server.args(); i++)
  {
    Serial1.println(" - " + server.argName(i) + " = " + server.arg(i));
  }
}

/////////////////////
// the setup routine
/////////////////////
void setup()
{
  // setup globals
  ulReqcount = 0;
  ulReconncount = 0;
  sResponse.reserve(15000);

  // start serial
  Serial.begin(38400);
  Serial1.begin(38400);
  delay(500);
  Serial1.println("  ");
  Serial1.println("  ");
  Serial1.println(F("ESP8266Server send() error test)"));
  Serial1.setDebugOutput(true);
}

///////////////////
// (re-)start WiFi
///////////////////
void WiFiStart(char* ssid, char* password)
{
  unsigned int uiTimeOut = 0;
  ulReconncount++;

  // Connect to WiFi network
  Serial1.println();
  Serial1.print(F("Connecting to "));
  Serial1.print(ssid);
  Serial1.print(F(" with password "));
  Serial1.println(password);

  WiFi.begin((const char*)ssid, (const char*)password);

  while (WiFiStatus != WL_CONNECTED)
  {
    WiFiStatus = WiFi.status();
    delay(500);
    Serial1.print(".");
    uiTimeOut++;
    if(uiTimeOut == 20)
    {      
      Serial1.println("retry");
      WiFi.begin((const char*)ssid, (const char*)password);
    }
    if(uiTimeOut == 40)
    {      
      break;
    }
  }
  if(uiTimeOut >= 40)
  {    
    Serial1.println(F("Could not establish connection!"));
    switch (WiFiStatus)
    {
      case WL_IDLE_STATUS:
      {
        Serial1.println(F("WL_IDLE_STATUS"));
        break;
      }
      case WL_CONNECTED:
      {
        Serial1.println(F("WL_CONNECTED"));
        break;
      }
      case WL_NO_SHIELD:
      {
        Serial1.println(F("WL_NO_SHIELD"));
        break;
      }
      case WL_CONNECT_FAILED:
      {
        Serial1.println(F("WL_CONNECT_FAILED"));
        break;
      }
      case WL_NO_SSID_AVAIL:
      {
        Serial1.println(F("WL_NO_SSID_AVAIL"));
        break;
      }
      case WL_SCAN_COMPLETED:
      {
        Serial1.println(F("WL_SCAN_COMPLETED"));
        break;
      }
      case WL_CONNECTION_LOST:
      {
        Serial1.println(F("WL_CONNECTION_LOST"));
        break;
      }
      case WL_DISCONNECTED:
      {
        Serial1.println(F("WL_DISCONNECTED"));
        break;
      }
      default:
      {
        Serial1.println(F("No idea what went wrong..."));
      }
    }        
    return;
  }
  Serial1.println();
  Serial1.println(F("WiFi connected"));
  
  server.on("/", sendTable);
  
  server.on("/favicon.ico", []()
  {
    server.send(404, "text/plain", "");
  });  
  server.onNotFound(notFoundHandler);
  
  // Start the server
  server.begin();
  Serial1.println(F("Server started"));

  // Print the IP address
  Serial1.print("IP: ");
  Serial1.println(WiFi.localIP());
}

/////////////
// main look
/////////////
void loop()
{ 
  delay(50);

  //////////////////////////////
  // check if WLAN is connected
  //////////////////////////////

  if(WiFi.status() != WL_CONNECTED)
  {
    Serial1.println(F("restart server if not connected"));
    WiFiStart(Receive_ssid, Receive_password);
  }

  server.handleClient();
}
```

 debug output:

>     URL is: /
>     HTTP Method on request was: GET
>     Number of query properties: 0
>     :wr
>     :sent 117
>     :ww
>     :wr
>     :sent 1460
>     :ww
>     :wr
>     :sent 1460
>     :ww
>     :wr
>     :er -8 1460 1
>     :ww
>     :ur 1
>     WS:dis
>     :del

ba-us, 2016-03-21:
Any idea how I can catch this error to handle it somehow?
I'm hoping to build a workaround since there seems to be no cure to this bug!

devyte, 2017-10-18:
I can't reproduce the -8 error with latest git. Closing.
If you think there is still a problem, please open a new issue, and fill in the requested info.

closed 2017-10-18 by devyte