esp8266/Arduino #1773: WiFiClientSecure doesn't work together with ESP8266WebServer

PiDiBi, 2016-03-14:
### Basic Infos

#### Hardware
Hardware:			ESP-8266 - Wemos D1 mini
Core Version:      	2.1.0?

### Description

Hi, I'm sending data over WiFiClientSecure  - everything works fine, but after adding only this line

`ESP8266WebServer server(80);`	

WiFiClientSecure  stops to send data. Probably some problem with non reentrant IP Stack? any idea how to solve it? 

`bytesWritten = tlsClient.print(buildHttpRequest(serializeData(data))); `
returns some bytes if it's work and zero after adding (declaring) ESP8266WebServer

igrr, 2016-03-14:
I think there is not enough RAM for WiFiClientSecure to work along with the web server. WiFiClientSecure RAM usage peaks at 22kb-25kb, so I can imagine that adding a WebServer will make it run out of RAM, hence it will be unable to send or receive anything.

PiDiBi, 2016-03-14:
version with server (not working) before sending - Memory 33576
version wothout server (working) - before sending - Memory 39184 after sending Memory 20080

PiDiBi, 2016-03-15:
you are right, it works in new sketch with server and client only
I will investigate more about it ...

closed 2016-03-15 by PiDiBi

jindrichsirucek, 2017-03-20:
Hi, igrr can You explain why the WiFiClientSecure takes so much of RAM? Its happening to me, whe I connect to Google servers it takes 20kB.. but when I connect to my website it consumes only 10kB.. What is the reason, is it possible to change it somehow? another library or infulenci it in any ways please?