esp8266/Arduino #1774: OTA - magic header, new bin not fits SPI Flash

ikin77, 2016-03-14:
Hi. HTTP OTA is not working to me. If I enable debug I see the following in Serial output:

HTTP_UPDATE_FAILD
[httpUpdate] Header read fin.
[httpUpdate] Server header:
[httpUpdate]  - code: 200
[httpUpdate]  - len: 313152
[httpUpdate] ESP8266 info:
[httpUpdate]  - free Space: 733184
[httpUpdate]  - current Sketch Size: 313140
[httpUpdate] runUpdate flash...
[httpUpdate] magic header, new bin not fits SPI Flash
HTTP_UPDATE_FAILD

The URL where new .bin is being downloaded from ESP8266 is the following:

http://85.54.133.105/OTA/acquasolar.cpp.bin

Any ideas? Thanks in advance!!!!!!!!!!!!!

krzychb, 2016-03-14:
Hi ikin77,

Basing on the above log `new bin not fits SPI Flash`
Have you tried flashing a smaller file?

Krzysztof

ikin77, 2016-03-14:
Hi, krzychb. Thanks for your help. Some thing with a smaller .bin. I've tried to upload just a simple Blink sketch with OTA.

HTTP_UPDATE_FAILD
[httpUpdate] Header read fin.
[httpUpdate] Server header:
[httpUpdate]  - code: 200
[httpUpdate]  - len: 233040
[httpUpdate] ESP8266 info:
[httpUpdate]  - free Space: 733184
[httpUpdate]  - current Sketch Size: 313140
[httpUpdate] runUpdate flash...
[httpUpdate] magic header, new bin not fits SPI Flash
HTTP_UPDATE_FAILD

Anyway, in my first attempt you can see that output says that  733184 is free space, and new sketch is just 313152. It should fit. 

The problem is with this ESP module: 

   http://s1.electrodragon.com/wp-content/uploads/2014/08/ESP-01-new-101.jpg

Up to now, I where using ESP-12E and ESP-12F. No problem with them.

Thanks again!!

mtnbrit, 2016-03-14:
you might want to upgrade to a newer version, if I remember correctly, the error message with the typo FAILD was in a much older version.

krzychb, 2016-03-14:
@mtnbrit,

Good point!

@ikin77,

Besides upgrading I would still verify available flash memory versus the sketch size. The module on the photo likely has 1MB. I guess you have selected 256K SPIFFS and the rest remaining for sketch is shown as the `free Space: 733184`. To use OTA you need at lest twice as much flash as the sketch size plus some extra space for EEPROM and WiFi config - see [Flash layout](https://github.com/esp8266/Arduino/blob/master/doc/filesystem.md#flash-layout).

If this is the case, then I would try to select 64K SPIFFS, load ESP-01 again using serial and then once more but with OTA.

Krzysztof

vlast3k, 2016-03-15:
i was getting this message when i tried flashing a sketch built for 4mb
flash on a 1 mb chip
the sketch was 300k
after few iterations i had success when i configured 1 mb flash and 256 kb
spiffs
(and also using a recent version of the core for both - the flashed stuff
and the bin to be flashed OTA)

ikin77, 2016-03-15:
Ok, thaks all of you :o)

I'll try with a newer version...but I'm having problems with them. My sketch can run for weeks with my current version, and it crashes suddenly if I use stable version 2.1.0. I'll have to do some research on it, surely next weekend. I'll let you know.

@vlast3k  => I'll try your suggestion too. I think my chip is 4Mb, anyway I'll check it.

Links2004, 2016-03-16:
the ```new bin not fits SPI Flash``` mean you have configured a bigger flash size in the IDE during build time then the real hardware have.

ti check the real size you can use:
https://github.com/esp8266/Arduino/blob/master/libraries/esp8266/examples/CheckFlashConfig/CheckFlashConfig.ino

ikin77, 2016-03-16:
Hi, Links2004. I've updated to lastest GIT and run the CheckFlashConfig.ino, with the following output:

Flash real id:   001440E0
Flash real size: 1048576
Flash ide  size: 4194304
Flash ide speed: 40000000
Flash ide mode:  DIO
Flash Chip configuration wrong!

The interest thing is the first value, Flash real id => 001440E0. In EspClass::getFlashChipSizeByChipId you can see it's a Bergmicro 1Mb, so, the problem seems to be found.

Just a question: why can I flash from Arduino 4Mb if flash is just 1Mb?

Thanks for your help and best regards

closed 2016-03-19 by ikin77

Links2004, 2016-03-20:
the Arduino IDE / esptool not checks the flash size and as long your sketch is smaller then 1048576 you not will get a upload / write error.