esp8266/Arduino #1775: Sharing RX/TX electrically?!

tuxedo0801, 2016-03-14:
Hi,

this is maybe not a software stack question, but maybe others faced same problem and solved it already:

How do I connect my ESP's RX/TX pins to a serial device and having at the same time the CH340G connected to the ESP for programming purpose?!

When both are connected (prog-adapter + serial device), programming fails. I need to remove at least RX connection on ESP for the serial device. Of course I can add a kind of jumper to open the RX-connection to the serial device when programming the ESP. But is there something that works "automatically"? 

If I look into the schematic, I often found 340ohm to 4.7kohm between rx/tx and the ESP... 

For testing purpose I have a Lolin/NodeMCU 1.0 boards as well as a ESP12E module.

Regards,
Alex

tuxedo0801, 2016-03-14:
Seems that I've found a solution myself:

https://github.com/esp8266/Arduino/issues/143

Will test it.

me-no-dev, 2016-03-15:
there are many ways to tackle this situation. TX/RX pins can be changed (no need for SoftSerial) or you can add inline resistors, but if your UART to USB is on-board and does not have resistors inline, then that would not be an option.
Watch out for using software serial and low bandwidths. You may run into watchdog being not fed enough.

tuxedo0801, 2016-03-15:
My plan is to put the ESP12 module on my board and have a pin-header for programing. So there's no UART-USB Adapter during runtime.

I will go for the rx/tx pin-swap and check if this solves my issue.
Struggled last night with WDT resets. Used SoftSerial. So thanks for the hint.

tuxedo0801, 2016-03-15:
Another question:

If I connect my serial device to 13/15 and use swap(), the serial device is not affected or disturbs programming/flashing. Works.

But what about the Pins of of UART0? If I connect them to relays (via transistor driver) or something else they might cause some troubles when programming/flashing... 
Is there a recommendation on how this is handled?

closed 2016-06-23 by igrr