esp8266/Arduino #1778: Crypto library ?

HW-Siew, 2016-03-15:
Does Esp8266 support any cryto library?

igrr, 2016-03-15:
You could probably start by looking at the crypto library which is part of axTLS:
https://github.com/igrr/axtls-8266/tree/master/crypto
It has some bits which you may find useful.

HW-Siew, 2016-03-20:
Thanks @igrr.

closed 2016-03-20 by HW-Siew

capablegh, 2016-04-19:
Do you mean that the 8266 SSL sources are based on axTLS? Would [https://tls.mbed.org/](url) offer more features? Definitely mbed TLS build into a larger binary, but perhaps it could be configured to what is needed and the resulting binary could be smaller in size. My interest are validating X509 signatures.

capablegh, 2016-04-19:
I wrote before I investigated more, so attempting correct myself. The ESP8266 port of axTLS provided here does have X509 support. At least I need to investigate that first for my work.