esp8266/Arduino #1779: Version 2.1.0. Module connection time > 2-3 seconds even if static address is used.

robertgregor, 2016-03-15:
Hello, this topic was here already, but still I don't have clear answer about the solution of this.
In fact, I have battery powered device, and I am going to deep sleep. Before going to deep sleep I will set the AP_STA mode, I will assign static IP address:

void connectToWifi() {
  if (WiFi.status() == WL_CONNECTED) return;
  WiFi.mode(WIFI_STA);
  WiFi.config(getIpAddrFromArr(ip), getIpAddrFromArr(gw), getIpAddrFromArr(mask),getIpAddrFromArr(dns));
  WiFi.begin();
  int attempts = 0;
  while (WiFi.status() != WL_CONNECTED) {
      if (++attempts < 150) {
        delay(100);
      } else break;
  }
} 

void setup() {
  connectToWifi();
}

void loop() {
.....
        delay(100);
        ESP.deepSleep(period*1000000, RF_CAL);
        delay(100);
}

Based on some other threads here, it has to connect within 300 ms to wifi. However, in my case, it takes much longer - around 3 seconds.
Also, it is interresting, that the WiFi.config has to be called each time, otherwise the dhcp is done and connection time is even longer. So seems, that WiFi.config doesn't save the configuration to the flash???
Any Idea? I am using latest 2.1.0. I have found in some other threads, that I have to re-flash ESP firmware? Or to erase the flash ?

vlast3k, 2016-03-30:
i was just trying the same. In my case the connection takes ~150 ms. The code is like this
```
  Serial <<endl<< millis() << endl;
  WiFi.mode(WIFI_STA);
  WiFi.config(IPAddress(192,168,1,19), IPAddress(192,168,1,1), IPAddress(255,255,255,0),IPAddress(192,168,1,1));
  WiFi.begin("vladiHome", "0888414447");
  for (int i=0; i <1000; i++) {
    if (WiFi.status() == WL_CONNECTED) {
      break;
    }
    delay(10);
    //Serial << i << ".";
  }
  Serial << WiFi.localIP() << endl;
  Serial << millis();
```

robertgregor, 2016-03-30:
Hello, OK, great! I will try this and will let you know!

robertgregor, 2016-04-01:
Hmmm, I have taken the latest Git version, but still I cannot go below 2 seconds. The code is exactly like you have:

void connectToWifi() {
  if (WiFi.status() == WL_CONNECTED) return;
  Serial.println("Connecting: "+String(millis(),DEC));
  WiFi.mode(WIFI_STA);
  WiFi.config(IPAddress(192,168,1,10),IPAddress(192,168,1,1),IPAddress(255,255,255,0),IPAddress(192,168,1,1));
  //WiFi.begin("Bob","XXXXX");
  WiFi.begin();
  int attempts = 0;
  while (WiFi.status() != WL_CONNECTED) {
      if (++attempts < 500) {
        delay(10);
      } else break;
  }
  if (WiFi.status() == WL_CONNECTED) {
    Serial.println("Connected: "+String(millis(),DEC));
    for (byte thisByte = 0; thisByte < 4; thisByte++) {
      Serial.print(String(WiFi.localIP()[thisByte], DEC));
      if (thisByte != 3) Serial.print(".");
    }
    Serial.println();
  } else {
    Serial.println("Not connected...");
  }
}

void setup() {
  delay(500);
  Serial.begin(115200);
  Serial.print("\nESP chip ID: ");
  Serial.println(ESP.getChipId(), DEC);
  connectToWifi();
}

void loop() {
        connectToWifi();
        delay(10);
        WiFi.forceSleepBegin();
        for (int i=0; i<10; i++) {
          delay(500);
          Serial.print(".");
        }
        WiFi.forceSleepWake();
        delay(10);    
  }

And the output:

ESP chip ID: 1193309
Connecting: 21496
Connected: 23756
192.168.1.10
..........Connecting: 28985
Connected: 31133
192.168.1.10
..........Connecting: 36360
Connected: 38509
192.168.1.10
..........Connecting: 43738
Connected: 45886
192.168.1.10
..........Connecting: 51116
Connected: 53263
192.168.1.10
..........Connecting: 58494
Connected: 60641
192.168.1.10
..........Connecting: 65870
Connected: 68017
192.168.1.10
..
So the connection time is 2 - 2.5 seconds...

Any idea?

lbischof, 2017-07-26:
Any progress? Having the same problem

devyte, 2017-10-18:
The connection time depends on several things, e.g.: persistent, encryption type, calling begin()/begin(ssid)/begin(ssid,pass)/begin(ssid,pass,bssid), static/dynamic IP, etc. I believe I saw a post on a discussion forum about it with a comparison table.
No concrete issue for the repo => closing.

closed 2017-10-18 by devyte