esp8266/Arduino #1780: WeMos D1 R2 - Incorrect behaviour

lxnx0, 2016-03-15:
### Basic Infos

#### Hardware
Hardware:			WeMos D1 R2
Core Version:      	N/A

### Description

 I'm not sure this really a bug or I just did something wrongly. But based on my testing the digitalWrite return the wrong result.

digitalWrite(BUILTIN_LED, HIGH) --> Behaves as LOW
digitalWrite(BUILTIN_LED, LOW) --> Behaves as HIGH

### Settings in IDE

*platformio.ini*

[env:d1_mini]
platform = espressif
framework = arduino
board = d1_mini
upload_speed = 115200

### Sketch

*Sample 1*
```cpp

#include <Arduino.h>

void setup() {
  pinMode(BUILTIN_LED, OUTPUT);  // initialize onboard LED as output
  Serial.begin(115200);
}

void loop() {
  digitalWrite(BUILTIN_LED, HIGH);  // turn on LED with voltage HIGH
  Serial.println("ON");
  delay(1000);                      // wait one second
  digitalWrite(BUILTIN_LED, LOW);   // turn off LED with voltage LOW
  Serial.println("OFF");
  delay(1000);                      // wait one second
}
```

*Sample 2*

```cpp
  server.on("/on", [](){
    server.send(200, "text/plain", "ON");
    digitalWrite(BUILTIN_LED, 1);  // turn on LED with voltage HIGH
  });

  server.on("/off", [](){
    server.send(200, "text/plain", "OFF");
    digitalWrite(BUILTIN_LED, 0);  // turn off LED with voltage HIGH
  });

```

### Debug Messages

```
N/A
```

WereCatf, 2016-03-15:
@lxnx0 The LED behaves the same on my Nodemcu-clone boards. I do not have one of these Wemos-boards, but on the Nodemcu it's just how the LED is wired physically.

pieman64, 2016-03-15:
See schematic at http://www.wemos.cc/Products/d1_r2.html the BUILTIN-LED has a 10K pull up resistor so it will be lit when LOW.

lxnx0, 2016-03-16:
Thanks @WereCatf & @pieman64 for the clarification.

closed 2016-03-16 by lxnx0