esp8266/Arduino #1782: Reboot Stuck after ESPhttpUpdate

craigfoo, 2016-03-16:
#### Hardware
Hardware:		ESP-12 (ESP8266 Feather HUZZAH)
Core Version:      	2.1.0-rc2

### Description

After the .bin file is downloaded from the server, the ESP8266 reboots but gets stuck and doesn't run the new program until a physical reset happens (a button in this case on the ESP8266 Feather).

### Settings in IDE

Module:  			ESP-12
Flash Size: 		4MB
CPU Frequency:	80Mhz
Upload Using:		SERIAL

### Sketch

```
/**
 * httpUpdate.ino
 *
 *  Created on: 27.11.2015
 *
 */

#include <Arduino.h>

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>

#include <ESP8266HTTPClient.h>
#include <ESP8266httpUpdate.h>

#define USE_SERIAL Serial

ESP8266WiFiMulti WiFiMulti;

void setup() {

    USE_SERIAL.begin(115200);
    // USE_SERIAL.setDebugOutput(true);

    USE_SERIAL.println();
    USE_SERIAL.println();
    USE_SERIAL.println();

    for(uint8_t t = 4; t > 0; t--) {
        USE_SERIAL.printf("[SETUP] WAIT %d...\n", t);
        USE_SERIAL.flush();
        delay(1000);
    }

    WiFiMulti.addAP("SSID", "PASSWORD");

}

void loop() {
    // wait for WiFi connection
    if((WiFiMulti.run() == WL_CONNECTED)) {

        t_httpUpdate_return ret = ESPhttpUpdate.update("http://server/file.bin");
        //t_httpUpdate_return  ret = ESPhttpUpdate.update("https://server/file.bin");

        switch(ret) {
            case HTTP_UPDATE_FAILED:
                USE_SERIAL.printf("HTTP_UPDATE_FAILD Error (%d): %s", ESPhttpUpdate.getLastError(), ESPhttpUpdate.getLastErrorString().c_str());
                break;

            case HTTP_UPDATE_NO_UPDATES:
                USE_SERIAL.println("HTTP_UPDATE_NO_UPDATES");
                break;

            case HTTP_UPDATE_OK:
                USE_SERIAL.println("HTTP_UPDATE_OK");
                break;
        }
    }
}
```

### Console Output

```
[SETUP] WAIT 3...
[SETUP] WAIT 2...
[SETUP] WAIT 1...

 ets Jan  8 2013,rst cause:2, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset
```

igrr, 2016-03-16:
Have you done physical reset after serial programming and before OTA update?

craigfoo, 2016-03-16:
@igrr, hmmm. Here's what I just did and it seemed to work:

1. Build and flash the OTA example (using PlatformIO with Eclispe). Once this is done loading it runs immediately (which I was doing before).
2. Physical reset before it gets to the update function.
3. Program runs, gets the update, and runs the new code. 
### Console Ouput

```
[SETUP] WAIT 4...
[SETUP] WAIT 3...
[SETUP] WAIT 2...
[SETUP] WAIT 1...

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
@cp:0
ld
```
I guess it would work if I didn't have to serial program it first. Not sure what's happening there.

igrr, 2016-03-16:
This is a pretty old issue which isn't resolved yet: #1017.
Basically reset after OTA doesn't happen correctly if the board was not reset after serial upload.

craigfoo, 2016-03-16:
@igrr, aha, didn't see that one.

closed 2016-03-17 by igrr

deepikavira, 2016-08-10:
@craigfoo Hi I am trying to understand how did you develop your server from where ESP was supposed to download the .bin file? If I am not wrong "http://server/file.bin" this is the server for which you would have to write a simple http code and have the bin file saved so that whenever ESP sends HTTP request from anywhere in the world, this server can respond (based on whether update is required or no?)

craigfoo, 2016-08-11:
@deepikavira, it's as simple as just dumping a .bin file on some server via FTP and then pointing your ESP code to that file location. There is nothing fancy on the server side.

deepikavira, 2016-08-11:
@craigfoo Thank you I will try it out.

deepikavira, 2016-08-12:
@craigfoo have you tried connecting to SFTP? Last  time checked support for secured connection is still being worked on, correct me if I am wrong.

craigfoo, 2016-08-12:
@deepikavira, not sure about SFTP. Mine is regular ol' FTP.