esp8266/Arduino #1783: free() function error

ikin77, 2016-03-16:
I'm having problems with free() function. With older releases, the problem didn't exist. For instance, consider the following sample sketch:

#include "time.h"

void setup() {
  Serial.begin(115200);
}

void loop() {
  time_t localeTime = 100000 + (millis());
  struct tm* aux = localtime(&localeTime);
  free(aux);
  Serial.println("loop");
  delay(500);
}

It crashes suddenly, but I should run for ages. If you comment the "free" statement, it works (up to memory is filled, I guess).

Any ideas?

Thanks in advance and best regards!!

sticilface, 2016-03-16:
Looks like you are trying to free something not allocated with malloc... it'll be on the stack and cause the crash!  As aux is declared in the loop, it is destroyed when it goes out of scope.  So no need to call free.

Does the program really crash if you leave it?  

Change print loop to 
```
Serial.printf("loop %y\n",ESP.getFreeHeap()); 
```
and see if you really are loosing memory..  I suspect not.

igrr, 2016-03-16:
Localtime doesn't allocate memory. Please check 
http://www.cplusplus.com/reference/ctime/localtime/

    The returned value points to an internal object whose validity or value may be altered by any subsequent call to gmtime or localtime.

ikin77, 2016-03-16:
Thanks to all of you. I'm closing the issue, the problem is now clear to me. Anyway, this was working on older versions...that's I wasn't migrating to new versions. Now, I can do it :o)

Thanks again!

closed 2016-03-16 by ikin77